ros-navigation/navigation2 #2418: main branch needed enviroment

badaolaziang, 2021-06-19:
Hi, at first show my sincere appreciation for your excellent work. I used ROS1 navigation a lot and just begin to touch ROS2 navigation2, and I saw several branches: main/foxy-devel/galactic with different version update frequency...... Currently I installed ros2 foxy version in ubuntu20.04, and I wondered what environment(ros/ubuntu version) is needed for main branch？

SteveMacenski, 2021-06-19:
ROS2 master builds right now, but `rolling` in the somewhat near term future. It expects Ubuntu 20.04

SteveMacenski, 2021-06-19:
This is a question better suited to ROS Answers or checking out our documentation website https://navigation.ros.org/! Closing but will follow up with any questions you might have on ROS Answers if you have followup.

closed 2021-06-19 by SteveMacenski