ros-navigation/navigation2 #2422: Regulated pure pursuit on large robots

stevemartinov, 2021-06-23:
- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - ROS2 Foxy, binaries
- Version or commit hash:
  - ros-foxy-navigation2
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
Just run the planner with default configurations but increse the footprint of the robot to 1 x 1.5 meters.

#### Expected behavior
Robot should navigate and if there is an obstacle next to it, trigger the collision check to true and stop the robot

#### Actual behavior
When visualizing the arc, it actually starts from the base_link (middle of the robot) and the length of the arc does not go beyond the footrpint as the footprint is so long. Therefore, in many cases the planner fails to detect the collision

#### Additional information
I am not sure whether it is a bug as I was not able to properly understand. In README the picture has some lookeahead radius and I presume that the arc should start from the footprint and not from the base link, otherwise, we would need to set the lookahead time to a very large number for the arc to check the collisions beyound the footprint
----

SteveMacenski, 2021-06-23:
What is your path planner? If you planner isn't generating kinematically feasible paths (e.g. paths your robot can literally drive and be collision free driving) then any Pure Pursuit controller will have problems because they're pure path following (pure "pursuit"). They only see the path and have interest in following that path. 

We also check for collision https://github.com/ros-planning/navigation2/blob/main/nav2_regulated_pure_pursuit_controller/src/regulated_pure_pursuit_controller.cpp#L404-L416 but assumes you have your costmap properly setup with a footprint and inflation layer. We use this to forward project in time to make sure there is no collision in the immediate future https://github.com/ros-planning/navigation2/blob/main/nav2_regulated_pure_pursuit_controller/src/regulated_pure_pursuit_controller.cpp#L373-L397 based on this pure position 2D costmap collision check. 

It is possible to improve this by using the footprint and using the full SE2 footprint checking, but it really depends on "why" your collisions are happening to know if that would help at all.

stevemartinov, 2021-06-23:
@SteveMacenski I am using a simple global planner (Djikstra, old ROS1 implementation) but I am not sure whether it generates kinematically feasable paths, do you have one in mind that does that?

Yes, I saw that function and I think I can implement a collision checking incorporating a whole footprint. I have a question though, in your function:
https://github.com/ros-planning/navigation2/blob/main/nav2_regulated_pure_pursuit_controller/src/regulated_pure_pursuit_controller.cpp#L404-L416

I assume you are already passing points where robot will be based on the lookeahead distance and time. If I use those points and check the whole footprint (meaning that if I treat the passed point as a middle of the robot and check that the point itself + all the bounding footprint is within the NO_OBSTACLE zone), will that be ok?

vinnnyr, 2021-06-28:
@stevemartinov  RE: planner that generates kinetically feasible paths: If you wanted to use something in ros2, you can use smac_planner https://navigation.ros.org/configuration/packages/configuring-smac-planner.html

SteveMacenski, 2021-06-28:
> but I am not sure whether it generates kinematically feasable paths, do you have one in mind that does that?

Any naive grid search planner is not kinematically feasible. Since this controller is purely following the path, if your path is not follow-able, then ergo, you would not expect this controller (and your planner) to be suitable for your needs. Please read the documentation on this controller and make sure you use an appropriate planner who generates plans that can actually be followed. I added another paragraph making this explicit https://github.com/ros-planning/navigation2/blob/main/nav2_regulated_pure_pursuit_controller/README.md#pure-pursuit-basics 

Paired with Smac planner, or something else, you should see this work well. Or if you can make your planner plan based on the **outer most radius** as the `robot_radius` such that you can treat your robot like a giant particle for planning, assuming your envionment will allow that.

> I assume you are already passing points where robot will be based on the lookeahead distance and time.

I do not understand. The plans are in the robot base frame, so you should assume that at the look ahead points / path points, you're referencing your robot base reference frame

stevemartinov, 2021-06-30:
@SteveMacenski Yes, I now read everything in great detail. I have one question though, when you check for the potential collision, I think there might be a bit better solution.

Suppose there is an obstacle at 1 second ahead and robot stopped. Then on the next callback, your code checks for collisions but because the robot has already stopped, it will not check the same obstacle that was 1 second infront when the robot was driving at higher velocity. Thus, the robot will start to move abruptly until it comes very close to the obstacle.

Do you think it will be reasonable to add some "saved" linear and angular velocities to rechech that the previous obstacle has gone?

SteveMacenski, 2021-06-30:
> Suppose there is an obstacle at 1 second ahead and robot stopped.

If the robot stopped, then there is no "next callback" because the controller failed and it will return to the server that failure. If you have the controller patience set to 0, then it would return to the BT server and start processing recovery behaviors / replanning before re-trying to navigate. If that patience is non-zero then you could try to move forward, but you'd still be safely 1 seconds from collision at all times. 

> Do you think it will be reasonable to add some "saved" linear and angular velocities to rechech that the previous obstacle has gone?

I don't think that's reasonable, no. It would be reasonable though to have checking a minimum time in the future **and** a minimum distance such that even if stopped, there's some distance (say 10cm or something) it always checks anyway even if > 1 second in the future. That could help with the slower speed issues you describe and still be generalized. I'd approve a PR adding that parameter / feature!

SteveMacenski, 2021-09-09:
@stevemartinov Please test out PR #2556 or at least confirm this solves your concerns, I believe it should.

It updates the collision checker to use the full footprint rather than the point costs in the robot pose. That makes it so that during the projected collision checking, it should now check outside of the robot's pose when projecting forward for any arbitrary robot, using the footprint itself.

closed 2021-11-11 by SteveMacenski