ros-navigation/navigation2 #2428: Cannot run slam demo from documentation

Serafadam, 2021-06-29:
## Bug report

**Required Info:**

- Operating System:
- Kubuntu 20.04
- ROS2 Version:
  - Foxy binaries
- Version or commit hash:
  - ros-foxy-navigation2	0.4.7-1focal.20210513.224843->
- DDS implementation:
  - Default

#### Steps to reproduce issue
- Install Nav stack as per https://navigation.ros.org/tutorials/docs/navigation2_with_slam.html
- Run `ros2 launch nav2_bringup tb3_simulation_launch.py slam:=True `

#### Expected behavior
- Output as in tutorial

#### Actual behavior
- The program crashes, info below:
```
ros2 launch nav2_bringup tb3_simulation_launch.py slam:=True                                                                                                                                                                   ✔  7s    15:20:37   
[INFO] [launch]: All log files can be found below /home/username/.ros/log/2021-06-29-15-22-58-726754-username-pc
[INFO] [launch]: Default logging verbosity is set to INFO
/opt/ros/foxy/share/nav2_bringup/launch/slam_launch.py:77: UserWarning: The parameter 'node_executable' is deprecated, use 'executable' instead
  start_map_saver_server_cmd = Node(
/opt/ros/foxy/share/nav2_bringup/launch/slam_launch.py:83: UserWarning: The parameter 'node_executable' is deprecated, use 'executable' instead
  start_lifecycle_manager_cmd = Node(
/opt/ros/foxy/share/nav2_bringup/launch/slam_launch.py:83: UserWarning: The parameter 'node_name' is deprecated, use 'name' instead
  start_lifecycle_manager_cmd = Node(
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:271> exception=ImportError("cannot import name 'HasNodeParams' from 'nav2_common.launch' (/opt/ros/foxy/lib/python3.8/site-packages/nav2_common/launch/__init__.py)")>
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 273, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 293, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 6 more times]
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/include_launch_description.py", line 125, in execute
    launch_description = self.__launch_description_source.get_launch_description(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_source.py", line 84, in get_launch_description
    self._get_launch_description(self.__expanded_location)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_sources/python_launch_description_source.py", line 51, in _get_launch_description
    return get_launch_description_from_python_launch_file(location)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_sources/python_launch_file_utilities.py", line 62, in get_launch_description_from_python_launch_file
    launch_file_module = load_python_launch_file_as_module(python_launch_file_path)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_sources/python_launch_file_utilities.py", line 37, in load_python_launch_file_as_module
    loader.exec_module(mod)
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/opt/ros/foxy/share/slam_toolbox/launch/online_sync_launch.py", line 9, in <module>
    from nav2_common.launch import HasNodeParams
ImportError: cannot import name 'HasNodeParams' from 'nav2_common.launch' (/opt/ros/foxy/lib/python3.8/site-packages/nav2_common/launch/__init__.py)
Task exception was never retrieved
future: <Task finished name='Task-20' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:271> exception=RuntimeError('Signal event received before subprocess transport available.')>
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 273, in _process_one_event
    await self.__process_event(next_event)
[... 17 lines cut ...]
[INFO] [gzserver-1]: sending signal 'SIGINT' to process[gzserver-1]
[robot_state_publisher-3] [INFO] [1624972978.837052772] [rclcpp]: signal_handler(signal_value=2)
[ERROR] [gzclient-2]: process has died [pid 82609, exit code -2, cmd 'gzclient'].
[ERROR] [gzserver-1]: process has died [pid 82607, exit code -2, cmd 'gzserver -s libgazebo_ros_init.so /opt/ros/foxy/share/nav2_bringup/worlds/waffle.model'].
[robot_state_publisher-3] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[robot_state_publisher-3]   what():  failed to initialize rcl node: the given context is not valid, either rcl_init() was not called or rcl_shutdown() was called., at /tmp/binarydeb/ros-foxy-rcl-1.1.10/src/rcl/guard_condition.c:67
[ERROR] [robot_state_publisher-3]: process has died [pid 82611, exit code -6, cmd '/opt/ros/foxy/lib/robot_state_publisher/robot_state_publisher /opt/ros/foxy/share/nav2_bringup/urdf/turtlebot3_waffle.urdf --ros-args -r __node:=robot_state_publisher -r __ns:=/ --params-file /tmp/launch_params_royvld_r -r /tf:=tf -r /tf_static:=tf_static'].
[rviz2-4] [INFO] [1624972979.141694589] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-4] [INFO] [1624972979.141906598] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-4] [INFO] [1624972979.169358429] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-4] [WARN] [1624972979.286279709] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[rviz2-4] [WARN] [1624972979.289134500] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[rviz2-4] [WARN] [1624972979.291791116] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[rviz2-4] [INFO] [1624972979.305378532] [rviz2]: Waiting for the lifecycle_manager_navigation/is_active service...
[rviz2-4] [INFO] [1624972980.305575604] [rviz2]: Waiting for the lifecycle_manager_navigation/is_active service...
[rviz2-4] [INFO] [1624972981.305758233] [rviz2]: Waiting for the lifecycle_manager_navigation/is_active service...
[rviz2-4] [INFO] [1624972982.305939631] [rviz2]: Waiting for the lifecycle_manager_navigation/is_active service...
[rviz2-4] [INFO] [1624972983.306214428] [rviz2]: Waiting for the lifecycle_manager_navigation/is_active service...
[ERROR] [rviz2-4]: process[rviz2-4] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[INFO] [rviz2-4]: sending signal 'SIGTERM' to process[rviz2-4]
[ERROR] [rviz2-4]: process has died [pid 82613, exit code -15, cmd '/opt/ros/foxy/lib/rviz2/rviz2 -d /opt/ros/foxy/share/nav2_bringup/rviz/nav2_default_view.rviz --ros-args -r __node:=rviz2'].
```

#### Additional information
- Running this example without `slam=True` launches everything correctly

cjh1995-ros, 2021-06-29:
I have same issue too. Even if I put it with slam=False or slam=True, it doesnt work.
[image]
[image]
[image]
plz somebody help me....

cjh1995-ros, 2021-06-29:
> I have same issue too. Even if I put it with slam=False or slam=True, it doesnt work.
> [image]
> [image]
> [image]
> plz somebody help me....

Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist

why that does happen?

SteveMacenski, 2021-06-29:
I do not see any such issues. I just went on a machine, did a clean install of ROS2 Foxy, installed nav2 / nav2-bringup packages, and launched the TB3 simulation with and without the SLAM field. It came up find each time (though with `slam:=True` you do get a couple of warnings about `node_name` vs `name` but you can safely ignore those). I'd recommend seeing what, if anything, you've changed in your system that might be causing this issue. 

@cjh1995-ros do not hijack someone elses ticket for your own problems. Please read the documentation and if you still cant find a solution to your problem, ask on ROS Answers.

Serafadam, 2021-06-29:
Ok, it seems like I had to upgrade my packages for this to work, sorry for the commotion. On a side note, this specific Gazebo simulation seems to take a lot of time to launch, is this something that is normal, or is this something tied to my pc? (I'm not sure whether I should create another issue for that)

SteveMacenski, 2021-06-29:
Its a mixture of a gazebo problem and the size of the meshes for the turtlebot model. Its nothing on your side, my computer goes in and out of having issues like that. I haven't quite been able to isolate exactly why that happens at some times but not at others. Suffice it to say its not your fault or nav2's fault, but some mixture of gazebo, gazebo ros, and turtlebot.

Does that fix your issue? Can we close this ticket?

Serafadam, 2021-06-29:
Yes, thank you for the answers :slightly_smiling_face:

closed 2021-06-29 by Serafadam

SteveMacenski, 2021-06-29:
No worries, I don't use the binaries much so I wanted to make sure I didn't release some messed up code