ros-navigation/navigation2 #2433: Fix broken tests created after Galactic release

SteveMacenski, 2021-07-02:
Since the Galactic release, our CI has been introduced with a number of failing tests. We fixed the trivial ones of linter profile changes, but there are still several nav2_utils tests that are failing that probably just need to be tweaked to pass again.

Goal: Look over the failures in CI, resolve them, and merge into the `main` branch

SteveMacenski, 2021-07-02:
e.x. https://app.circleci.com/pipelines/github/ros-planning/navigation2/5602/workflows/02427c64-ce4a-4e0b-af82-db0f574413c9/jobs/20522

karl-schulz, 2021-07-09:
Hi,
is this issue the reason why the Nav2 binaries for galactic are not yet available? Or are they just not released yet for other reasons?

`sudo -H apt-get install -y -qq ros-galactic-nav2-core` 
yields
`E: Unable to locate package ros-galactic-nav2-core`

SteveMacenski, 2021-07-09:
We're just waiting on a Galactic sync. I barely missed the first release of Galactic so some of the packages made it through and some did not. They're going to do a sync in the next few days so you should be able to get binaries then, the test failures in CI didn't block that.

SteveMacenski, 2021-08-03:
We have flaky tests now, which is a slightly different problem. The core of this has been addressed and CI is turning over

closed 2021-08-03 by SteveMacenski