ros-navigation/navigation2 #2437: Planner SIGSEGV, segmentation fault.

seasony-vp, 2021-07-06:
## Bug report

Operating system : Ubuntu 20.04
ROS2 version: Foxy
Nav2 branch: Foxy-devel
DDS: FastRTPS

#### Steps to reproduce issue

Run a navigation loop using SLAM toolbox and the SMAC planner.
Set the re-planning frequency to 1hz and the map update frequency to 1 second as well.

#### Expected behavior

The planner doesn't crash.

#### Actual behavior

The planner crashes within 15 minutes. 

#### Additional information

Setting the re-planning and map publishing frequency lower extends the time before a SIGSEGV crash.
I suspect this is caused because the global cost-map is cleared every one second due to the map update and there is a window of time where the planner needs to access it however the pointer to the actual cost-map value has just been cleared, or the planner is accesing a value and the map update clears it and thus it is not able to access it.

GBD backtrace: 

```
(gdb) backtrace
#0  0x00007ffff77dd68c in nav2_costmap_2d::Costmap2D::getCost(unsigned int, unsigned int) const ()
   from /home/watney-nuci7/seasony/lewis/install/nav2_costmap_2d/lib/libnav2_costmap_2d_core.so
#1  0x00007ffff02ec555 in smac_planner::UnconstrainedSmootherCostFunction::Evaluate(double const*, double*, double*) const ()
   from /home/watney-nuci7/seasony/lewis/install/smac_planner/lib/libsmac_planner_2d.so
#2  0x00007fffd9e56278 in ceres::GradientProblem::Evaluate(double const*, double*, double*) const
    () from /lib/libceres.so.1
#3  0x00007fffd9e5902c in ceres::internal::GradientProblemEvaluator::Evaluate(ceres::internal::Evaluator::EvaluateOptions const&, double const*, double*, double*, double*, ceres::internal::SparseMatrix*) () from /lib/libceres.so.1
#4  0x00007fffd9e5f592 in ceres::internal::LineSearchFunction::Evaluate(double, bool, ceres::internal::FunctionSample*) () from /lib/libceres.so.1
#5  0x00007fffd9e62179 in ceres::internal::WolfeLineSearch::BracketingPhase(ceres::internal::FunctionSample const&, double, ceres::internal::FunctionSample*, ceres::internal::FunctionSample*, bool*, ceres::internal::LineSearch::Summary*) const () from /lib/libceres.so.1
#6  0x00007fffd9e6579a in ceres::internal::WolfeLineSearch::DoSearch(double, double, double, ceres::internal::LineSearch::Summary*) const () from /lib/libceres.so.1
#7  0x00007fffd9e60562 in ceres::internal::LineSearch::Search(double, double, double, ceres::internal::LineSearch::Summary*) const () from /lib/libceres.so.1
#8  0x00007fffd9e6cc9d in ceres::internal::LineSearchMinimizer::Minimize(ceres::internal::Minimizer::Options const&, double*, ceres::Solver::Summary*) () from /lib/libceres.so.1
#9  0x00007fffd9e57a72 in ceres::GradientProblemSolver::Solve(ceres::GradientProblemSolver::Options const&, ceres::GradientProblem const&, double*, ceres::GradientProblemSolver::Summary*) ()
   from /lib/libceres.so.1
#10 0x00007fffd9e58721 in ceres::Solve(ceres::GradientProblemSolver::Options const&, ceres::GradientProblem const&, double*, ceres::GradientProblemSolver::Summary*) () from /lib/libceres.so.1
--Type <RET> for more, q to quit, c to continue without paging--
#11 0x00007ffff02ea43a in smac_planner::SmacPlanner2D::createPlan(geometry_msgs::msg::PoseStamped_<std::allocator<void> > const&, geometry_msgs::msg::PoseStamped_<std::allocator<void> > const&) ()
   from /home/watney-nuci7/seasony/lewis/install/smac_planner/lib/libsmac_planner_2d.so
#12 0x00007ffff7f6c621 in nav2_planner::PlannerServer::getPlan(geometry_msgs::msg::PoseStamped_<std::allocator<void> > const&, geometry_msgs::msg::PoseStamped_<std::allocator<void> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
   from /home/watney-nuci7/seasony/lewis/install/nav2_planner/lib/libplanner_server_core.so
#13 0x00007ffff7f70a7d in nav2_planner::PlannerServer::computePlan() ()
   from /home/watney-nuci7/seasony/lewis/install/nav2_planner/lib/libplanner_server_core.so
#14 0x00007ffff7f86cc6 in nav2_util::SimpleActionServer<nav2_msgs::action::ComputePathToPose, rclcpp::Node>::work() ()
   from /home/watney-nuci7/seasony/lewis/install/nav2_planner/lib/libplanner_server_core.so
#15 0x00007ffff7f87e30 in std::_Function_handler<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> (), std::__future_base::_Task_setter<std::unique_ptr<std::__future_base::_Result<void>, std::__future_base::_Result_base::_Deleter>, std::thread::_Invoker<std::tuple<nav2_util::SimpleActionServer<nav2_msgs::action::ComputePathToPose, rclcpp::Node>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::ComputePathToPose> >)::{lambda()#1}> >, void> >::_M_invoke(std::_Any_data const&) ()
   from /home/watney-nuci7/seasony/lewis/install/nav2_planner/lib/libplanner_server_core.so
#16 0x00007ffff7f73d3d in std::__future_base::_State_baseV2::_M_do_set(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*) ()
   from /home/watney-nuci7/seasony/lewis/install/nav2_planner/lib/libplanner_server_core.so
#17 0x00007ffff7cbf47f in __pthread_once_slow (once_control=0x7fffd0002e88, 
    init_routine=0x7ffff7ba0c20 <__once_proxy>) at pthread_once.c:116
#18 0x00007ffff7f78618 in std::thread::_State_impl<std::thread::_Invoker<std::tuple<std::__future_base::_Async_state_impl<std::thread::_Invoker<std::tuple<nav2_util::SimpleActionServer<nav2_msgs::ac--Type <RET> for more, q to quit, c to continue without paging--
tion::ComputePathToPose, rclcpp::Node>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::ComputePathToPose> >)::{lambda()#1}> >, void>::_Async_state_impl(std::tuple<nav2_util::SimpleActionServer<nav2_msgs::action::ComputePathToPose, rclcpp::Node>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::ComputePathToPose> >)::{lambda()#1}>&&)::{lambda()#1}> > >::_M_run() ()
   from /home/watney-nuci7/seasony/lewis/install/nav2_planner/lib/libplanner_server_core.so
#19 0x00007ffff7ba1de4 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#20 0x00007ffff7cb6609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#21 0x00007ffff79e0293 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

```

SteveMacenski, 2021-07-06:
My recommendation would be to remove smoothing parameter to false. In the not-too-far-off future we'll be migrating changes from main to foxy that includes a total rewrite of the smoother in SMAC that will fix this issue (as well as improve path quality and planning times)

SteveMacenski, 2021-07-06:
This is not good though, someone in #2332 is able to reproduce in another algorithm, so clearly there's something wrong here and its not actually an issue with the smoother but in accessing costmap data. I don't think we can safely ignore it as a smoother issue that's been implicitly solved by removing that code. 

Can you try reverting https://github.com/ros-planning/navigation2/commit/57d49e119964df93d07d9ed7b99ad361e5736975 and/or https://github.com/ros-planning/navigation2/commit/1160100d9689bea4c149031bb82bf7d182ec0645? That's the only change I can find that's recent in Foxy and it touches costmap_2d which makes me suspect. I don't see where that would have caused the issue, but its possible it has.

seasony-vp, 2021-07-07:
Okay, so setting the smoother parameter to `false` seems to have done the trick, 45 minutes in and no segfaults. Thank you!

To satisfy our curiosity though i looked at where the `getCost` function is being called within the `smoother_cost_function.hpp` file and found it's being called [here](https://github.com/ros-planning/navigation2/blob/10650836e639f6c6cdb529f78a049dd0623ee46a/smac_planner/include/smac_planner/smoother_cost_function.hpp#L427) multiple times, could this be culprit, calling for a value out of index? And as you've mentioned this is totally rewritten in the main branch so once back-ported the issue should be fixed..

As for reverting to the commits you mentioned, I'm already on a commit before those so i don't think those broke anything. 

I will let this run overnight and see how it progresses, if successful we can close this issue imo.     

Once again, thank you! :)

SteveMacenski, 2021-07-07:
Its odd, certainly. We lock the costmap mutex https://github.com/ros-planning/navigation2/blob/foxy-devel/smac_planner/src/smac_planner.cpp#L220 at the start of this function, so there shouldn't be any changes in size to the costmap during the execution (e.g. there should not be anything special about the smoother looking at cost values versus the planner itself). Though I think if my analysis is correct, all of the costmap layers are only locking their own internal costmaps, not the master costmap when updating with values. And the master costmap is only locked in those algorithm calls which should prevent resizing but not value changes. 

I would be interested if you could set that back to `true` but somehow log the index its trying to get and what the current costmap size is. If it really is an out of bounds issue, then it would be solved by removing this smoother entirely. If its something else, that is a bit concerning.

seasony-vp, 2021-07-08:
Okay, so I let it run overnight and in the morning it was still running, 12+ hours of constant navigation loops! So it seems that the issue was indeed within the smoother. 

I set the `smooth_path` parameter back to `true` and logged the current size of the costmap as defined [here](https://github.com/ros-planning/navigation2/blob/10650836e639f6c6cdb529f78a049dd0623ee46a/nav2_costmap_2d/src/costmap_2d.cpp#L72) together with the index being called [here](https://github.com/ros-planning/navigation2/blob/10650836e639f6c6cdb529f78a049dd0623ee46a/nav2_costmap_2d/src/costmap_2d.cpp#L210) like this:
(the `costmap_planner_` is just a bool flag so only the costmap inside the smoother activates this log)
```
  if (costmap_planner_){
    std::cout << "index:" << getIndex(mx, my) << std::endl;
    std::cout <<  "size: " << (size) << std::endl;
  }
  return costmap_[getIndex(mx, my)];
```
this is the output just before a sigsegv:

```
index:1710
size: 160400
index:106
size: 160400
index:4294966600
size: 160400
--Type <RET> for more, q to quit, c to continue without paging--

Thread 280 "planner_server" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffbef2a700 (LWP 816606)]
0x00007ffff77dd69e in nav2_costmap_2d::Costmap2D::getCost(unsigned int, unsigned int) const ()
   from /home/watney-nuci7/seasony/lewis/install/nav2_costmap_2d/lib/libnav2_costmap_2d_core.so
(gdb) 
```
I think we can safely assert that this is a out of bounds issue caused by the smoother implementation. You're welcome to close this issue unless you still want to see the costmap / smac planner configs and/or an example plan.  

Thank you for your help!

SteveMacenski, 2021-07-08:
I want to leave this open since its problematic, but I can retarget the intent to `backport smac changes to foxy/galactic`. Thanks for debugging and letting me know!

seasony-vp, 2021-07-09:
I most definitely have the inclination however the timing might be an issue. I think sometime in August - September I could squeeze this in (hopefully)!

SteveMacenski, 2021-07-13:
Backported to Galactic, unfortunately, it could not be (easily) backported to Foxy since the changes are _extensive_. I think it will be my tentative position that it won't be backported to Foxy, but it will be available to Galactic!

closed 2021-07-13 by SteveMacenski