ros-navigation/navigation2 #2438: NavigateToPose action msg field missing

Yousseflah, 2021-07-07:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Foxy/Galactic
- Version or commit hash:
  - nav2 from binary
- DDS implementation:
  - 

#### Steps to reproduce issue
ros2 interface show nav2_msgs/action/NavigateToPose
```

```

#### Expected behavior
status field need to be added in result section
#### Actual behavior
In result section, there is only std_msgs/Empty result
#### Additional information

----

SteveMacenski, 2021-07-07:
What is your suggestion to add to the result?

Yousseflah, 2021-07-07:
When i run the navigateToPose action, i see that there is a "status" field returned in the response which is not mentionned in the interface. I suggest to add the field in the "#result definition" section. That's all.

SteveMacenski, 2021-07-07:
I think what you're mixing up is the idea of the Action Result and the Action's Result (read that slowly :laughing: ) When you interact with the Action API you can get the result which is giving you the result code for the action https://github.com/ros2/rcl_interfaces/blob/master/action_msgs/msg/GoalStatus.msg but that is **not** provided by the Action msg definition, that is an attribute of the action server itself. It can return a failure / success / etc. The specific action's result is then the actual fields you can access on the message containing more information.

But if you're just looking for a status, its already there through the mechanics you implicitly reference in the above comment.

SteveMacenski, 2021-07-07:
Closing since already provided and nothing is missing here, so the feature request is `wontfix`.

closed 2021-07-07 by SteveMacenski