ros-navigation/navigation2 #2439: Error while starting navigation using local planner(DWB)

Soofiyan, 2021-07-07:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Foxy
- Version or commit hash:
  - Latest of foxy-devel branch(2nd July 2021)
- DDS implementation:
  

#### Steps to reproduce issue

```
This error is seen in simulation when the robot collides with obstacles(similar issue).

```

#### Expected behaviour
After giving a goal, the global planner is properly planning the path towards the goal. The expected behaviour is that the path must follow this path using a local planner.
[image]

#### Actual behavior
Here the path is always failing and starts to spin, And the bot is not hitting any obstacle, and transformations are perfect. The robot is not even moving a little bit; it directly starts the recoveries servers as soon as a global planner is planned.
[image]

#### Additional information
This is the error shown on the screen
[image]

SteveMacenski, 2021-07-07:
You can certainly tune the DWB controller into behavior where it has a hard time moving. Have you tuned the parameters of DWB from the defaults in nav2_bringup? (not that those are perfect by any stretch either, but its a baseline we can use to test that hypothesis) The error also looks to be pretty straight forward: https://github.com/ros-planning/navigation2/search?q=Trajectory+goes+off, is your costmap reasonably sized? Please provide your configuration of DWB / local costmap.

Soofiyan, 2021-07-07:
Thank you for replying to my query. I have tuned this for almost 5-6 hours. Sometimes it is going towards the goal but still failing in the middle of the trajectory. 9/10 times it is still going into the recovery mode without hitting any obstacle, as mentioned earlier. If you can suggest any change in a parameter, it will be beneficial.
Here is my nav2_param file, which I am using
```
amcl:
  ros__parameters:
    use_sim_time: True
    alpha1: 0.2
    alpha2: 0.2
    alpha3: 0.2
    alpha4: 0.2
    alpha5: 0.2
    base_frame_id: "base_footprint"
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.9
    beam_skip_threshold: 0.3
    do_beamskip: false
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 2.0
    laser_max_range: 100.0
    laser_min_range: -1.0
    laser_model_type: "likelihood_field"
    max_beams: 60
    max_particles: 2000
    min_particles: 500
    odom_frame_id: "odom"
    pf_err: 0.05
    pf_z: 0.99
    recovery_alpha_fast: 0.0
    recovery_alpha_slow: 0.0
    resample_interval: 1
    robot_model_type: "differential"
    save_pose_rate: 0.5
    sigma_hit: 0.2
    tf_broadcast: true
    transform_tolerance: 1.0
    update_min_a: 0.2
    update_min_d: 0.25
    z_hit: 0.5
    z_max: 0.05
    z_rand: 0.5
    z_short: 0.05
    scan_topic: scan1

amcl_map_client:
  ros__parameters:
    use_sim_time: True

amcl_rclcpp_node:
  ros__parameters:
    use_sim_time: True

bt_navigator:
  ros__parameters:
    use_sim_time: False
    global_frame: map
    robot_base_frame: base_link
    odom_topic: /odom
    enable_groot_monitoring: True
    groot_zmq_publisher_port: 1666
    groot_zmq_server_port: 1667
    default_bt_xml_filename: "default_bt_xml_filename"
    plugin_lib_names:
    - nav2_compute_path_to_pose_action_bt_node
    - nav2_follow_path_action_bt_node
    - nav2_back_up_action_bt_node
    - nav2_spin_action_bt_node
    - nav2_wait_action_bt_node
    - nav2_clear_costmap_service_bt_node
    - nav2_is_stuck_condition_bt_node
[... 200 lines cut ...]
    spin:
      plugin: "nav2_recoveries/Spin"
    backup:
      plugin: "nav2_recoveries/BackUp"
    wait:
      plugin: "nav2_recoveries/Wait"
    global_frame: odom
    robot_base_frame: base_link
    transform_timeout: 0.1
    use_sim_time: False
    simulate_ahead_time: 2.0
    max_rotational_vel: 1.0
    min_rotational_vel: 0.4
    rotational_acc_lim: 3.2

waypoint_follower:
  ros__parameters:
    loop_rate: 20
    stop_on_failure: false
    waypoint_task_executor_plugin: "wait_at_waypoint"   
    wait_at_waypoint:
      plugin: "nav2_waypoint_follower::WaitAtWaypoint"
      enabled: True
      waypoint_pause_duration: 200
```

SteveMacenski, 2021-07-07:
You didn't answer my questions, please answer them before I can further help. Additionally, it doesn't sound like you followed my advice, have you tested with complete defaults as set up in the nav2_bringup package for the costmaps and controllers? It's not worth any time to look at anything until that question is properly answered.

Soofiyan, 2021-07-08:
Yes, firstly, I tuned the parameters of the param file provided in the default nav2_bringup, but it was, again and again, going in the recovery mode. Also, I had added the footprint in the default param file instead of radius. So the default parameters were not giving me any progress whatsoever; thus, I tuned the parameters. Also, the costmap is reasonably sized for my use case.

SteveMacenski, 2021-07-08:
Please ask on ROS Answers -- I can't provide personalized tuning assistance to every user. If you identify a tangible bug or issue, I'm happy to discuss a resolution to that.

closed 2021-07-08 by SteveMacenski