ros-navigation/navigation2 #2442: [Smac Smoother] handle reverse direction reeds sheep models as individual segments

SteveMacenski, 2021-07-08:
Right now the smoother will smooth the entire path, meaning that it will smooth the cusp in direction changes. That probably isn't good. Instead we should be treating each segment in a velocity direction separately to retain the cusp and directional changes.

dpm-seasony, 2021-11-08:
Could you confirm that the PR  #2660 solves this issue?

salmagro, 2021-11-18:
I have experienced a similar effect, and I would like to clarify if it corresponds to the same issue treated here.

The function `updateApproximatePathOrientations` recalculate the heading based on the order of the XY coordinates without considering the driving direction.

This generates a continuous path to drive forward, provoking the misbehavior of the local planner.

[image]
[image]

SteveMacenski, 2021-11-18:
@salmagro try testing the PR #2660. It should work, I've asked for more programming style updates. Let us know if that fixes your issue

salmagro, 2021-11-19:
@SteveMacenski thank you for the information. Yes, PR #2660 solved the problem with the wrong heading. However, I have experienced a bug described in https://github.com/ros-planning/navigation2/pull/2660#issuecomment-974087387 .

SteveMacenski, 2021-12-02:
It's my intention to address this tomorrow and get something merged in hopefully before the end of the week. The issue with the smoother making the plans potentially no longer kinematically feasible remains, but at least reversing will not be negatively impacted anymore (e.g. I can stop commenting out the smoother when I do tests in reverse).

 I'll plan on addressing, or asking someone else to address, the issue with feasibility as a separate topic covered in #2690. I want to start working on these problems as individuals so we can make incremental progress.

SteveMacenski, 2021-12-03:
I have a working solution to this, but discovered some quirks with the State Lattice planner that need to be resolved before I can merge it. We need the primitive markers to be some distance apart so I can use the dot product to discover cusps. Right now, it over-fires on some situations that I'm trying to fully analyze so that I can fix them all.

For now since I'm in a bit of a waiting cycle, I'm also going to try and tackle the smoother feasibility challenge, which might also encompass re-adding the parameter for smoothing or not so that it can be disabled.

SteveMacenski, 2021-12-03:
https://github.com/ros-planning/navigation2/pull/2727

SteveMacenski, 2021-12-06:
Merging shortly

closed 2021-12-06 by SteveMacenski