ros-navigation/navigation2 #2452: Nav2 plans path through walls after clear global costmap

charlielito, 2021-07-15:
## Bug report

- Operating System:
  - Docker + Ubuntu 20.04, 10th gen i7
- ROS2 Version:
  - Foxy binaries
- Version or commit hash:
  - Foxy
- DDS implementation:
  - cyclone 

#### Steps to reproduce issue
I am facing this same issue https://github.com/ros-planning/navigation2/issues/1999 where I see it was fixed with https://github.com/ros-planning/navigation2/pull/2090
Nonetheless I think this was not ported to foxy branch. 

#### Expected behavior
Do plan through walls.

#### Actual behavior
Sometimes it plans through walls.

Would it be possible to introduce it to foxy? Do you think the fix here https://github.com/ros-planning/navigation2/pull/2090 will be compatible with foxy so a PR could be made?

SteveMacenski, 2021-07-15:
You pointed towards the same things I was going to link you to, the discussion there is pretty clear for Foxy this is a wont fix. The patch is not ABI compatible with Foxy and you should move to Galactic or newer if you wish to use the patch. Else, you can attempt to migrate it yourself into a local fork but we cannot merge that solution into Foxy because it significantly breaks ABI policies. If you have a solution that wouldn't do that, I'm all ears and happy to merge.

charlielito, 2021-07-15:
Ok, since this is a wont fix for foxy I'll close this. Thanks for the fast response

closed 2021-07-15 by charlielito