ros-navigation/navigation2 #2453: [Recovery] Backup never terminating when robot is stuck

karl-schulz, 2021-07-16:
## Bug report

**Required Info:**

- Operating System:
  - Any
- ROS2 Version:
  - Galactic
- Version or commit hash:
  - master or 5a70275 
- DDS implementation:
  - Any

#### Steps to reproduce issue

Run any robot simulation using a backup recovery. Example behavior tree XML:

````
<root main_tree_to_execute="MainTree">
  <BehaviorTree ID="MainTree">
    <RecoveryNode number_of_retries="3" name="NavigateRecovery">
      <PipelineSequence name="SimpleBT">
        <RateController hz="4.0">
          <ComputePathToPose goal="{goal}" path="{path}" planner_id="GridBased"/>
        </RateController>
        <FollowPath path="{path}" controller_id="FollowPath"/>
      </PipelineSequence>
      <ReactiveFallback name="RecoveryFallback">
        <GoalUpdated/> 
        <RoundRobin name="RecoveryActions">
          <BackUp backup_dist="0.65" backup_speed="0.15"/>
        </RoundRobin>
      </ReactiveFallback>
    </RecoveryNode>
  </BehaviorTree>
</root>
````

Now, block the robot from behind to it cannot move backwards. The backup movement will be stuck at:

    [controller_server-3] [ERROR] [1626426808.548096680] [controller_server]: Failed to make progress
    [controller_server-3] [WARN] [1626426808.548129815] [controller_server_rclcpp_node]: [follow_path] [ActionServer] Aborting handle.
    [recoveries_server-4] [INFO] [1626426808.570189476] [recoveries_server]: Attempting backup

And navigation never terminates and keeps publishing`cmd_vel` = `... linear: x: -0.15  ...`.

In the real world, this can be triggered by multiple things:
* Failure in localization (and the robot will move backwards forever, not good)
* Being blocked by a non-visible object
* Motor failure, robot being broken and not moving despite non-negative `cmd_vel`

#### Expected behavior

The recovery fails and the navigator returns an error.

#### Actual behavior

The navigator is stuck in the backup node on RUNNING.

#### Proposed Solution

Provide an (optional) argument `timeout` and check if it expired in the `BackUp::onCycleUpdate`. If the timeout expired, return FAILURE.

If you want, I could create a PR.

SteveMacenski, 2021-07-16:
That sounds reasonable, I'd merge that.

SteveMacenski, 2021-07-27:
@karl-schulz is there a PR coming for this?

karl-schulz, 2021-07-27:
Yes definitely, we are just a bit too busy with other issues :) It's coming in the next days.

SteveMacenski, 2021-08-24:
@karl-schulz I see the PR you linked to this ticket from @philison. I would approve that PR (+ a couple of parameter renames and doc updates) to the main stack if that PR was directed here.

philison, 2021-08-24:
Great. Yes, it was meant to solve this issue. I just haven't prepared the pull request for the merge with the nav2 repo, like you said. I also think there might be a simpler solution that adopts/copies the "waiting" functionality from the `Wait` node. I will check this tmr CET and will open a PR.

SteveMacenski, 2021-10-12:
Merging shortly