ros-navigation/navigation2 #2455: Build error (initial_ros_setup.sh)

kledom, 2021-07-16:
## Bug report

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - latest (according to https://github.com/ros2/ros2/commit/3d98527)
- Version or commit hash:
  - main @ 5a7027536068183602f10b419bd711fafdad6784

#### Steps to reproduce issue

```console
mkdir nav2
cd nav2
wget https://raw.githubusercontent.com/ros-planning/navigation2/main/tools/initial_ros_setup.sh
chmod a+x initial_ros_setup.sh
./initial_ros_setup.sh
```

#### Expected behavior

Successful build.

#### Actual behavior

Build fails with the following error message:

```
--- stderr: nav2_util                                
CMake Error at CMakeLists.txt:18 (find_package):
  By not providing "Findbondcpp.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "bondcpp", but
  CMake did not find one.

  Could not find a package configuration file provided by "bondcpp" with any
  of the following names:

    bondcppConfig.cmake
    bondcpp-config.cmake

  Add the installation prefix of "bondcpp" to CMAKE_PREFIX_PATH or set
  "bondcpp_DIR" to a directory containing one of the above files.  If
  "bondcpp" provides a separate development package or SDK, be sure it has
  been installed.
```

#### Additional information

Probably related to https://github.com/ros-planning/navigation2/commit/083f3214c5f62010aa4a83496263561459ceef65 where all entries from tools/underlay.repos were commented out.

SteveMacenski, 2021-07-16:
Certainly related, yes. Sorry that script hasn't been well maintained, I want to apologize for that. This has slipped my mind. 

If you are building on an existing distribution, https://github.com/ros-planning/navigation2/blob/main/tools/initial_ros_setup.sh#L92 this will grab the dependencies because they've been released as binaries.

While this effort is still ongoing https://github.com/ros-planning/navigation2/issues/2337 we are planning on moving to Rolling as our basis of development, CI, and testing. So in that context, all of those dependencies are also released. So I would recommend working on Rolling and not ROS2 master branches and then this should work (if you find any other issues, I'll merge in a PR to resolve). Else, you can uncomment out that underlay file and build that as well and that should solve your issue in the meantime.

kledom, 2021-07-18:
Thank you very much for the prompt reply. Compiling succeeded after I switched to ros2-rolling and installed the dependencies using  rosdep.

closed 2021-07-18 by kledom