ros-navigation/navigation2 #2463: [Smac Planner] Planning fails without analytic expansion

HosseinSheikhi, 2021-07-20:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu 20.04
- ROS2 Version:
  - Rolling
- Version or commit hash:
  - 21a44bcb559682f67289ac72571206597c00855b

- DDS implementation:
  - 

#### Steps to reproduce issue
Just commenting the analytic expansion in the A* algorithm [lines 266-271](https://github.com/ros-planning/navigation2/blob/21a44bcb559682f67289ac72571206597c00855b/nav2_smac_planner/src/a_star.cpp#L266). 

#### Expected behavior
Find a path.
#### Actual behavior
`[planner_server-2] [WARN] [1626764628.205177091] [nav2_costmap_2d]: Robot is out of bounds of the costmap!
[planner_server-2] [WARN] [1626764628.205215696] [global_costmap.global_costmap]: Sensor origin at (-2.06, 4.00) is out of map bounds. The costmap cannot raytrace for it.
[planner_server-2] [WARN] [1626764629.205208316] [nav2_costmap_2d]: Robot is out of bounds of the costmap!
[planner_server-2] [WARN] [1626764629.205239394] [global_costmap.global_costmap]: Sensor origin at (-2.06, 4.00) is out of map bounds. The costmap cannot raytrace for it.
[planner_server-2] [WARN] [1626764630.205251944] [nav2_costmap_2d]: Robot is out of bounds of the costmap!
[planner_server-2] [WARN] [1626764630.205301593] [global_costmap.global_costmap]: Sensor origin at (-2.06, 4.00) is out of map bounds. The costmap cannot raytrace for it.
[planner_server-2] [WARN] [1626764631.205221296] [nav2_costmap_2d]: Robot is out of bounds of the costmap!
[planner_server-2] [WARN] [1626764631.205256366] [global_costmap.global_costmap]: Sensor origin at (-2.06, 4.00) is out of map bounds. The costmap cannot raytrace for it.
[planner_server-2] [INFO] [1626764631.822458317] [global_costmap.global_costmap]: StaticLayer: Resizing costmap to 101 X 83 at 0.050000 m/pix
[bt_navigator-4] [INFO] [1626764650.554547341] [bt_navigator]: Begin navigating from current location to (-1.22, 3.36)

[controller_server-1] [INFO] [1626764650.575388070] [controller_server]: Received a goal, begin computing control effort.
[controller_server-1] [WARN] [1626764650.575434888] [controller_server]: No goal checker was specified in parameter 'current_goal_checker'. Server will use only plugin loaded general_goal_checker . This warning will appear once.
[planner_server-2] [WARN] [1626764651.598629032] [planner_server]: GridBased: failed to create plan, no valid path found.
[planner_server-2] [WARN] [1626764651.598699293] [planner_server]: Planning algorithm GridBased failed to generate a valid path to (-1.22, 3.36)
[planner_server-2] [WARN] [1626764651.598723302] [planner_server_rclcpp_node]: [compute_path_to_pose] [ActionServer] Aborting handle.
[planner_server-2] [INFO] [1626764651.615494237] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
[planner_server-2] [WARN] [1626764652.220077266] [planner_server]: GridBased: failed to create plan, no valid path found.
[planner_server-2] [WARN] [1626764652.220136763] [planner_server]: Planning algorithm GridBased failed to generate a valid path to (-1.22, 3.36)
[planner_server-2] [WARN] [1626764652.220154502] [planner_server_rclcpp_node]: [compute_path_to_pose] [ActionServer] Aborting handle.
[controller_server-1] [INFO] [1626764652.235371881] [local_costmap.local_costmap]: Received request to clear entirely the local_costmap
[planner_server-2] [INFO] [1626764652.235859397] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
[controller_server-1] [INFO] [1626764652.275766203] [controller_server]: Goal was canceled. Stopping the robot.
[controller_server-1] [WARN] [1626764652.275941525] [controller_server_rclcpp_node]: [follow_path] [ActionServer] Client requested to cancel the goal. Cancelling.
[planner_server-2] [WARN] [1626764653.218986119] [planner_server]: GridBased: failed to create plan, no valid path found.
[planner_server-2] [WARN] [1626764653.219051948] [planner_server]: Planning algorithm GridBased failed to generate a valid path to (-1.22, 3.36)
[planner_server-2] [WARN] [1626764653.219070525] [planner_server_rclcpp_node]: [compute_path_to_pose] [ActionServer] Aborting handle.
[planner_server-2] [INFO] [1626764653.234858641] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
[planner_server-2] [WARN] [1626764654.216365642] [planner_server]: GridBased: failed to create plan, no valid path found.
[planner_server-2] [WARN] [1626764654.216417258] [planner_server]: Planning algorithm GridBased failed to generate a valid path to (-1.22, 3.36)
[planner_server-2] [WARN] [1626764654.216434684] [planner_server_rclcpp_node]: [compute_path_to_pose] [ActionServer] Aborting handle.
[recoveries_server-3] [INFO] [1626764654.235644048] [recoveries_server]: Attempting spin
[recoveries_server-3] [INFO] [1626764654.235849994] [recoveries_server]: Turning 1.57 for spin recovery.
[recoveries_server-3] [INFO] [1626764656.036079620] [recoveries_server]: spin completed successfully
[planner_server-2] [WARN] [1626764656.089889252] [planner_server]: GridBased: failed to create plan, no valid path found.
[planner_server-2] [WARN] [1626764656.089975756] [planner_server]: Planning algorithm GridBased failed to generate a valid path to (-1.22, 3.36)
[planner_server-2] [WARN] [1626764656.090003404] [planner_server_rclcpp_node]: [compute_path_to_pose] [ActionServer] Aborting handle.
[planner_server-2] [INFO] [1626764656.104810242] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
[planner_server-2] [WARN] [1626764656.235658419] [planner_server]: GridBased: failed to create plan, no valid path found.
[planner_server-2] [WARN] [1626764656.235744506] [planner_server]: Planning algorithm GridBased failed to generate a valid path to (-1.22, 3.36)
[planner_server-2] [WARN] [1626764656.235776133] [planner_server_rclcpp_node]: [compute_path_to_pose] [ActionServer] Aborting handle.
[recoveries_server-3] [INFO] [1626764656.255445034] [recoveries_server]: Attempting wait
[planner_server-2] [INFO] [1626764656.822819052] [global_costmap.global_costmap]: StaticLayer: Resizing costmap to 101 X 86 at 0.050000 m/pix
[recoveries_server-3] [INFO] [1626764661.255815799] [recoveries_server]: wait completed successfully
[planner_server-2] [WARN] [1626764661.306500428] [planner_server]: GridBased: failed to create plan, no valid path found.
[planner_server-2] [WARN] [1626764661.306557835] [planner_server]: Planning algorithm GridBased failed to generate a valid path to (-1.22, 3.36)
[planner_server-2] [WARN] [1626764661.306575698] [planner_server_rclcpp_node]: [compute_path_to_pose] [ActionServer] Aborting handle.
[planner_server-2] [INFO] [1626764661.324856438] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
[planner_server-2] [WARN] [1626764662.217128102] [planner_server]: GridBased: failed to create plan, no valid path found.
[planner_server-2] [WARN] [1626764662.217182337] [planner_server]: Planning algorithm GridBased failed to generate a valid path to (-1.22, 3.36)
[... 9 lines cut ...]
[planner_server-2] [WARN] [1626764669.235991877] [planner_server_rclcpp_node]: [compute_path_to_pose] [ActionServer] Aborting handle.
[controller_server-1] [INFO] [1626764669.255115689] [local_costmap.local_costmap]: Received request to clear entirely the local_costmap
[planner_server-2] [INFO] [1626764669.255892776] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
[planner_server-2] [WARN] [1626764670.226795367] [planner_server]: GridBased: failed to create plan, no valid path found.
[planner_server-2] [WARN] [1626764670.226881004] [planner_server]: Planning algorithm GridBased failed to generate a valid path to (-1.22, 3.36)
[planner_server-2] [WARN] [1626764670.226909750] [planner_server_rclcpp_node]: [compute_path_to_pose] [ActionServer] Aborting handle.
[planner_server-2] [INFO] [1626764670.245020106] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
[planner_server-2] [WARN] [1626764671.250231090] [planner_server]: GridBased: failed to create plan, no valid path found.
[planner_server-2] [WARN] [1626764671.250306837] [planner_server]: Planning algorithm GridBased failed to generate a valid path to (-1.22, 3.36)
[planner_server-2] [WARN] [1626764671.250334382] [planner_server_rclcpp_node]: [compute_path_to_pose] [ActionServer] Aborting handle.
[recoveries_server-3] [INFO] [1626764671.265304320] [recoveries_server]: Attempting spin
[... cut ...]
[planner_server-2] [WARN] [1626764673.216301218] [planner_server]: Planning algorithm GridBased failed to generate a valid path to (-1.22, 3.36)
[planner_server-2] [WARN] [1626764673.216317966] [planner_server_rclcpp_node]: [compute_path_to_pose] [ActionServer] Aborting handle.
[bt_navigator-4] [ERROR] [1626764673.244682558] [bt_navigator]: Goal failed
[bt_navigator-4] [WARN] [1626764673.244736501] [bt_navigator]: [navigate_to_pose] [ActionServer] Aborting handle.
[planner_server-2] [INFO] [1626764676.823228700] [global_costmap.global_costmap]: StaticLayer: Resizing costmap to 101 X 87 at 0.050000 m/pix
`
#### Additional information

HosseinSheikhi, 2021-07-20:
One main reason is without analytic expansion robot can't EXACTLY reach the goal, so I have to enable the tolerance feature. But still does not work well, some times find the path -usually finds paths to the goals close to the robot- sometimes not.

SteveMacenski, 2021-07-20:
```
`[planner_server-2] [WARN] [1626764628.205177091] [nav2_costmap_2d]: Robot is out of bounds of the costmap!
[planner_server-2] [WARN] [1626764628.205215696] [global_costmap.global_costmap]: Sensor origin at (-2.06, 4.00) is out of map bounds. The costmap cannot raytrace for it.
[planner_server-2] [WARN] [1626764629.205208316] [nav2_costmap_2d]: Robot is out of bounds of the costmap!
[planner_server-2] [WARN] [1626764629.205239394] [global_costmap.global_costmap]: Sensor origin at (-2.06, 4.00) is out of map bounds. The costmap cannot raytrace for it.
[planner_server-2] [WARN] [1626764630.205251944] [nav2_costmap_2d]: Robot is out of bounds of the costmap!
[planner_server-2] [WARN] [1626764630.205301593] [global_costmap.global_costmap]: Sensor origin at (-2.06, 4.00) is out of map bounds. The costmap cannot raytrace for it.
[planner_server-2] [WARN] [1626764631.205221296] [nav2_costmap_2d]: Robot is out of bounds of the costmap!
[planner_server-2] [WARN] [1626764631.205256366] [global_costmap.global_costmap]: Sensor origin at (-2.06, 4.00) is out of map bounds. The costmap cannot raytrace for it.
```

This has nothing to do with the planner, this has to do with your costmap configuration so you should look into that first. 

Surely, if you just choose to comment out large chunks of my code, you're going to break the code. I don't know what you should expect. This doesn't seem like an issue to me. This seems like it should be closed as invalid.

HosseinSheikhi, 2021-07-20:
> ```
> `[planner_server-2] [WARN] [1626764628.205177091] [nav2_costmap_2d]: Robot is out of bounds of the costmap!
> [planner_server-2] [WARN] [1626764628.205215696] [global_costmap.global_costmap]: [...]

Absolutely, but it is not the issue, It always appears at the beginning, before costmap resizes itself. 

> Surely, if you just choose to comment out large chunks of my code, you're going to break the code. I don't know what you should expect. This doesn't seem like an issue to me. Of course this was added to the planner because it drastically speeds things up. This seems like it should be closed as invalid.
To me, it seems fair to expect the planner to find a path to the vicinity of the goal without analytic expansion.
P.S. am working on it and will close it within the next few hours.

SteveMacenski, 2021-07-20:
There is no support or intention of support of "finding a path in the vicinity" for kinematically feasible path generators. What you see inline there is for the 2D A* but is broadly unused for the Hybrid-A* implementation. I don't even think I expose the dials for that into Hybrid-A*'s plugin. 

If there's some improvements you've made, I'd be open to hear them / look over a PR with them, but generally it is the intent of Hybrid-A* to always be paired with the analytic expansion phase of planning. It is core to the Hybrid-A* planner academic description and as you've found, it is important for run-time efficiency since most planners will spend the bulk of their time on the final approach for exact fitting. This eliminates that largely.

HosseinSheikhi, 2021-07-22:
By disabling analytic expansion one has to set up the `_tolerance`  and `approach_iterations` SmacPlannerHybrid finds a path to a vicinity. I set them to 0.25 and 100 respectively.

closed 2021-07-22 by HosseinSheikhi