ros-navigation/navigation2 #2467: Set BT::NodeStatus based on service result msg

philison, 2021-07-21:
**Required Info:**

- Operating System:
  - Ubuntu 20.04.2 LTS
- ROS2 Version:
  - Galactic binaries
- Version or commit hash:
  - 5a7027536068183602f10b419bd711fafdad6784
- DDS implementation:
  - 

## Feature request

#### Feature description

It is the goal to set the BT::NodeStatus of a behavior tree leave node based on the result send from the service server.
More specifically, I tried to implement a _Condition-Node_ that makes a request to a service server via the ROS2 service interface and sets its BT::NodeStatus based on the result.. Therefore, the BtServiceNode class ([_bt\_service\_node.cpp_](https://github.com/angsa-robotics/navigation2/blob/033cdc49b4c106f2747772f2081fca47bee6e9ec/nav2_behavior_tree/include/nav2_behavior_tree/bt_service_node.hpp)) seemed to me to be the best suited class to use. However, in contrast to the BtActionNode class ([_bt\_action\_node.cpp_](https://github.com/ros-planning/navigation2/blob/0e470415f0d6b2672e350e2cc9271002b143cb10/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_node.hpp)) it missed the virtual _on\_success()_ function, that can be used to define the custom BT::NodeStatus behavior based on the service's answer. The current BtServiceNode class sets the BT::NodeStatus solely based on the success of the service request.

I am not sure if I missed a class that implements this feature, if there is a simpler solution or if I should just use the BT::ConditionNode base class with a normal service client after all. 

#### Implementation considerations

The proposed implementation simply adds a call to the virtual _on\_success()_ function in the _check\_future()_ function (Similar to the BtActionNode).

The changes do not change do not affect the default behavior of the BtServiceNode as the default definition of _on\_success()_ does still simply return BT::NodeStatus::SUCCESS.

Changes:
https://github.com/angsa-robotics/navigation2/blob/033cdc49b4c106f2747772f2081fca47bee6e9ec/nav2_behavior_tree/include/nav2_behavior_tree/bt_service_node.hpp#L150-L187

Original:
https://github.com/ros-planning/navigation2/blob/0764e5af5b23ea7a08629876792a8da1465ef261/nav2_behavior_tree/include/nav2_behavior_tree/bt_service_node.hpp#L145-L171

SteveMacenski, 2021-07-21:
>  The current BtServiceNode class sets the BT::NodeStatus solely based on the success of the service request.

Yes, but its virtual, if you made your own custom BTServiceNode implementation using it, you can override it with whatever you like.

SteveMacenski, 2021-07-21:
Please show a diff, I can't know what you changed just looking at that easily.

philison, 2021-07-21:
Sorry for the confusion. I thought I had double checked, so exactly that wouldn't happen, but it is the wrong link.
Yes exactly, I was using the BtActionNode as an inspiration. There, a differentiation is made between all of the possible request results of the action interface (each with the corresponding callback).
https://github.com/ros-planning/navigation2/blob/c417e2fd267e1dfa880b7ff9d37aaaa7b5eab9ca/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_node.hpp#L239-L255
However as cancelling and aborting is not possible with services those would not be relevant. 
A on-function that might be relevant would be the `on_wait_for_result()`, as implemented in the BtActionNode.
https://github.com/ros-planning/navigation2/blob/c417e2fd267e1dfa880b7ff9d37aaaa7b5eab9ca/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_node.hpp#L133-L139

Regarding the wording, are you alluding to the to the two different types of success ? 

1. the success that refers to whether the ros2 interface interaction/communication was successful (more of a `complete` ?)
2. the success that refers to whether the actual task, that was triggered by calling the service, was successful

I agree with you about the wording issue, as the `on_success()` name refers to the rclcpp::FutureReturnCode::SUCCESS of the service call (case 1.), but does not necessarily mean that the BtNodeStatus is also a success.

SteveMacenski, 2021-07-21:
I think if you open a Pr, this would be much easier so we're talking over specific code vs abstract ideas

philison, 2021-07-27:
I finally got to open a Pr  #2481. I had to do some commit/feature de-cluttering, therefore it took a while.

SteveMacenski, 2021-08-03:
Merging imminent

closed 2021-08-03 by SteveMacenski