ros-navigation/navigation2 #2468: Nav2 B

marcomasa, 2021-07-21:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04.2 LTS, AMD Ryzen 3600
- ROS2 Version:
  - Foxy Debian
- Version or commit hash:
ros-foxy-nav2-amcl	0.4.7-1focal.20210513.213303
ros-foxy-nav2-behavior-tree	0.4.7-1focal.20210513.213233
ros-foxy-nav2-bringup	0.4.7-1focal.20210619.024606
ros-foxy-nav2-bt-navigator	0.4.7-1focal.20210513.214421
ros-foxy-nav2-common	0.4.7-1focal.20210513.203711
ros-foxy-nav2-controller	0.4.7-1focal.20210513.214739
ros-foxy-nav2-core	0.4.7-1focal.20210513.214550
ros-foxy-nav2-costmap-2d	0.4.7-1focal.20210513.213728
ros-foxy-nav2-dwb-controller	0.4.7-1focal.20210513.215352
ros-foxy-nav2-lifecycle-manager	0.4.7-1focal.20210513.213306
ros-foxy-nav2-map-server	0.4.7-1focal.20210513.213306
ros-foxy-nav2-msgs	0.4.7-1focal.20210513.205442
ros-foxy-nav2-navfn-planner	0.4.7-1focal.20210513.214739
ros-foxy-nav2-planner	0.4.7-1focal.20210513.214742
ros-foxy-nav2-recoveries	0.4.7-1focal.20210513.214741
ros-foxy-nav2-regulated-pure-pursuit-controller	0.4.7-1focal.20210513.214745
ros-foxy-nav2-rviz-plugins	0.4.7-1focal.20210619.022344
ros-foxy-nav2-util	0.4.7-1focal.20210513.212614
ros-foxy-nav2-voxel-grid	0.4.7-1focal.20210513.210024
ros-foxy-nav2-waypoint-follower	0.4.7-1focal.20210513.213311

- DDS implementation:
  - Default (Fast-RTPS)

#### Steps to reproduce issue
https://navigation.ros.org/getting_started/index.html

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

SteveMacenski, 2021-07-21:
> libnav2_navigate_through_poses_action_bt_node.so

Where did you get that from? That isn't available in Foxy. It shouldn't be anywhere in the Foxy branch.

marcomasa, 2021-07-21:
Tbh i dont know. I think i never had it, but it is now requested by the bt_navigator package, as shown above.

SteveMacenski, 2021-07-21:
Please try to find where that is specified. Your options are: behavior tree, configuration file, or bt_navigator.cpp defaults. Its not in the defaults https://github.com/ros-planning/navigation2/blob/foxy-devel/nav2_bt_navigator/src/bt_navigator.cpp#L40-L67. So that means either you're trying to use a behavior tree containing that node or your configuration file is calling out to pick that library up that doesn't exist.

marcomasa, 2021-07-21:
You are right, it should not be included in the launchfile.
I cloned the main branch to copy the existing nav2_bringup files and modify them:
https://github.com/ros-planning/navigation2/blob/main/nav2_bringup/bringup/params/nav2_params.yaml

Apparently they are not on foxy-devel state but rather on galactic?
Idk which one should be on the master in your opinion, 
maybe update this to foxy? Its kinda unintuitive to have the previous ones as master.

Anyways, adjusting this params in the launchfile did the job. Thank you for your help!

closed 2021-07-21 by marcomasa