ros-navigation/navigation2 #2487: Will the stability updates in Galactic be brought to Foxy as well?

jacobwaller, 2021-08-02:
I notice that there are quite a few changes for this package from Foxy to Galactic, including some stability updates.

Are any of these changes on the roadmap for Foxy? Specifically the stability with bt_navigator.

SteveMacenski, 2021-08-02:
You'll have to be more specific.

jacobwaller, 2021-08-02:
[this](https://github.com/ros-planning/navigation2/issues/2272) to be more specific. Was wondering if this was on the roadmap, because I am running in to the same issues.

SteveMacenski, 2021-08-03:
Well, you can see my exact answer to your question in your linked ticket https://github.com/ros-planning/navigation2/issues/2272

jacobwaller, 2021-08-03:
Are the larger timeouts you mentioned in your recent comment on ros2/ros2#1074 currently implemented in the foxy binaries? Or is that what went out with galactic?

jacobwaller, 2021-08-03:
Also, maybe slightly unrelated, is there a workaround for this? Instead of using ActionSrvs to pass a goal pose, are there other options? I'd really prefer to not change my whole stack to fix this for my use case haha

SteveMacenski, 2021-08-03:
Please read #2272, all I'm going to do is copy paste my exact comments from there so you should just read it there to see the full context.

closed 2021-08-03 by jacobwaller