ros-navigation/navigation2 #2491: bt_navigator gets stuck in on_activate()

jacobwaller, 2021-08-07:
## Bug report

- Operating System:
Ubuntu 20.04
- ROS2 Version:
Galactic Apt Binaries 
- Version or commit hash:
ros-galactic-navigation2        1.0.6-1focal.20210701.071948

#### Steps to reproduce issue
[relevant section of config on pastebin](https://pastebin.com/erEAHxCQ)
Launched with
```
    <include file="/home/ubuntu/beer-bot/bringup/launch/nav2_launch.py">
        <arg name="map" value="/home/ubuntu/beer-bot/apt.yaml" />
        <arg name="params_file" value="/home/ubuntu/beer-bot/bringup/config/nav_config.yaml" />
    </include>
```
The nav2 launch file found at `home/ubuntu/beer-bot/bringup/launch/nav2_launch.py`:
[here](https://pastebin.com/LUWEnVgA) (It is a barely modified copy of nav_bringup.py)

#### Expected behavior
nav2 stack should launch

#### Actual behavior
nav2 stack launches with the exception of bt_navigator.

nothing happens after
```
[lifecycle_manager-13] [INFO] [1628302332.755854970] [lifecycle_manager_navigation]: Activating bt_navigator
[bt_navigator-11] [INFO] [1628302332.756855404] [bt_navigator]: Activating
```

#### Additional information
I'm expecting the launch process to also print something along the lines of:
```
[bt_navigator-11] [INFO] [1628302332.639718606] [bt_navigator]: Creating bond (bt_navigator) to lifecycle manager.
[lifecycle_manager-13] [INFO] [1628302332.755689046] [lifecycle_manager_navigation]: Server bt_navigator connected with bond.
```

jacobwaller, 2021-08-07:
I have been driving myself mad for the past few days trying to figure this out. I'm not sure if this is a bug or an issue with my configuration. But as far as I can see, I should be launching everything properly and my config seems correct

mdtoom, 2021-08-09:
It might be waiting for a service or action that is not available. Are you sure that all the action/service servers that the behavior tree nodes use are up and functional?

jacobwaller, 2021-08-09:
@mdtoom Is lifecycle manager order dependent? I'm just wondering if nav_bringup.py is incorrect as of right now. Or if navigation_launch.py in nav2_bringup is incorrect..

SteveMacenski, 2021-08-09:
Does this happen when using the nav2_bringup example launch files without changes? It works for me and other production users. If that works for you, then it would be good for you to change one thing at a time until you back out where the issue is. If that doesn't work for you, then there may be something deeper afoot to talk about.

jacobwaller, 2021-08-09:
Hey @SteveMacenski its looking like even when using the example launch file, i'm getting the same issue. 
```
    <include file="$(find-pkg-share nav2_bringup)/launch/bringup_launch.py">
        <arg name="map" value="/home/ubuntu/beer-bot/apt.yaml" />
        <arg name="params_file" value="/home/ubuntu/beer-bot/bringup/config/nav_config.yaml" />
    </include>
```

jacobwaller, 2021-08-09:
& I wanted to share some more detail that mdtoom had mentioned.
(Both outputs are from while bt_navigator is stuck in on_activate())

the output of `ros2 action list`:
```
/back_up
/backup
/compute_path_through_poses
/compute_path_to_pose
/follow_path
/follow_waypoints
/navigate_through_poses
/navigate_to_pose
/spin
/wait
```

and of `ros2 service list`

```
/amcl/change_state
/amcl/describe_parameters
/amcl/get_available_states
/amcl/get_available_transitions
/amcl/get_parameter_types
/amcl/get_parameters
/amcl/get_state
/amcl/get_transition_graph
/amcl/list_parameters
/amcl/set_parameters
/amcl/set_parameters_atomically
/amcl_rclcpp_node/describe_parameters
/amcl_rclcpp_node/get_parameter_types
/amcl_rclcpp_node/get_parameters
/amcl_rclcpp_node/list_parameters
/amcl_rclcpp_node/set_parameters
/amcl_rclcpp_node/set_parameters_atomically
/bt_navigator/change_state
/bt_navigator/describe_parameters
/bt_navigator/get_available_states
/bt_navigator/get_available_transitions
/bt_navigator/get_parameter_types
/bt_navigator/get_parameters
/bt_navigator/get_state
/bt_navigator/get_transition_graph
/bt_navigator/list_parameters
/bt_navigator/set_parameters
/bt_navigator/set_parameters_atomically
/bt_navigator_rclcpp_node/describe_parameters
/bt_navigator_rclcpp_node/get_parameter_types
/bt_navigator_rclcpp_node/get_parameters
/bt_navigator_rclcpp_node/list_parameters
/bt_navigator_rclcpp_node/set_parameters
/bt_navigator_rclcpp_node/set_parameters_atomically
/controller_server/change_state
/controller_server/describe_parameters
/controller_server/get_available_states
/controller_server/get_available_transitions
/controller_server/get_parameter_types
/controller_server/get_parameters
/controller_server/get_state
/controller_server/get_transition_graph
/controller_server/list_parameters
/controller_server/set_parameters
/controller_server/set_parameters_atomically
/controller_server_rclcpp_node/describe_parameters
/controller_server_rclcpp_node/get_parameter_types
/controller_server_rclcpp_node/get_parameters
/controller_server_rclcpp_node/list_parameters
/controller_server_rclcpp_node/set_parameters
[... 151 lines cut ...]
/stop_motor
/tf2_ros_fp_laser/describe_parameters
/tf2_ros_fp_laser/get_parameter_types
/tf2_ros_fp_laser/get_parameters
/tf2_ros_fp_laser/list_parameters
/tf2_ros_fp_laser/set_parameters
/tf2_ros_fp_laser/set_parameters_atomically
/waypoint_follower/change_state
/waypoint_follower/describe_parameters
/waypoint_follower/get_available_states
/waypoint_follower/get_available_transitions
/waypoint_follower/get_parameter_types
/waypoint_follower/get_parameters
/waypoint_follower/get_state
/waypoint_follower/get_transition_graph
/waypoint_follower/list_parameters
/waypoint_follower/set_parameters
/waypoint_follower/set_parameters_atomically
/waypoint_follower_rclcpp_node/describe_parameters
/waypoint_follower_rclcpp_node/get_parameter_types
/waypoint_follower_rclcpp_node/get_parameters
/waypoint_follower_rclcpp_node/list_parameters
/waypoint_follower_rclcpp_node/set_parameters
/waypoint_follower_rclcpp_node/set_parameters_atomically
```

SteveMacenski, 2021-08-09:
You are clearly using something else by showing me that xml. Use the exact commands from the exact nav2-bringup package only, without anything else you've added or modifications.

jacobwaller, 2021-08-09:
I'm not sure what you mean by "using something else".

I'm using the launch file that comes with nav2_bringup. I'm calling it with the arguments that are required for my use case.

SteveMacenski, 2021-08-09:
https://navigation.ros.org/getting_started/index.html please follow the general getting started tutorials.

jacobwaller, 2021-08-09:
I have..

and now integrating this with actual hardware has its own set of challenges. Running in to issues I never saw with the foxy version.

SteveMacenski, 2021-08-09:
Then I'd point you back to 

> If that works, then it would be good for you to change one thing at a time until you back out where the issue is.

Please ask any follow up configuration / use questions on ROS Answers.

closed 2021-08-09 by SteveMacenski

eliasdc, 2021-09-20:
I had been having the same problem where the bt_navigator was stuck in onActivate(). I just found out what config was different and caused this to hang. 

My `recoveries_server` settings were as follows:
```
recoveries_server:
  ros__parameters:
    recovery_plugins: ["spin", "back_up", "wait"]
    spin:
      plugin: "nav2_recoveries/Spin"
    back_up:
      plugin: "nav2_recoveries/BackUp"
    wait:
      plugin: "nav2_recoveries/Wait"
```

Changing this to be the same as the nav2 main branch makes it (`back_up` -> `backup`): 
```
recoveries_server:
  ros__parameters:
    recovery_plugins: ["spin", "backup", "wait"]
    spin:
      plugin: "nav2_recoveries/Spin"
    backup:
      plugin: "nav2_recoveries/BackUp"
    wait:
      plugin: "nav2_recoveries/Wait"
```

Shouldn't this be just a configuration key? I'm not sure it should be blocking the `bt_navigator`. This does seem like a bug to me.

jacobwaller, 2021-09-22:
I agree. I think the bug is more that there is a lack of error messages, personally.

MoritzWillmann, 2022-01-12:
> @mdtoom Is lifecycle manager order dependent? I'm just wondering if nav_bringup.py is incorrect as of right now. Or if navigation_launch.py in nav2_bringup is incorrect..

hey @jacobwaller, I don't know if this is still relevant to you but I encountered the same "issue" using nav2 with galactic on tag 1.0.8 today. I think that lifecycle node order is important, yes. It seems that for `navigation_launch.py` (and especially `bt_navigator`) to work, `map_server` and `amcl` which are launched in `localization_launch.py` need to be added to the lifecycle nodes first. Making sure all nodes are added in the same order as they are when using `bringup_launch.py` solved the problem for me.

Kaetera, 2022-01-27:
Got the same error after migration from Foxy to Galactic.
Changing from `back_up` to `backup` as @eliasdc explained solved it for me.

Note: I have vague memory that I tried one day under Foxy to rename the `back_up` key in the .yaml param file and that it caused issues.

cf-zhang, 2022-03-10:
I also encountered this situation. It took me several hours to find the problem in my system. Make sure that all lifecycle nodes are activated by lifecycle manager. Check whether the nodes match the management list and spell it completely. Wasted hours dealing with a problem that shouldn't have happened.

Zephyr69, 2022-11-24:
> I had been having the same problem where the bt_navigator was stuck in onActivate(). I just found out what config was different and caused this to hang.
>
> My `recoveries_server` settings were as [...]

Amazing! Almost destroyed me when migrating from foxy onwards.