ros-navigation/navigation2 #2492: Single rviz for mult - robot navigation

Parv-Maheshwari, 2021-08-07:
## Feature request
Option for user to use a single rviz for mult -robot navigation

#### Feature description

It would be great to have a option for users to use a single rviz for mult -robot navigation as having individual rviz for each robot  as in the current multibot navigation setting, can act as a hindrance when increasing the number of robots in simulation

SteveMacenski, 2021-08-08:
PRs are appreciated if it is possible.

SteveMacenski, 2021-09-14:
@Parv-Maheshwari is this something you're interested in contributing? It's definitely a nice to have, but I'd probably close the ticket otherwise since this is a 'nice to have'. The rviz plugins for controlling the robot would only work with a single robot at a time so it would introduce a whole set of other command and control issues that probably aren't worth addressing unless there's real need for it.

SteveMacenski, 2021-10-07:
Closing due to lack of response. This is not a bad idea, but won't be worked on, happy to reopen if someone's wanting to work on it in the future. It would require having multiple robot's nav2 rviz panels in a single window as well since the nav2 panels would have to work for multiple instances of robots, each in namespaced.

This, in it of itself, is actually not a very big effort. But the problem lies in generalizing this to arbitrary number of `N` robots that can be spawned and and looked after at once. At some point, the Nav2 panels wont be able to appear or the number of things happening in a single window will exceed rviz's capabilities. Keeping them distributed to N rviz windows allows us to more easily scale larger number of robots that can still be commanded / controlled in the panels. 

The desire to do one or the other is purely preference. And it would be not very difficult to create a custom rviz configuration for some specific number, N, robots to appear in a single rviz window for a specific application. As the nav2_bringup package is just a reference for building your own custom nav2 bringup configs / rviz configs / launch of hardware / etc, it can be an additional your proprietary navigation launch package to have a particular rviz config to your preferences.

closed 2021-10-07 by SteveMacenski