ros-navigation/navigation2 #2493: Launching nav2_bringup tb3_simulation_launch.py - Gazebo can't find Child Link[camera_rgb_frame]

Darkproduct, 2021-08-07:
## Bug report

**Required Info:**

- Operating System:
Ubuntu 20.04
- ROS2 Version:
Foxy binaries
- Version or commit hash:
ros-foxy-navigation2	0.4.7-1focal.20210701.023143
- DDS implementation:
Fast-RTPS

#### Steps to reproduce issue
https://navigation.ros.org/getting_started/index.html#running-the-example
```
source /opt/ros/<ros2-distro>/setup.bash
export TURTLEBOT3_MODEL=waffle
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/opt/ros/<ros2-distro>/share/turtlebot3_gazebo/models
// Add --verbose to the gzserver in the launch file.
// Either change it directly or copy it and run it afterwards with
// ros2 launch tb3_simulation_launch.py
ros2 launch nav2_bringup tb3_simulation_launch.py
```
Result:
```
[gzserver-1] [Msg] Waiting for master.
[gzserver-1] [Msg] Connected to gazebo master @ http://127.0.0.1:11345
[gzserver-1] [Msg] Publicized address: 192.168.122.20
[gzserver-1] [Msg] Loading world file [/opt/ros/foxy/share/nav2_bringup/worlds/waffle.model]
[gzserver-1] [Err] [Joint.cc:297] EXCEPTION: Couldn't Find Child Link[camera_rgb_frame]
[gzserver-1] 
[gzserver-1] [Err] [Model.cc:272] LoadJoint Failed
```

#### Expected behavior
Running without errors

#### Actual behavior
Gazebo can't load the broken `waffle.model` from `nav2_bringup`.

#### Possible fixes
Use:
```
ros2 launch tb3_simulation_launch.py world:=/opt/ros/foxy/share/turtlebot3_gazebo/worlds/turtlebot3_worlds/waffle.model
```

Or change this back (from `tb3_simulation_launch.py`):
```python
declare_world_cmd = DeclareLaunchArgument(
        'world',
        # TODO(orduno) Switch back once ROS argument passing has been fixed upstream
        #              https://github.com/ROBOTIS-GIT/turtlebot3_simulations/issues/91
        # default_value=os.path.join(get_package_share_directory('turtlebot3_gazebo'),
        #                            'worlds/turtlebot3_worlds/waffle.model'),
        default_value=os.path.join(bringup_dir, 'worlds', 'waffle.model'),
        description='Full path to world model file to load')
```

SteveMacenski, 2021-08-09:
a PR to add that link to the URDF would be appreciated. Right now we're in the mid-stages of removing dependencies on the TB3 so I don't want to add another dependency by changing the world from our version of the TB3 world to the Robotis version.

Darkproduct, 2021-08-09:
Ok, np. Should I just copy to the waffle.model from the TB3 over, or revert to the previus option to just open the model from the tb3_worlds?

SteveMacenski, 2021-08-09:
No, I think you just need to add that frame that was missing (or remove the other frames that depend on it if not being used)

Darkproduct, 2021-08-09:
Ok. I'll have a look then.

commit d2628a95 (2021-08-10): Fix #2493 by removing the unused joint to camera_rgb_frame (#2497)

* Fix #2493 by removing the unused joint to camera_rgb_frame. It looks like camera_rgb_link link is now camera_link e3ee2da2d5a274b52d32fcdf7e270567ca1093d0

* Remove unwanted line
closed 2021-08-10 by SteveMacenski