ros-navigation/navigation2 #2504: AddressSanitizer report bt_navigator SEGV

zouyonghao, 2021-08-13:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04.2 LTS
- ROS2 Version:
  - Foxy
- Version or commit hash:
  - a7955d8
- DDS implementation:
  - no idea, using the default configuration

#### Steps to reproduce issue

Just run some random navigation tasks with ASAN enabled.

#### Additional information

----

```
[bt_navigator-11] [INFO] [1628499925.752805320] [bt_navigator]: Begin navigating from current location to (0.54, 1.66)
[bt_navigator-11] [ERROR] [1628499925.787389556] [bt_navigator]: Action server failed while executing action callback: "send_goal failed"
[bt_navigator-11] [WARN] [1628499925.787467902] [bt_navigator]: [navigate_to_pose] [ActionServer] Aborting handle.
[bt_navigator-11] [INFO] [1628499939.343112578] [bt_navigator]: Begin navigating from current location to (-2.86, -1.21)
[bt_navigator-11] AddressSanitizer:DEADLYSIGNAL
[bt_navigator-11] =================================================================
[bt_navigator-11] ==14078==ERROR: AddressSanitizer: SEGV on unknown address 0x0000000000a8 (pc 0x7f4c699d5fc4 bp 0x616000078980 sp 0x7f4c5c569328 T17)
[bt_navigator-11] ==14078==The signal is caused by a READ memory access.
[bt_navigator-11] ==14078==Hint: address points to the zero page.
[bt_navigator-11]     #0 0x7f4c699d5fc3 in pthread_mutex_lock (/lib/x86_64-linux-gnu/libpthread.so.0+0xbfc3)
[bt_navigator-11]     #1 0x7f4c6a5453fd in nav2_behavior_tree::BtActionNode<nav2_msgs::action::ComputePathToPose>::tick() (/root/dev_ws/install/nav2_behavior_tree/lib/libnav2_compute_path_to_pose_action_bt_node.so+0x353fd)
[bt_navigator-11]     #2 0x7f4c69ef5733 in BT::TreeNode::executeTick() (/opt/ros/foxy/lib/libbehaviortree_cpp_v3.so+0x6d733)
[bt_navigator-11]     #3 0x7f4c6a1b7e7d in nav2_behavior_tree::RecoveryNode::tick() (/root/dev_ws/install/nav2_behavior_tree/lib/libnav2_recovery_node_bt_node.so+0x6e7d)
[bt_navigator-11]     #4 0x7f4c69ef5733 in BT::TreeNode::executeTick() (/opt/ros/foxy/lib/libbehaviortree_cpp_v3.so+0x6d733)
[bt_navigator-11]     #5 0x7f4c6a281b59 in nav2_behavior_tree::RateController::tick() (/root/dev_ws/install/nav2_behavior_tree/lib/libnav2_rate_controller_bt_node.so+0x5b59)
[bt_navigator-11]     #6 0x7f4c69ef5733 in BT::TreeNode::executeTick() (/opt/ros/foxy/lib/libbehaviortree_cpp_v3.so+0x6d733)
[bt_navigator-11]     #7 0x7f4c69ef3602 in BT::DecoratorNode::executeTick() (/opt/ros/foxy/lib/libbehaviortree_cpp_v3.so+0x6b602)
[bt_navigator-11]     #8 0x7f4c6a16ab92 in nav2_behavior_tree::PipelineSequence::tick() (/root/dev_ws/install/nav2_behavior_tree/lib/libnav2_pipeline_sequence_bt_node.so+0x4b92)
[bt_navigator-11]     #9 0x7f4c69ef5733 in BT::TreeNode::executeTick() (/opt/ros/foxy/lib/libbehaviortree_cpp_v3.so+0x6d733)
[bt_navigator-11]     #10 0x7f4c6a1b7e7d in nav2_behavior_tree::RecoveryNode::tick() (/root/dev_ws/install/nav2_behavior_tree/lib/libnav2_recovery_node_bt_node.so+0x6e7d)
[bt_navigator-11]     #11 0x7f4c69ef5733 in BT::TreeNode::executeTick() (/opt/ros/foxy/lib/libbehaviortree_cpp_v3.so+0x6d733)
[bt_navigator-11]     #12 0x7f4c6a554bb2 in nav2_behavior_tree::BehaviorTreeEngine::run(BT::Tree*, std::function<void ()>, std::function<bool ()>, std::chrono::duration<long, std::ratio<1l, 1000l> >) (/root/dev_ws/install/nav2_behavior_tree/lib/libnav2_behavior_tree.so+0x3bb2)
[bt_navigator-11]     #13 0x7f4c6966d2c2 in nav2_bt_navigator::BtNavigator::navigateToPose() /root/dev_ws/src/navigation2/nav2_bt_navigator/src/bt_navigator.cpp:310:42
[bt_navigator-11]     #14 0x7f4c6977b453 in std::function<void ()>::operator()() const /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/std_function.h:688:14
[bt_navigator-11]     #15 0x7f4c6977b453 in nav2_util::SimpleActionServer<nav2_msgs::action::NavigateToPose, rclcpp::Node>::work() /root/dev_ws/install/nav2_util/include/nav2_util/simple_action_server.hpp:144:9
[bt_navigator-11]     #16 0x7f4c6977a9ba in nav2_util::SimpleActionServer<nav2_msgs::action::NavigateToPose, rclcpp::Node>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::NavigateToPose> >)::'lambda'()::operator()() const /root/dev_ws/install/nav2_util/include/nav2_util/simple_action_server.hpp:135:68
[bt_navigator-11]     #17 0x7f4c6977a9ba in void std::__invoke_impl<void, nav2_util::SimpleActionServer<nav2_msgs::action::NavigateToPose, rclcpp::Node>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::NavigateToPose> >)::'lambda'()>(std::__invoke_other, nav2_util::SimpleActionServer<nav2_msgs::action::NavigateToPose, rclcpp::Node>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::NavigateToPose> >)::'lambda'()&&) /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/invoke.h:60:14
[bt_navigator-11]     #18 0x7f4c6977a9ba in std::__invoke_result<nav2_util::SimpleActionServer<nav2_msgs::action::NavigateToPose, rclcpp::Node>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::NavigateToPose> >)::'lambda'()>::type std::__invoke<nav2_util::SimpleActionServer<nav2_msgs::action::NavigateToPose, rclcpp::Node>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::NavigateToPose> >)::'lambda'()>(nav2_util::SimpleActionServer<nav2_msgs::action::NavigateToPose, rclcpp::Node>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::NavigateToPose> >)::'lambda'()&&) /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/invoke.h:95:14
[bt_navigator-11]     #19 0x7f4c6977a9ba in void std::thread::_Invoker<std::tuple<nav2_util::SimpleActionServer<nav2_msgs::action::NavigateToPose, rclcpp::Node>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::NavigateToPose> >)::'lambda'()> >::_M_invoke<0ul>(std::_Index_tuple<0ul>) /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/thread:244:13
[bt_navigator-11]     #20 0x7f4c6977a9ba in std::thread::_Invoker<std::tuple<nav2_util::SimpleActionServer<nav2_msgs::action::NavigateToPose, rclcpp::Node>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::NavigateToPose> >)::'lambda'()> >::operator()() /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/thread:251:11
[bt_navigator-11]     #21 0x7f4c6977a9ba in std::__future_base::_Task_setter<std::unique_ptr<std::__future_base::_Result<void>, std::__future_base::_Result_base::_Deleter>, std::thread::_Invoker<std::tuple<nav2_util::SimpleActionServer<nav2_msgs::action::NavigateToPose, rclcpp::Node>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::NavigateToPose> >)::'lambda'()> >, void>::operator()() const /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/future:1362:6
[bt_navigator-11]     #22 0x7f4c6977a7d9 in std::_Function_handler<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> (), std::__future_base::_Task_setter<std::unique_ptr<std::__future_base::_Result<void>, std::__future_base::_Result_base::_Deleter>, std::thread::_Invoker<std::tuple<nav2_util::SimpleActionServer<nav2_msgs::action::NavigateToPose, rclcpp::Node>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::NavigateToPose> >)::'lambda'()> >, void> >::_M_invoke(std::_Any_data const&) /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/std_function.h:285:9
[bt_navigator-11]     #23 0x7f4c6a53150c in std::__future_base::_State_baseV2::_M_do_set(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*) (/root/dev_ws/install/nav2_behavior_tree/lib/libnav2_compute_path_to_pose_action_bt_node.so+0x2150c)
[bt_navigator-11]     #24 0x7f4c699dc47e in __pthread_once_slow (/lib/x86_64-linux-gnu/libpthread.so.0+0x1247e)
[bt_navigator-11]     #25 0x7f4c69779f02 in __gthread_once(int*, void (*)()) /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/x86_64-linux-gnu/c++/9/bits/gthr-default.h:700:12
[bt_navigator-11]     #26 0x7f4c69779f02 in void std::call_once<void (std::__future_base::_State_baseV2::*)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*>(std::once_flag&, void (std::__futur [... cut ...] :688:14
[bt_navigator-11]     #15 0x7f4c697650be in rclcpp_action::Server<nav2_msgs::action::NavigateToPose>::call_goal_accepted_callback(std::shared_ptr<rcl_action_goal_handle_t>, std::array<unsigned char, 16ul>, std::shared_ptr<void>) /opt/ros/foxy/include/rclcpp_action/server.hpp:429:5
[bt_navigator-11]     #16 0x7f4c6998db2c in rclcpp_action::ServerBase::execute_goal_request_received() (/opt/ros/foxy/lib/librclcpp_action.so+0x15b2c)
[bt_navigator-11] 
[bt_navigator-11] ==14078==ABORTING

```

SteveMacenski, 2021-08-13:
I’m not clear what this is in reference to. If this is for your other ticket on goal cancellation, close this ticket and the other one like this and add it to that ticket’s thread. Do not make new tickets for a single issue. 

If this is unrelated, you need to provide alot more information about what/why/when/etc

SteveMacenski, 2021-08-13:
After more thinking, I'm almost entirely sure that these are with regards to #2499 so I'm moving this content over and closing the ticket. Lets discuss there, lets keep all discussions and new information on that topic in that ticket please.

closed 2021-08-13 by SteveMacenski