ros-navigation/navigation2 #2506: nodes crash because of uncaught exceptions

easylyou, 2021-08-13:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04.2 LTS
- ROS2 Version:
  - Foxy
- Version or commit hash:
  - a7955d8
- DDS implementation:
  - default

#### Steps to reproduce issue

I had done some tests.These are mainly owing to malformed messages from laser sensor.
```
  try {
    // in fact, the function will throw std::rumtime_error in some situation, 
    // but the exception is not caught.
    projector_.transformLaserScanToPointCloud(message->header.frame_id, *message, cloud, *tf_);
  } catch (tf2::TransformException & ex) {
    ...
    projector_.projectLaser(*message, cloud);
  }
```

#### Expected behavior
Process should not crash.

#### Actual behavior
```
amcl:
	amcl lifecycle node launched. 
	Waiting on external lifecycle transitions to activate
	See https://design.ros2.org/articles/node_lifecycle.html for more information.
terminate called after throwing an instance of 'std::runtime_error'
  what():  cannot store a negative time point in rclcpp::Time
```
```
controller_server:
	controller_server lifecycle node launched. 
	Waiting on external lifecycle transitions to activate
	See https://design.ros2.org/articles/node_lifecycle.html for more information.
	local_costmap lifecycle node launched. 
	Waiting on external lifecycle transitions to activate
	See https://design.ros2.org/articles/node_lifecycle.html for more information.
terminate called after throwing an instance of 'std::runtime_error'
  what():  cannot store a negative time point in rclcpp::Time
```
```
planner_server:
	planner_server lifecycle node launched. 
	Waiting on external lifecycle transitions to activate
	See https://design.ros2.org/articles/node_lifecycle.html for more information.
	global_costmap lifecycle node launched. 
	Waiting on external lifecycle transitions to activate
	See https://design.ros2.org/articles/node_lifecycle.html for more information.
terminate called after throwing an instance of 'std::runtime_error'
  what():  cannot store a negative time point in rclcpp::Time

```
#### Additional information
In fact, there are the same problems in ros1 navigation.
https://github.com/ros-planning/navigation/issues/1151

SteveMacenski, 2021-08-13:
PR would be appreciated, but if you are sending intentionally malformed messages, clearly that’s not a normal runtime situation. But it would always be good to cover these situations. 

It sounds like you’ve been digging into this, I look forward to a PR to review!

SteveMacenski, 2021-08-17:
Merged

closed 2021-08-17 by SteveMacenski