ros-navigation/navigation2 #2507: [Recover] possible use-after-free bug in nav2_costmap_2d::FootprintSubscriber::getFootprint()

easylyou, 2021-08-13:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04.2 LTS
- ROS2 Version:
  - Foxy
- Version or commit hash:
  - a7955d8
- DDS implementation:
  - default

#### Steps to reproduce issue

I had done some tests. And it can be reproduced.

I manually reviewed the related code and found the possible cause.
It looks like a bug of data race, but I can't be sure.

```
// in navigation2/nav2_costmap_2d/src/footprint_subscriber.cpp

bool
FootprintSubscriber::getFootprint(...)
{
  ...
  //1: footprint_->polygon is doing copy construct, but not finished
  //3: copying, boom! use after free.
  footprint = toPointVector(
    std::make_shared<geometry_msgs::msg::Polygon>(footprint_->polygon));
  auto & footprint_stamp = footprint_->header.stamp;
  ...
}

void
FootprintSubscriber::footprint_callback()
{
  //2: set the new footprint_, so the old footprint_ is released 
  footprint_ = msg;
  ...
}
```

#### Expected behavior
Process should not crash.

#### Actual behavior
```
=================================================================
==12174==ERROR: AddressSanitizer: heap-use-after-free on address 0x610000211f40 at pc 0x0000004fef2a bp 0x7f99e7154430 sp 0x7f99e7153bf8
READ of size 12 at 0x610000211f40 thread T13
    #0 0x4fef29 in __asan_memcpy (/home/r1/ros2_clang_navigation/build/nav2_recoveries/recoveries_server+0x4fef29)
    #1 0x7f99f4760060 in void std::_Construct<geometry_msgs::msg::Point32_<std::allocator<void> >, geometry_msgs::msg::Point32_<std::allocator<void> > const&>(geometry_msgs::msg::Point32_<std::allocator<void> >*, geometry_msgs::msg::Point32_<std::allocator<void> > const&) /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/stl_construct.h:75:38
    #2 0x7f99f475fd61 in geometry_msgs::msg::Point32_<std::allocator<void> >* std::__uninitialized_copy<false>::__uninit_copy<__gnu_cxx::__normal_iterator<geometry_msgs::msg::Point32_<std::allocator<void> > const*, std::vector<geometry_msgs::msg::Point32_<std::allocator<void> >, std::allocator<geometry_msgs::msg::Point32_<std::allocator<void> > > > >, geometry_msgs::msg::Point32_<std::allocator<void> >*>(__gnu_cxx::__normal_iterator<geometry_msgs::msg::Point32_<std::allocator<void> > const*, std::vector<geometry_msgs::msg::Point32_<std::allocator<void> >, std::allocator<geometry_msgs::msg::Point32_<std::allocator<void> > > > >, __gnu_cxx::__normal_iterator<geometry_msgs::msg::Point32_<std::allocator<void> > const*, std::vector<geometry_msgs::msg::Point32_<std::allocator<void> >, std::allocator<geometry_msgs::msg::Point32_<std::allocator<void> > > > >, geometry_msgs::msg::Point32_<std::allocator<void> >*) /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/stl_uninitialized.h:83:3
    #3 0x7f99f475fb15 in geometry_msgs::msg::Point32_<std::allocator<void> >* std::uninitialized_copy<__gnu_cxx::__normal_iterator<geometry_msgs::msg::Point32_<std::allocator<void> > const*, std::vector<geometry_msgs::msg::Point32_<std::allocator<void> >, std::allocator<geometry_msgs::msg::Point32_<std::allocator<void> > > > >, geometry_msgs::msg::Point32_<std::allocator<void> >*>(__gnu_cxx::__normal_iterator<geometry_msgs::msg::Point32_<std::allocator<void> > const*, std::vector<geometry_msgs::msg::Point32_<std::allocator<void> >, std::allocator<geometry_msgs::msg::Point32_<std::allocator<void> > > > >, __gnu_cxx::__normal_iterator<geometry_msgs::msg::Point32_<std::allocator<void> > const*, std::vector<geometry_msgs::msg::Point32_<std::allocator<void> >, std::allocator<geometry_msgs::msg::Point32_<std::allocator<void> > > > >, geometry_msgs::msg::Point32_<std::allocator<void> >*) /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/stl_uninitialized.h:137:14
    #4 0x7f99f475f062 in geometry_msgs::msg::Point32_<std::allocator<void> >* std::__uninitialized_copy_a<__gnu_cxx::__normal_iterator<geometry_msgs::msg::Point32_<std::allocator<void> > const*, std::vector<geometry_msgs::msg::Point32_<std::allocator<void> >, std::allocator<geometry_msgs::msg::Point32_<std::allocator<void> > > > >, geometry_msgs::msg::Point32_<std::allocator<void> >*, geometry_msgs::msg::Point32_<std::allocator<void> > >(__gnu_cxx::__normal_iterator<geometry_msgs::msg::Point32_<std::allocator<void> > const*, std::vector<geometry_msgs::msg::Point32_<std::allocator<void> >, std::allocator<geometry_msgs::msg::Point32_<std::allocator<void> > > > >, __gnu_cxx::__normal_iterator<geometry_msgs::msg::Point32_<std::allocator<void> > const*, std::vector<geometry_msgs::msg::Point32_<std::allocator<void> >, std::allocator<geometry_msgs::msg::Point32_<std::allocator<void> > > > >, geometry_msgs::msg::Point32_<std::allocator<void> >*, std::allocator<geometry_msgs::msg::Point32_<std::allocator<void> > >&) /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/stl_uninitialized.h:307:14
    #5 0x7f99f475e58c in std::vector<geometry_msgs::msg::Point32_<std::allocator<void> >, std::allocator<geometry_msgs::msg::Point32_<std::allocator<void> > > >::vector(std::vector<geometry_msgs::msg::Point32_<std::allocator<void> >, std::allocator<geometry_msgs::msg::Point32_<std::allocator<void> > > > const&) /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/stl_vector.h:555:4
    #6 0x7f99f3e84d68 in geometry_msgs::msg::Polygon_<std::allocator<void> >::Polygon_(geometry_msgs::msg::Polygon_<std::allocator<void> > const&) /opt/ros/foxy/include/geometry_msgs/msg/detail/polygon__struct.hpp:35:8
    #7 0x7f99f3e94a64 in void __gnu_cxx::new_allocator<geometry_msgs::msg::Polygon_<std::allocator<void> > >::construct<geometry_msgs::msg::Polygon_<std::allocator<void> >, geometry_msgs::msg::Polygon_<std::allocator<void> >&>(geometry_msgs::msg::Polygon_<std::allocator<void> >*, geometry_msgs::msg::Polygon_<std::allocator<void> >&) /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/ext/new_allocator.h:147:23
    #8 0x7f99f3e94950 in void std::allocator_traits<std::allocator<geometry_msgs::msg::Polygon_<std::allocator<void> > > >::construct<geometry_msgs::msg::Polygon_<std::allocator<void> >, geometry_msgs::msg::Polygon_<std::allocator<void> >&>(std::allocator<geometry_msgs::msg::Polygon_<std::allocator<void> > >&, geometry_msgs::msg::Polygon_<std::allocator<void> >*, geometry_msgs::msg::Polygon_<std::allocator<void> >&) /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/alloc_traits.h:484:8
    #9 0x7f99f39354e6 in std::_Sp_counted_ptr_inplace<geometry_msgs::msg::Polygon_<std::allocator<void> >, std::allocator<geometry_msgs::msg::Polygon_<std::allocator<void> > >, (__gnu_cxx::_Lock_policy)2>::_Sp_counted_ptr_inplace<geometry_msgs::msg::Polygon_<std::allocator<void> >&>(std::allocator<geometry_msgs::msg::Polygon_<std::allocator<void> > >, geometry_msgs::msg::Polygon_<std::allocator<void> >&) /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/shared_ptr_base.h:548:4
    #10 0x7f99f3934f36 in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count<geometry_msgs::msg::Polygon_<std::allocator<void> >, std::allocator<geometry_msgs::msg::Polygon_<std::allocator<void> > >, geometry_msgs::msg::Polygon_<std::allocator<void> >&>(geometry_msgs::msg::Polygon_<std::allocator<void> >*&, std::_Sp_alloc_shared_tag<std::allocator<geometry_msgs::msg::Polygon_<std::allocator<void> > > >, geometry_msgs::msg::Polygon_<std::allocator<void> >&) /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/shared_ptr_base.h:680:6
    #11 0x7f99f3934c07 in std::__shared_ptr<geometry_msgs::msg::Polygon_<std::allocator<void> >, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<geometry_msgs::msg::Polygon_<std::allocator<void> > >, geometry_msgs::msg::Polygon_<std::allocator<void> >&>(std::_Sp_alloc_shared_tag<std::allocator<geometry_msgs::msg::Polygon_<std::allocator<void> > > >, geometry_msgs::msg::Polygon_<std::allocator<void> >&) /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/shared_ptr_base.h:1344:14
    #12 0x7f99f39349cc in std::shared_ptr<geometry_msgs::msg::Polygon_<std::allocator<void> > >::shared_ptr<std::allocator<geometry_msgs::msg::Polygon_<std::allocator<void> > >, geometry_msgs::msg::Polygon_<std::allocator<void> >&>(std::_Sp_alloc_shared_tag<std::allocator<geometry_msgs::msg::Polygon_<std::allocator<void> > > >, geometry_msgs::msg::Polygon_<std::allocator<void> >&) /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/shared_ptr.h:359:4
[... cut ...]
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==83523==ABORTING
```

SteveMacenski, 2021-08-13:
What is your actual issue? You've provided alot of GDB output, but its unrealistic for me to try to fish through that if I don't understand what your problem is. If you have a solution, please submit a PR.

SteveMacenski, 2021-08-13:
Get footprint checks
```
  if (!footprint_received_) {
    return false;
  }
```

so it would fail if the callback was not yet triggered. that boolean is only set to true after the `footprint_` has been set 

```
  footprint_ = msg;
  if (!footprint_received_) {
    footprint_received_ = true;
  }
```

easylyou, 2021-08-16:
Thanks for your reply! 
I try to figure out the cause. It is helpful to discuss about the cause of the output. If it is a real bug, recover node will die after the bug is trigger.  Here are my further analyses.

**Every times a new message received from footprint_topic,  footprint_callback will be called.**
**create_subscription(). Here are the related codes**
```
//in navigation2/nav2_recoveries/src/recovery_server.cpp
nav2_util::CallbackReturn
RecoveryServer::on_configure(const rclcpp_lifecycle::State & /*state*/)
{
 ...
  footprint_sub_ = std::make_unique<nav2_costmap_2d::FootprintSubscriber>(
    shared_from_this(), footprint_topic, 1.0);
  ... 
}

//in navigation2/nav2_costmap_2d/src/footprint_subscriber.cpp
FootprintSubscriber::FootprintSubscriber() 
{
  footprint_sub_ = rclcpp::create_subscription<geometry_msgs::msg::PolygonStamped>(
    node_topics_,
    topic_name, rclcpp::SystemDefaultsQoS(),
    std::bind(&FootprintSubscriber::footprint_callback, this, std::placeholders::_1));
}
```

**Every times the recover try to spin, getFootprint will be called**
**ActionServer(). Here are the related codes**
```
//in navigation2/nav2_recoveries/include/nav2_recoveries/recovery.hpp
void configure()
{
  ...
    action_server_ = std::make_shared<ActionServer>(
      node_, recovery_name_,
      std::bind(&Recovery::execute, this));
  ...
}

void execute()
{
  ...
  switch (onCycleUpdate()) {
  ...
}

//in navigation2/nav2_recoveries/plugins/spin.cpp
Status Spin::onCycleUpdate()
{
  ...
  if (!isCollisionFree(relative_yaw_, cmd_vel.get(), pose2d))
  ...
}

//in navigation2/nav2_costmap_2d/src/costmap_topic_collision_checker.cpp
bool CostmapTopicCollisionChecker::isCollisionFree(
  const geometry_msgs::msg::Pose2D & pose)
{
  ...
   if (scorePose(pose) >= LETHAL_OBSTACLE) 
   ...

}

double CostmapTopicCollisionChecker::scorePose(
  const geometry_msgs::msg::Pose2D & pose)
{
  ...
  return collision_checker_.footprintCost(getFootprint(pose));
}
```
**Those mean that getFootprint() and footprint_callback() could run multiple times and run concurrently**
Because of multiple runs, the footprint checks can not prevent it.

SteveMacenski, 2021-08-16:
Lets chat in the PR

SteveMacenski, 2021-08-17:
Merging once we have CI turning over, closing issue since solution was approved.

closed 2021-08-17 by SteveMacenski