ros-navigation/navigation2 #2508: [Planner] possible use-after-free bug in nav2_costmap_2d

easylyou, 2021-08-13:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04.2 LTS
- ROS2 Version:
  - Foxy
- Version or commit hash:
  - a7955d8
- DDS implementation:
  - default

#### Steps to reproduce issue

I had done some tests. And it can be reproduced.

```
[INFO] [1628747203.189238115] [global_costmap.global_costmap]: Configuring
[INFO] [1628747203.199641398] [global_costmap.global_costmap]: Using plugin "static_layer"
[INFO] [1628747203.205795632] [global_costmap.global_costmap]: Subscribing to the map topic (/map) with transient local durability
[INFO] [1628747203.208597874] [global_costmap.global_costmap]: Initialized plugin "static_layer"
[INFO] [1628747203.208680902] [global_costmap.global_costmap]: Using plugin "obstacle_layer"
[INFO] [1628747203.211488645] [global_costmap.global_costmap]: Subscribed to Topics: scan
[INFO] [1628747203.218346306] [global_costmap.global_costmap]: Initialized plugin "obstacle_layer"
[INFO] [1628747203.218425410] [global_costmap.global_costmap]: Using plugin "voxel_layer"
[INFO] [1628747203.220963960] [global_costmap.global_costmap]: StaticLayer: Resizing costmap to 384 X 384 at 0.050000 m/pi
```

According to the log of a real launch, it is possible that a resizing costmap of staticLayer happens before voxel_layer complete the initializing.

```
in navigation2/nav2_voxel_grid/src/voxel_grid.cpp

void VoxelGrid::resize(unsigned int size_x, unsigned int size_y, unsigned int size_z)
{
  ...
  //2. a resize happends
  delete[] data_;
  
  ...

  data_ = new uint32_t[size_x_ * size_y_];

  ...
}

void VoxelGrid::reset()
{
  //1. VoxelGrid is initializing
  //3. boom! use after free
  uint32_t * col = data_;
  for (unsigned int i = 0; i < size_x_ * size_y_; ++i) {
    *col = unknown_col;
    ++col;
  }
}
```

#### Expected behavior
Process should not crash.

#### Actual behavior
```
==178418==ERROR: AddressSanitizer: heap-use-after-free on address 0x6020000560f0 at pc 0x7fc6b81ae43a bp 0x7ffc328e9410 sp 0x7ffc328e9408
WRITE of size 4 at 0x6020000560f0 thread T0
    #0 0x7fc6b81ae439 in nav2_voxel_grid::VoxelGrid::reset() /home/r1/ros2_clang_navigation/src/navigation2/nav2_voxel_grid/src/voxel_grid.cpp:103:10
    #1 0x7fc6b98b235c in nav2_costmap_2d::VoxelLayer::resetMaps() /home/r1/ros2_clang_navigation/src/navigation2/nav2_costmap_2d/plugins/voxel_layer.cpp:126:15
    #2 0x7fc6b8e42a38 in nav2_costmap_2d::Costmap2D::resizeMap(unsigned int, unsigned int, double, double, double) /home/r1/ros2_clang_navigation/src/navigation2/nav2_costmap_2d/src/costmap_2d.cpp:88:3
    #3 0x7fc6b90e0fc4 in nav2_costmap_2d::CostmapLayer::matchSize() /home/r1/ros2_clang_navigation/src/navigation2/nav2_costmap_2d/src/costmap_layer.cpp:59:3
[... 286 lines cut ...]
  0x0c0480002e90: fa fa fd fd fa fa fd fa fa fa fd fa fa fa fd fa
  0x0c0480002ea0: fa fa fd fa fa fa 00 00 fa fa 00 00 fa fa 00 00
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==65951==ABORTING

```

SteveMacenski, 2021-08-13:
> it is possible that a resizing costmap of staticLayer happens before voxel_layer complete the initializing

How? Please show in the code where you're talking about. You've provided alot of GDB output, but its unrealistic for me to try to fish through that if I don't understand what your problem is. If you have a solution, please submit a PR.

easylyou, 2021-08-16:
Thanks for your reply! Here are my further analyses. It seems a real bug related to multi-threading hard to be detected. If that happens, robot will be out of control.

**layered_costmap_ is a global value that all plugins have.**
**Every plugin initialize one by one.**
```
in navigation2/nav2_costmap_2d/src/costmap_2d_ros.cpp
nav2_util::CallbackReturn
Costmap2DROS::on_configure(const rclcpp_lifecycle::State & /*state*/)
{
  ...
  // Create the costmap itself
  layered_costmap_ = new LayeredCostmap(global_frame_, rolling_window_, track_unknown_space_);
  ...
  // Then load and add the plug-ins to the costmap
  for (unsigned int i = 0; i < plugin_names_.size(); ++i) {

    std::shared_ptr<Layer> plugin = plugin_loader_.createSharedInstance(plugin_types_[i]);
    layered_costmap_->addPlugin(plugin);

    plugin->initialize(
      layered_costmap_, plugin_names_[i], tf_buffer_.get(),
      shared_from_this(), client_node_, rclcpp_node_);
  }
  ...
}
```
**incomingMap() can be called  at any time after the plugin of static_layer have done onInitialize().**
**That is, layered_costmap_->resizeMap() may be called before other plugins complete  onInitialize().**
**That is the key problem.**
```
//in navigation2/nav2_costmap_2d/plugins/static_layer.cpp
void
StaticLayer::onInitialize()
{
  ...
  map_sub_ = node_->create_subscription<nav_msgs::msg::OccupancyGrid>(
    map_topic_, map_qos,
    std::bind(&StaticLayer::incomingMap, this, std::placeholders::_1));
  ...
}

void
StaticLayer::processMap(const nav_msgs::msg::OccupancyGrid & new_map)
{
  ...
  layered_costmap_->resizeMap(
      size_x, size_y, new_map.info.resolution,
      new_map.info.origin.position.x,
      new_map.info.origin.position.y,
      true);
  ...
}
```

- First problem:  std::vector is unsafe in multithreading.
**thread one : layered_costmap_->addPlugin(); vector->push_back()**
Because vector is added an element, it may expand capacity. All elements will be copied to a new place and the olds will be freed.
```
Costmap2DROS::on_configure(const rclcpp_lifecycle::State & /*state*/)
{
  for (unsigned int i = 0; i < plugin_names_.size(); ++i) {

    std::shared_ptr<Layer> plugin = plugin_loader_.createSharedInstance(plugin_types_[i]);
    layered_costmap_->addPlugin(plugin);

    plugin->initialize(
      layered_costmap_, plugin_names_[i], tf_buffer_.get(),
      shared_from_this(), client_node_, rclcpp_node_);
  }
  ...
}
```
**thread two: iterate vector**
It is possible that (*plugin) is pointed to a freed memory.  That is the second debug output in my first post.
```
void LayeredCostmap::resizeMap(...)
{
  for (vector<std::shared_ptr<Layer>>::iterator plugin = plugins_.begin();
    plugin != plugins_.end(); ++plugin)
  {
    (*plugin)->matchSize();
  }
}
```

- Second problem: onInitialize() and  matchSize() can be called concurrently! Let's see what will happen in voxel_layer.
```
void VoxelLayer::onInitialize()
{
  ...
  matchSize();
}
```
```
void VoxelLayer::matchSize()
{
  voxel_grid_.resize(size_x_, size_y_, size_z_);
}
```
Those codes above show that voxel_grid_.resize() can be called concurrently.

```
void VoxelGrid::resize(unsigned int size_x, unsigned int size_y, unsigned int size_z)
{
   ...

  delete[] data_;
  size_x_ = size_x;
  size_y_ = size_y;
  size_z_ = size_z;

  ...

  data_ = new uint32_t[size_x_ * size_y_];
  uint32_t unknown_col = ~((uint32_t)0) >> 16;
  uint32_t * col = data_;
  for (unsigned int i = 0; i < size_x_ * size_y_; ++i) {
    *col = unknown_col;
    ++col;
  }
}
```
**Depends on the order of execution in multithreading, it could be use-after-free or double-free. **

SteveMacenski, 2021-08-16:
I have to admit, these seem like incredibly unlikely things to occur, but happy to have fixes! First merged, second I'm thinking about for a little bit but likely to also be merged

SteveMacenski, 2021-08-16:
Both merged! I think that closes this ticket out, feel free to comment back if I am incorrect

closed 2021-08-16 by SteveMacenski