ros-navigation/navigation2 #2509: amcl in ros2 has higher errors than amcl in ros1,even though the parameters are the same.

poornimajd, 2021-08-13:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu 18.04
- ROS2 Version:
  -  Foxy
- Version or commit hash:
  - ros-foxy-navigation2	0.4.7-1focal.20210619.022653

- DDS implementation:
  - 

#### Steps to reproduce issue

```

```
The amcl parameter files have been attached here,for both [ros1](https://github.com/poornimajd/newdemo/blob/main/amcl_files/amcl_ros1.txt) and [ros2](https://github.com/poornimajd/newdemo/blob/main/amcl_files/amcl_ros2.txt).Along with the [waypoints](https://github.com/poornimajd/newdemo/blob/main/amcl_files/pose.csv) 
Map file is attached [here](https://github.com/poornimajd/newdemo/blob/main/amcl_files/map_default.yaml)

#### Expected behavior
The localization error should be same in both versions of ros,since same map,same world file and same parameters are used.
#### Actual behavior
The average errors in localization are as follows-
 
    Ros1-Avg err in x=4cm ,y=5cm ,yaw=0.015 rad
    Ros2-Avg err in x=10cm,y=12cm,yaw=0.07rad

#### Additional information
The parameter files,initial pose estimation,map file and same way points are used for both the ros versions.The world file is also the same for both the versions.The experiment is carried out on burger model in both the versions.
The average error in localization over the entire path in x direction and y direction has been reported above.

----
## Feature request

#### Feature description

#### Implementation considerations

SteveMacenski, 2021-08-13:
PR would be appreciated but not likely something maintainers will be spending much time to analyze in the foreseeable future.

poornimajd, 2021-08-14:
Thanks for your speedy reply.

> This could be a result of absolutely anything, including different planners controllers amcl or even the robot model drivers itself.

Yes,sounds reasonable.

> You’d have to do experiments with only ROS1 or ROS2 navigation and take a single run as a dataset and then running both on that exact same dataset.

Yes I tried to take the exact same waypoints and used only the navigation stack in the respective versions,to get the results.As the documentation says that AMCL is similar in both ROS1 and ROS2,hence the avg errors in ROS2 should have been at least close to that of ROS1,but here they are almost the double.

I am also suspecting the QOS setting in ROS2,which is best effort UDP,while in ROS1 it is TCP,but other than this,I feel there is nothing much on the AMCL side,as I made sure that all the parameters are same in ROS1, and ROS2.

Do you have any thoughts/comments on the last point I made?
Thanks a lot once again.
Awaiting for your reply.

SteveMacenski, 2021-08-16:
Please ask on ROS answers, this is not an isolated enough experiment to note any specific errors in AMCL with. I certainly recognize that it could be possible, but without a more scientific experiment, this isn't something actionable

closed 2021-11-21 by poornimajd

muratkoc503, 2023-02-05:
@poornimajd,hello, you can look Kaiyu Zheng - navguide.pdf. (page 19-20)
if you set alpha1 to alpha5 parameter to small (decrease), particles are more condensed, otherwise the particles are more spread-out.

my param:
    alpha1: 0.005
    alpha2: 0.005
    alpha3: 0.005
    alpha4: 0.005
    alpha5: 0.005
    pf_err: 0.01
    pf_z: 0.99

mikeferguson, 2023-02-06:
Looks like this is an ancient ticket, but since someone digged it up and I saw it cross my inbox:

You are probably using slightly different models - for ROS1, somewhere around Kinetic/Melodic, it was pointed out that the odom models were NOT implemented the same as the book - and so we have "diff" and "diff-corrected" models - the default is the "old" diff model in ROS1, but I expect that in porting to ROS2, the old "diff" model was dropped and only the "diff-corrected" version remains.

muratkoc503, 2023-02-06:
@mikeferguson thanks for reply. ı use omnidirectional robot at simulation.

gtoff, 2023-10-06:
I stumbled upon this issue today.
I also have the purely anecdotal "impression" that there is some sort of regression in acml between ROS1 (kinetic) and ROS2 (humble) on all our turtlebot2. But, as Steve already pointed out, system level results won't cut it.
I wonder if it has to do with what @mikeferguson  pointed out.

Rather than trying the proper experiment as suggested (isolating AMCL), I just did a quick comparison on the TB3 in simulation in noetic vs humble. 
With the same initial position error (0.0,0.4) I have:
- In noetic particles converge very quickly (one in place 180 rotation does it) and "error" (measured as translation between map and odom frames) is always below 4cm
- In humble particles never really converge (although after 180 in place rotation error goes down to 1cm), but error can go up to 10cm after reaching few goals