ros-navigation/navigation2 #2514: nav2_bringup doesn't load the map if current directory doesn't contain the file

jonathanTIE, 2021-08-16:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10
- ROS2 Version:
  - Foxy binaries from chocolatey/ms IoT - version 20201211.0.0.2107200001-pre
- Version or commit hash:
  - 0.4.5
- DDS implementation:
  - CycloneDDS

#### Steps to reproduce issue
Following the instructions to launch nav2 on a robot : https://github.com/ros-planning/navigation2/tree/foxy-devel/nav2_bringup/bringup
I try to launch the bringup with a map on windows using this command without being in the same directory as the map.
```
ros2 launch nav2_bringup bringup_launch.py map:=<full/path/to/map.yaml> map_type:=occupancy
```

#### Expected behavior
The map should load properly.
``` 
[map_server.EXE-1] [INFO] [map_io]: Loading image_file: ././map.png
```

#### Actual behavior
I can't see the map on rviz, nav2 doesn't work, and "OpenBlob" is unable to load the image file
```
[map_server.EXE-1] [INFO] [map_io]: Loading yaml file: C:\opt\ros\foxy\x64\src\robot_simu_enac\map.yaml
[map_server.EXE-1] [DEBUG] [map_io]: resolution: 0.01
[map_server.EXE-1] [DEBUG] [map_io]: origin[0]: 0
[map_server.EXE-1] [DEBUG] [map_io]: origin[1]: 0
[map_server.EXE-1] [DEBUG] [map_io]: origin[2]: 0
[map_server.EXE-1] [DEBUG] [map_io]: free_thresh: 0.196
[map_server.EXE-1] [DEBUG] [map_io]: occupied_thresh: 0.65
[map_server.EXE-1] [DEBUG] [map_io]: mode: trinary
[map_server.EXE-1] [DEBUG] [map_io]: negate: 0
[map_server.EXE-1] [INFO] [map_io]: Loading image_file: ./C:/opt/ros/foxy/x64/src/robot_simu_enac/map.png
[map_server.EXE-1] [INFO] [1629139320.231949600] [map_server]:
[map_server.EXE-1]      map_server lifecycle node launched.
[map_server.EXE-1]      Waiting on external lifecycle transitions to activate
[map_server.EXE-1]      See https://design.ros2.org/articles/node_lifecycle.html for more information.
[map_server.EXE-1] [INFO] [1629139320.232021100] [map_server]: Creating
[map_server.EXE-1] [INFO] [1629139320.251053600] [map_server]: Configuring
[map_server.EXE-1] [ERROR] [map_io]: Failed to load image file ./C:/opt/ros/foxy/x64/src/robot_simu_enac/map.png for reason: map_server: Unable to open file (./C:/opt/ros/foxy/x64/src/robot_simu_enac/map.png) reported by D:\bld\graphicsmagick\src\bf3ade807d-d804a40dca.clean\magick\blob.c:3058 (OpenBlob)
```
#### Additional information
I tried both to put only the file name and the full path to the image in the map.yaml but it doesn't work too, the only workaround that in found is to do a cd to be in the same directory, and I don't think it's an intended behaviour.

Also the bringup instructions are not written for Windows, can I write a PR to add these instructions?

SteveMacenski, 2021-08-16:
What is your specific command you ran, with specific map, etc

SteveMacenski, 2021-08-19:
any updat?

jonathanTIE, 2021-08-20:
Yes, I changed the path separator from the usual Windows one `\` to `/`. So I guess that when map_io read `map.yaml` and look for the image, it looks for the current directory or it takes the full path given, and it search in the same folder for the image file. I can't build the code to test, but I think for the image_file, it expects normal slash, but instead I have `C:\opt\ros\foxy\x64\src\robot_simu_enac\map.yaml` and then it generates ` ./C:\opt\ros\foxy\x64\src\robot_simu_enac\map.png`. 

I don't know how to modify map_io to support windows filepaths for now.

SteveMacenski, 2021-08-20:
Not sure, I'd be happy to merge a PR to support, but still stand that this isn't something I know much about to give you any meaningful assistance (I haven't used a Windows machine since college). 

@seanyen this is a windows user having issues with Nav2, is this something you'd be able to help with?

SteveMacenski, 2021-10-07:
Currently, windows is not a supported platform. Happy to merge a PR to enable, but closing this ticket to remove from the queue since there's no action from the issuer and it is not in our maintenance plan to do so in the foreseeable future.

closed 2021-10-07 by SteveMacenski