ros-navigation/navigation2 #2520: smac_planner seems to be ignoring all cost values other than LETHAL and INSCRIBED

saadehmd, 2021-08-18:
Not sure if this is a bug, so removing the bug report tag. 

#### DESCRIPTION
Unlike the navfn planner smac planner only reacts to the costs that are >= INSCRIBED cost value. A visual representation of this is shown in the GIFs attached. The GIF shows an infinite loop of replanning, where obstacles are registered from the [Social navigation Layer](http://wiki.ros.org/social_navigation_layers). So the costs are laid out in a velocity-skewed gaussian-fashion around the moving obstacles. A lobe at the center is forcefully turned to be equal to INSCRIBED cost for ease of visualization ( although in reality they change smoothly starting from the center).  

#### Expected behavior
 - The planner stays away from the high cost areas as much as possible atleast until MAX_NON_OBSTACLE cost threshold.
 - The planner should at any cost stay away from INSCRIBED and OCCUPIED cells

#### Actual Behavior 
The planner stays away from INSCRIBED and OCCUPIED cells but doesn't react at all to any other cost values unlike the NavFn planner

#### Additional information
Navfn_planner GIF
[image]

Smac_planner GIF
[image]

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  -

saadehmd, 2021-08-18:
planner_server:
  ros__parameters:
    expected_planner_frequency: 10.0
    use_sim_time: True
    planner_plugins: ["GridBased"]
    GridBased:
      plugin: "nav2_smac_planner/SmacPlannerHybrid"
      tolerance: 0.5                      # tolerance for planning if unable to reach exact pose, in meters, for 2D node
      downsample_costmap: false           # whether or not to downsample the map
      downsampling_factor: 1              # multiplier for the resolution of the costmap layer (e.g. 2 on a 5cm costmap would be 10cm)
      allow_unknown: false                # allow traveling in unknown space
      max_iterations: 45000000            # maximum total iterations to search for before failing (in case unreachable), set to -1 to disable
      max_on_approach_iterations: 1000    # maximum number of iterations to attempt to reach goal once in tolerance, 2D only
      max_planning_time: 3.5              # max time in s for planner to plan, smooth, and upsample. Will scale maximum smoothing and upsampling times based on remaining time after planning.
      motion_model_for_search: "MOORE"    # 2D Moore, Von Neumann; Hybrid Dubin, Redds-Shepp; State Lattice set internally
      cost_travel_multiplier: 2.0         # For 2D: Cost multiplier to apply to search to steer away from high cost areas. Larger values will place in the center of aisles more exactly (if non-`FREE` cost potential field exists) but take slightly longer to compute. To optimize for speed, a value of 1.0 is reasonable. A reasonable tradeoff value is 2.0. A value of 0.0 effective disables steering away from obstacles and acts like a naive binary search A*.
      angle_quantization_bins: 64         # For Hybrid nodes: Number of angle bins for search, must be 1 for 2D node (no angle search)
      analytic_expansion_ratio: 3.5       # For Hybrid/Lattice nodes: The ratio to attempt analytic expansions during search for final approach.
      minimum_turning_radius: 2.0        # For Hybrid/Lattice nodes: minimum turning radius in m of path / vehicle
      reverse_penalty: 4.1                # For Reeds-Shepp model: penalty to apply if motion is reversing, must be => 1
      change_penalty: 1.25                # For Hybrid nodes: penalty to apply if motion is changing directions, must be >= 0
      non_straight_penalty: 1.5          # For Hybrid nodes: penalty to apply if motion is non-straight, must be => 1
      cost_penalty: 3.0                   # For Hybrid nodes: penalty to apply to higher cost areas when adding into the obstacle map dynamic programming distance expansion heuristic. This drives the robot more towards the center of passages. A value between 1.3 - 3.5 is reasonable.
      lookup_table_size: 20.0               # For Hybrid nodes: Size of the dubin/reeds-sheep distance window to cache, in meters.
      cache_obstacle_heuristic: false      # For Hybrid nodes: Cache the obstacle map dynamic programming distance expansion heuristic between subsiquent replannings of the same goal location. Dramatically speeds up replanning performance (40x) if costmap is largely static.     
      smoother:
        max_iterations: 1000
        w_smooth: 0.3
        w_data: 0.2
        tolerance: 1e-10

SteveMacenski, 2021-08-18:
A view with the costmap in rviz would be valuable. The smac planner most certainly does take into account the costmap values that are non-lethal/inscribed, that's how we stay in the center of aisles and round corners without clipping them. 

I cannot, off hand, explain the behavior you're presenting, but since this isn't showing the actual cost grid, I can't really know what it is that I'm truly looking at or what the planner has access to (or how that all looks). There are tunable weights for how much we penalize non-straight / turning / costed paths that could be better tuned if you're in an environment where you would rather your robot make very curvy / wobbly decisions even at the expense of higher lengthed paths. 

What I slightly expect is happening (and again, because not showing me the full picture, I can't say with any confidence) is that you're setting a trivial path in plain view of the robot so the analytic expansions are finding the path. When the analytic expansion occurs on occasion, it tries to find the path from the currently expanded node to the goal using the Dubin/Reeds-Sheep motion models ignoring everything about the environment. It then does periodic collision checks along the path to make sure its valid. If so, then planning is done. If not, expansion continues and it'll try again in 10-100 (or something) iterations. In that case when the goal is in plain site and the analytic expansion is valid, then you're right it would straight up cut off something if that was kinematically feasible and without collision.  

PS those gifs are painfully slow, in the future please speed them up like 5x at minimum before posting so we can actually see what's happening :wink:

SteveMacenski, 2021-08-18:
so tl;dr
- This may be tunable in smac with penalties or adjusting the turning radius to be representative of your robot's actual drivetrain if you want tighter turns. Tuning penalty functions isn't sexy but you can often get the behavior you want.
-  This may benefit greatly from tuning the social layer's Gaussian distribution to be scaled appropriately to your penalty settings in smac. The cost laid down in the costmap by the distribution and the cost penalty in search are going to be always linked, so add that dial in your tuning process. For the "never cut off" situation, you could also try to replace the Gaussian with just a straight Gaussian-shaped block of high cost, that would definitely make the search algorithm think twice about entering it regardless of how well you've tuned the smac penalty functions. 
- If this is due to analytic expansions, that may need a code solution
- To some level, a local trajectory planner will help in this situation as well

saadehmd, 2021-08-19:
Thanks a lot for a very thorough explanation on this and sorry about the poor slow GIFs. 
So a few variations a tried based on your arguments about the penalty params, the lack of freely swirling paths due to maneuverability constraints and the diminishing cost values from the gaussians. I still can't definitively say, if what i am seeing is exactly this behavior as i'd actually have to look at the expanded nodes and print out some cost values to say for sure. But just from my own intuition, i went with following corner cases that somewhat indirectly point to the planner being somewhat in-reactive to the costs that are neither OCCUPIED nor INSCRIBED:-

cost_penalty: 50 (i know it's quite high, but i always like to start tuning things from a ridiculous imbalance and work towards the required balance)
non_straight_penalty: 1.0
reverse_penalty:5.0
path_change_penalty: 1.0
analitycal_expansion_ratio:3.5
angle_quantization_bins: 72
global__costmap: resolution: 1.0 
minimum_turning radius:  1.0 , 10.0 , 100.0, 200.0 (I tried 4 different radii just to go from one extreme to the other with the same values above). GIFs are as follows :
[image]
turning radius: 1.0m

[image]
turning radius: 10.0m

[image]
turning radius: 100.0m

[image]
turning radius: 200.0m

As for the costmap issue, instead of the smooth gaussian, i changed it into a four-sector gaussian with costs 253, 128, 64 and 32 from middle to the circumference. And tested the same setup as above instead this time i kept the min_turning_radius fixed at 1m in order to keep the maneuverability restriction at bay. So with these tests i changed the INSCRIBED COST threshold in smac_planner/include/constants.hpp  to 253, 128, 64 and 32. GIFs are as follows:

[image]
INSCRIBED_COST_253

[image]
INSCRIBED_COST_128

[image]
INSCRIBED_COST_64

[image]
INSCRIBED_COST_32

And lastly checked the INSCRIBED THRESHOLD = 32 with 200m of turning radius, cost_penalty = 2.0, non-straight_penalty = 1.5, path_change_penalty = 2.0 and angle_quantization_bins = 128
[image]
INSCRIBED_COST_32_200m

The GIFs are not probably that much clear. Couldn't find a really good GIF tool online. but i uploaded the videos just in case:-
https://drive.google.com/drive/folders/1NE8B33LVfjIzz3V2qRSVxUu0iFRra7IS?usp=sharing

I hope, i'm not complicating it even more. Feel like i have only been messing around with the planner rather than actually getting into it's nuts and bolts. But the short conclusion is that , even with all the apparent manuevaribility restrictions, cost factors and cost thresholds, the planner seems pretty resilient against the cost values that are rejected within the checkCollision() function and always manages to plan around them but pretty much just reacts to the rest of the values as free space.

SteveMacenski, 2021-08-19:
Your first 4 examples (where the background color is "light") looks like its doing more or less what you want, its planning around the gaussian. Any issues there? I'm not sure exactly what you were trying to describe there or if that was just a confirmation that you're getting closer to what you want. Clearly there's some tuning to be done, but I think that's a proof point that you can get the behavior you need with some penalty / radius / guassian tuning.

The next 4 examples feels the same way (where the background is "dark") -- getting closer to the behavior you'd want, just need to play with things and get to honed in to something you're happy with. Personally looking at the size of your space, I'd make that entire guassian cloud at least 150 at the lowest value and raising up to 200 as something to play with. It's not like there isn't plenty of space to go around for the robot to really ever need to enter the "egg".

I wouldn't use those really big turning radiuses, that would really limit your maneuverability. I think on the turning radius, you should pick a value that lets you negotiate around the turns needed for your environments and stick to that number and tune the rest with that set statically. For ackermann / legged these are parameters of the base itself that could be set higher to be conservative. For diff / omni, you can set it based on the spaces you're in and what you expect your robot to need to be able to do. I don't want to give any specific number suggestions because I don't know your application / robot, but looking at those gifs, I think you can see where the turning radius being too large is just not workable. 

Thanks for that detailed analysis, that's nice to see. I always love seeing my work on people's robots! My email is on my github profile (not linking here just for scanning bots), I'd love to hear a little about what you're working on if you're open to sharing. Anything else we need to chat about here or are we good to close this out?

saadehmd, 2021-08-19:
I do plan to mitigate this with a trajectory planner but my first concern was to get planner as close to the actual maneuver as possible particularly because my use-case is a marine application and the dynamic obstacles aren't moving that fast anyway. And since i could easily fair 5 - 8 HZ of frequency out of SMAC so i thought why shouldn't i tune this as much as i can before moving on. One alternative i am thinking of is to just keep using a very low INSCRIBED COST THRESHOLD somewhere around 64 although that would completely defeat the point of proxemic layer and might over-restrict the planner when in over-crowded situations. Your thoughts on the alternatives ?

SteveMacenski, 2021-08-19:
It's a tuning game depending on your environment / robot. I give penalty functions so you can tune it to your liking and gave defaults I thought were "reasonable" for a large number of applications to  start with. If you think crowding is a concern, maybe don't charge up the costs a ton, but depends on how often you think that will happen. The robot will still plan but the costs will be high, which isn't inherently a problem.

saadehmd, 2021-08-19:
Thanks a lot for your time. The project is an AR-based navigation guidance system for containers, fishing vessels and leisure boats. Was just trying to see, how much i can re-use from the world of robotics, although there's quite substantial differences in marine motion planning but anyways let's see how well this plays out.

I consider the issue CLOSED now. Thanks and cheers.

closed 2021-08-19 by saadehmd

SteveMacenski, 2021-08-19:
Super cool, do you have a link?