ros-navigation/navigation2 #2523: [bt_navigator] A bug of NULL pointer dereference in bt_action_node.hpp.

easylyou, 2021-08-19:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04.2 LTS
- ROS2 Version:
  - Foxy
- Version or commit hash:
  - a7955d8
- DDS implementation:
  - default

#### Steps to reproduce issue

I had done some tests. And it can be reproduced.
```cpp
// in navigation2/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_node.hpp

void on_new_goal_received()
{
    ...
    auto future_goal_handle = action_client_->async_send_goal(goal_, send_goal_options);
    ...
    // goal_handle_ could be set NULL here.
    // The exception is not handled well.
    goal_handle_ = future_goal_handle.get();
    if (!goal_handle_) {
        throw std::runtime_error("Goal was rejected by the action server");
    }
}

BT::NodeStatus tick() override
{
    if (status() == BT::NodeStatus::IDLE) {
        ...
        setStatus(BT::NodeStatus::RUNNING);
        on_new_goal_received();
    }

    if (rclcpp::ok() && !goal_result_available_) {
        ...
        // goal_handle_ could be NULL here.
        auto goal_status = goal_handle_->get_status();
```

#### Expected behavior
Process should not crash.

#### Actual behavior
NULL pointer dereference.
I think #2504 is the same issue.
```
==22779==ERROR: AddressSanitizer: SEGV on unknown address 0x0000000000a8 (pc 0x7f01525dafc4 bp 0x7f0144fc7f20 sp 0x7f0144fc7ef8 T17)
==22779==The signal is caused by a READ memory access.
==22779==Hint: address points to the zero page.
    #0 0x7f01525dafc4 in pthread_mutex_lock /build/glibc-eX1tMB/glibc-2.31/nptl/../nptl/pthread_mutex_lock.c:67:23
    #1 0x7f01543b0bc0 in __gthread_mutex_lock(pthread_mutex_t*) /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/x86_64-linux-gnu/c++/9/bits/gthr-default.h:749:12
    #2 0x7f01543c203b in std::mutex::lock() /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/std_mutex.h:100:17
    #3 0x7f01543d81fa in std::lock_guard<std::mutex>::lock_guard(std::mutex&) /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/std_mutex.h:159:19
    #4 0x7f0154415fc2 in rclcpp_action::ClientGoalHandle<nav2_msgs::action::ComputePathToPose>::get_status() /opt/ros/foxy/include/rclcpp_action/client_goal_handle_impl.hpp:108:31
    #5 0x7f01543b9eea in nav2_behavior_tree::BtActionNode<nav2_msgs::action::ComputePathToPose>::tick() /home/r1/ros2_clang_navigation/src/navigation2/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_node.hpp:149:40
    #6 0x7f0152d7b733 in BT::TreeNode::executeTick() (/opt/ros/foxy/lib/libbehaviortree_cpp_v3.so+0x6d733)
    #7 0x7f015319c602 in nav2_behavior_tree::RecoveryNode::tick() /home/r1/ros2_clang_navigation/src/navigation2/nav2_behavior_tree/plugins/control/recovery_node.cpp:44:53
    #8 0x7f0152d7b733 in BT::TreeNode::executeTick() (/opt/ros/foxy/lib/libbehaviortree_cpp_v3.so+0x6d733)
    #9 0x7f01535a241e in nav2_behavior_tree::RateController::tick() /home/r1/ros2_clang_navigation/src/navigation2/nav2_behavior_tree/plugins/decorator/rate_controller.cpp:60:53
    #10 0x7f0152d7b733 in BT::TreeNode::executeTick() (/opt/ros/foxy/lib/libbehaviortree_cpp_v3.so+0x6d733)
    #11 0x7f0152d79602 in BT::DecoratorNode::executeTick() (/opt/ros/foxy/lib/libbehaviortree_cpp_v3.so+0x6b602)
    #12 0x7f015301cb82 in nav2_behavior_tree::PipelineSequence::tick() /home/r1/ros2_clang_navigation/src/navigation2/nav2_behavior_tree/plugins/control/pipeline_sequence.cpp:39:39
    #13 0x7f0152d7b733 in BT::TreeNode::executeTick() (/opt/ros/foxy/lib/libbehaviortree_cpp_v3.so+0x6d733)
[... 16 lines cut ...]
    #30 0x7f01522ee8d8 in std::thread::_Invoker<std::tuple<nav2_util::SimpleActionServer<nav2_msgs::action::NavigateToPose, rclcpp::Node>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::NavigateToPose> >)::'lambda'()> >::operator()() /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/thread:251:11
    #31 0x7f01522ee5df in std::__future_base::_Task_setter<std::unique_ptr<std::__future_base::_Result<void>, std::__future_base::_Result_base::_Deleter>, std::thread::_Invoker<std::tuple<nav2_util::SimpleActionServer<nav2_msgs::action::NavigateToPose, rclcpp::Node>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::NavigateToPose> >)::'lambda'()> >, void>::operator()() const /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/future:1362:6
    #32 0x7f01522ee23e in std::_Function_handler<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> (), std::__future_base::_Task_setter<std::unique_ptr<std::__future_base::_Result<void>, std::__future_base::_Result_base::_Deleter>, std::thread::_Invoker<std::tuple<nav2_util::SimpleActionServer<nav2_msgs::action::NavigateToPose, rclcpp::Node>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::NavigateToPose> >)::'lambda'()> >, void> >::_M_invoke(std::_Any_data const&) /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/std_function.h:285:9
    #33 0x7f01543db2ff in std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>::operator()() const /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/std_function.h:688:14
    #34 0x7f01543dabd4 in std::__future_base::_State_baseV2::_M_do_set(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*) /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/future:561:27
    #35 0x7f01543db243 in void std::__invoke_impl<void, void (std::__future_base::_State_baseV2::*)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*>(std::__invoke_memfun_deref, void (std::__future_base::_State_baseV2::*&&)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*&&, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*&&, bool*&&) /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/invoke.h:73:14
    #36 0x7f01543dafa7 in std::__invoke_result<void (std::__future_base::_State_baseV2::*)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*>::type std::__invoke<void (std::__future_base::_State_baseV2::*)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*>(void (std::__future_base::_State_baseV2::*&&)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*&&, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*&&, bool*&&) /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/bits/invoke.h:95:14
    #37 0x7f01543daf18 in void std::call_once<void (std::__future_base::_State_baseV2::*)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*>(std::once_flag&, void (std::__future_base::_State_baseV2::*&&)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*&&, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*&&, bool*&&)::'lambda'()::operator()() const /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/mutex:671:4
    #38 0x7f01543dadf6 in void std::call_once<void (std::__future_base::_State_baseV2::*)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()> [... cut ...] ted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::NavigateToPose> >)::'lambda'()> >&&)::'lambda'()> > >::_M_run() /usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/thread:195:13
    #50 0x7f0151bd8de3  (/lib/x86_64-linux-gnu/libstdc++.so.6+0xd6de3)
    #51 0x7f01525d8608 in start_thread /build/glibc-eX1tMB/glibc-2.31/nptl/pthread_create.c:477:8
    #52 0x7f01518bb292 in clone /build/glibc-eX1tMB/glibc-2.31/misc/../sysdeps/unix/sysv/linux/x86_64/clone.S:95

```

SteveMacenski, 2021-08-19:
PR would be appreciated! Though in this case, I'm not sure why `if(!nullptr)` isn't evaluating and instead crashing. I'm pretty sure the boolean evaluation of a nullptr to false is in the C++11 standard. Is this still an issue in the `main` branch? That file's received a bit of attention in the last year and some non-trivial changes have been made to handle some particular crashing cases that may have been covering this. 

Just curious, you've brought up about 3-4 issues with solutions -- are you finding any more or just those few? Just wanting to know if I should try to find more of these issues or if you covered them.

easylyou, 2021-08-23:
Hi! I think this bug had been fixed in `main`. I tried several simple fixes in `foxy` but failed. It seems hard to fix in a few lines. I will close this issue.
My testing environment relies on gazebo simulation but the simulation is not supporting `Galactic` well now.
The found bugs related to nav2 is by fuzzing technology and most of them have been posted, so you probably needn't worry..

closed 2021-08-23 by easylyou