ros-navigation/navigation2 #2525: smac_planner_hybrid makes loops on short paths

Aposhian, 2021-08-19:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - rolling
- Version or commit hash:
  - main
- DDS implementation:
  - CycloneDDS

#### Steps to reproduce issue
```
--- a/nav2_smac_planner/test/test_a_star.cpp
+++ b/nav2_smac_planner/test/test_a_star.cpp
@@ -16,6 +16,7 @@
 #include <memory>
 #include <string>
 #include <vector>
+#include <iostream>
 
 #include "gtest/gtest.h"
 #include "rclcpp/rclcpp.hpp"
@@ -198,9 +199,10 @@ TEST(AStarTest, test_se2_single_pose_path)
   EXPECT_TRUE(a_star.createPath(path, num_it, tolerance));
 
   // Check that the path is length one
-  // With the current implementation, this produces a longer path
-  // EXPECT_EQ(path.size(), 1u);
-  EXPECT_GE(path.size(), 1u);
+  for (auto pose : path) {
+    std::cout << pose.x << ", " << pose.y << ", " << pose.theta << std::endl;
+  }
+  EXPECT_EQ(path.size(), 1u);
 
   delete costmapA;
 }
```

Then the output of colcon test is

```
12: [ RUN      ] AStarTest.test_se2_single_pose_path
12: 11, 10, 0                                     
12: 9.55402, 10.0124, 71.3608
12: 8.12518, 10.2228, 69.2893
12: 6.75744, 10.6866, 67.2178
12: 5.49539, 11.3888, 65.1462
12: 4.38014, 12.3064, 63.0747
12: 3.44805, 13.4096, 61.0032
12: 2.72949, 14.6624, 58.9316
12: 2.24789, 16.024, 56.8601
12: 2.01893, 17.4499, 54.7885
12: 2.05009, 18.8938, 52.717
12: 2.34035, 20.3086, 50.6454
12: 2.88024, 21.6481, 48.5739
12: 3.65217, 22.8688, 46.5024
12: 4.63098, 23.9307, 44.4308
12: 5.78478, 24.7994, 42.3593
12: 7.07596, 25.4465, 40.2877
12: 8.46243, 25.8509, 38.2162
12: 9.89901, 25.9994, 36.1447
12: 11.3389, 25.8872, 34.0731
12: 12.7351, 25.5179, 32.0016
12: 14.0422, 24.9037, 29.93
12: 15.2176, 24.0644, 27.8585
12: 16.2229, 23.0275, 25.787
12: 17.0254, 21.8267, 23.7154
12: 17.5989, 20.5012, 21.6439
12: 17.9248, 19.0943, 19.5723
12: 17.9924, 17.6516, 17.5008
12: 17.7995, 16.2203, 15.4292
12: 17.3525, 14.847, 13.3577
12: 16.6658, 13.5765, 11.2862
12: 15.7618, 12.4501, 9.21462
12: 14.6701, 11.5046, 7.14308
12: 13.4261, 10.7708, 5.07154
12: 12.0706, 10.2726, 3
12: /home/adam/github/aposhian/nav2_ws/navigation2/nav2_smac_planner/test/test_a_star.cpp:205: Failure
12: Expected equality of these values:
12:   path.size()
12:     Which is: 35
12:   1u
12:     Which is: 1
```

As you can see, the first pose is the goal, so it should just stop there, but it adds all these other poses, which if you plot are a loop.

If I change the model to Reeds-Shepp on this test, I get the following output
```
12: [ RUN      ] AStarTest.test_se2_single_pose_path
12: 11, 10, 0                                     
12: 14.1768, 10, 0
12: 12.0884, 10, 0
12: /home/adam/github/aposhian/nav2_ws/navigation2/nav2_smac_planner/test/test_a_star.cpp:205: Failure
12: Expected equality of these values:
12:   path.size()
12:     Which is: 3
12:   1u
12:     Which is: 1
```

#### Expected behavior
`AStarAlgorithm` should plan a single pose path to the adjacent costmap cell.

#### Actual behavior
`AStarAlgorithm` plans a loop after the goal when using the Dubin model, and plans a forward and backup step with Reeds-Shepp.

This is disastrous for `ComputePlanThroughPoses` when the goals are close together.

[image]

#### Additional information
It appears that the offending code is the path "refinement." When I remove this block, the issue is resolved. https://github.com/ros-planning/navigation2/blob/7b29e471098fa291e7bcf3223a4b5f42643b4296/nav2_smac_planner/src/a_star.cpp#L505-L526

Removing this will require a refactor of the other tests to expect different paths.

Aposhian, 2021-08-19:
Also, this is a blocker for implementing the proposed fix for #2517

SteveMacenski, 2021-08-19:
Well, this is where a couple of things are at play for very short paths. The 2 phases of what might happen are the analytic expansion can get a result first or searching. 

- 1) If A->B is really short and analytic expansion can find a solution, it may cause a loop because where your robot is is slightly off from where it would need to be for kinematic feasibility based on your minimum turning radius to just go straight and turn a little. So as a result it would have to loop back in order to achieve the exact requested goal. 

- 2) If that isn't actually what is getting the goal and instead its from search, it might actually be that the next cell is less than the minimum motion primitive length. In order for this algorithm to be robust for a broad range of radiuses, we dynamically set the motion primitives at the turning radius's length such that it must both leave the current 2D resolution cell (e.g. minimum of sqrt(2) in length) and also must move at least 1 angular bin, in whole numbered bin increments. So if your goal is, lets say 1 cell up and 1 cell to the left, in order to achieve that exact cell, you'd have to loop around since its shorter than the smallest primitive length if you've set your turning radius to something large. For very short paths though, it almost certainly isn't this since you're so close to the goal there's no reason that the analytic expansion wouldn't be the thing to immediately find a solution. So its pretty much always going to be (1) above unless your goals are in close proximity to obstacles making the loop-back pattern in a collision state. 

This is a known behavior that there's some looping on replans when in very close proximity to the goal for the 2 reasons described above. I'd absolutely love a workable solution to that, but to keep kinematic feasibility, I'm not exactly sure how that could occur.

---

What is that image I'm looking at? What are the green arrows / dashed lines? What the goals you requested being compounded with NavThroughPoses? What's the output look like based on your solution vs the initial issue?

That is interesting, that code is actually there to **prevent** looping in analytic expansions. Something we saw happening for "normal" analytic expansions (e.g. lets say ignoring the super short ones) a particular search position might be at X, Y at an angle 90 degrees from the goal. The shortest Dubin path to the  goal would be to rotate the robot in a circle to get the heading back to 0 deg relative and then drive straight. That would introduce a loop in the path at the end of the search expansion where it meets with the analytic expansion. This was added so that if we did find a solution, that we'd backtrack a few nodes and try again, and try again, and try again, until we could not find a workable solution. That way we could try to find the first time a valid analytic expansion is available to use which would (hopefully) not include a loop.

So we  want to keep that around since that solved a pretty real problem. But maybe it would be good to make a condition on smaller paths to either A) not backtrack or B) backtrack less aggressively. But for longer paths, do this. 

Thoughts?

Aposhian, 2021-08-19:
Sorry, should have clarified. The image is of a sequence of goals (green arrows) being provided to our robot via `NavigateThroughPoses`, and the plan output by smac_planner_hybrid (purple arrows).

Aposhian, 2021-08-19:
>That could also be prevented by making your path not based on the minimum turning radius but add some epsilon error to air on the side of conservative for your predefined route's turning radius if you want the points to be super close together.

Where is that `epsilon` configurable somewhere in the current code? We thought that being more lenient on orientation would help, but couldn't figure out how to do it.

SteveMacenski, 2021-08-19:
I’m saying that your code for generating the points for nav through poses would need to add some tolerance for tracking error. 

But we can also work on the backtrack solution too, but wont solve it 100% if you’re generating paths for NavThroughPose assuming a perfect robot system

Aposhian, 2021-08-23:
Not yet. We are considering some different options on the path forward (knee slap) with our team, but this is definitely still on our radar.

SteveMacenski, 2021-08-25:
Makes sense, I'd still be interested to see the changes you mention exist when you remove those lines of code. Even if you work around this problem by sampling larger increments between GPS waypoints, I do want to understand that issue since that will come up again with someone else.

SteveMacenski, 2021-09-14:
Closing from discussions in working group / slack discussions. The cause of this is a short, pre-recorded path from the robot. The frequency of sampling was high enough that the short distance making up the path segments cannot be found by a shorter analytic expansion (due to the settings of the planner being slightly lower than the actual vehicle recorded data).  The fix is to either sample less frequently, lower the planner's turning radius to be reflective of the real data your robot collected of its capabilities, or sample more frequently and instead use that as the path for the local trajectory planner to track. 

I would be curious to learn about how / why you say that removing the analytic refinement fixes the problem however and come up with a solution to that as well, but that's out of the scope of this ticket

closed 2021-09-14 by SteveMacenski