ros-navigation/navigation2 #2541: Costmap not being published when using custom params

marcomasa, 2021-08-30:
## Bug report

**Required Info:**

- Operating System:   Ubuntu 20.04
- ROS2 Version: Galactic (apt-get)
- Version or commit hash:
ros-galactic-nav2-amcl	1.0.6-1focal.20210701.002235
ros-galactic-nav2-behavior-tree	1.0.6-1focal.20210701.002952
ros-galactic-nav2-bringup	1.0.6-1focal.20210701.015912
ros-galactic-nav2-bt-navigator	1.0.6-1focal.20210701.004934
ros-galactic-nav2-common	1.0.6-1focal.20210630.232638
ros-galactic-nav2-controller	1.0.6-1focal.20210701.004401
ros-galactic-nav2-core	1.0.6-1focal.20210701.004136
ros-galactic-nav2-costmap-2d	1.0.6-1focal.20210701.002823
ros-galactic-nav2-dwb-controller	1.0.6-1focal.20210701.005251
ros-galactic-nav2-lifecycle-manager	1.0.6-1focal.20210701.002253
ros-galactic-nav2-map-server	1.0.6-1focal.20210701.002229
ros-galactic-nav2-msgs	1.0.6-1focal.20210630.235639
ros-galactic-nav2-navfn-planner	1.0.6-1focal.20210701.004410
ros-galactic-nav2-planner	1.0.6-1focal.20210701.004411
ros-galactic-nav2-recoveries	1.0.6-1focal.20210701.004929
ros-galactic-nav2-regulated-pure-pursuit-controller	1.0.6-1focal.20210701.004418
ros-galactic-nav2-rviz-plugins	1.0.6-1focal.20210701.014516
ros-galactic-nav2-smac-planner	1.0.6-1focal.20210701.004430
ros-galactic-nav2-util	1.0.6-1focal.20210701.001550
ros-galactic-nav2-voxel-grid	1.0.6-1focal.20210701.002338
ros-galactic-nav2-waypoint-follower	1.0.6-1focal.20210701.004452

- DDS implementation: default

 

#### Steps to reproduce issue
I am trying to get nav2 working with a custom simulation setup.
I modified the launchfiles from the turtlebot example and used them in the waffle sim.
Then i got my custom model loaded in gazebo & rviz2 and made a slam of my custom world.

After adjusting AMCL and the global planner plugin (SMAC), 
i cannot get the costmaps to show in rviz2.
Weird thing about this is, it seems its just not published, bc sometimes i can still get the robot to navigate.

nav_params.yaml
```
amcl:
  ros__parameters:
    use_sim_time: True
    alpha1: 0.2
    alpha2: 0.2
    alpha3: 0.2
    alpha4: 0.2
    alpha5: 0.2
    base_frame_id: "base_footprint"
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.9
    beam_skip_threshold: 0.3
    do_beamskip: false
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 2.0
    laser_max_range: 100.0
    laser_min_range: -1.0
    laser_model_type: "likelihood_field"
    max_beams: 600
    max_particles: 2000
    min_particles: 500
    odom_frame_id: "odom"
    pf_err: 0.05
    pf_z: 0.99
    recovery_alpha_fast: 0.0
[... 515 lines cut ...]

Subscription count: 1

Node name: rviz
Node namespace: /
Topic type: nav_msgs/msg/OccupancyGrid
Endpoint type: SUBSCRIPTION
GID: 86.49.10.01.34.f8.b9.fa.99.f0.6b.ad.00.00.3f.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  Durability: TRANSIENT_LOCAL
  Lifespan: 9223372036854775807 nanoseconds
  Deadline: 9223372036854775807 nanoseconds
  Liveliness: AUTOMATIC
  Liveliness lease duration: 9223372036854775807 nanoseconds

```

So the QoS settings are matching i guess,
but i try to find out the frequency nothing happens...

```
ros2 topic hz /local_costmap/costmap
1630363248.914410 [0]       ros2: using network interface enp5s0 (udp/10.0.7.21) selected arbitrarily from: enp5s0, docker0
```

SteveMacenski, 2021-08-31:
Please ask on ROS answers / search through issue trackers for rviz / QoS.

closed 2021-08-31 by SteveMacenski

myxiaojun, 2021-12-07:
i have the same problem of "Rviz2 not showing the costmaps",   @marcomasa  do you find a way to solve this problem?

marcomasa, 2021-12-07:
@myxiaojun you need to launch everything in one launchfile.
Idk why, but starting everything seperately does NOT work for me..
You also need to check your QoS settings, but that does not help if you dont start everything at once (see turtle nav examples)

myxiaojun, 2021-12-08:
I use "ros2 launch turtlebot3_bringup robot.launch.py" to launch robot, use "ros2 launch turtlebot3_navigation2 navigation2.launch.py map:=$HOME/map.yaml" to launch nav2.. but the costmaps still not show.

marcomasa, 2021-12-09:
I think u need to launch rviz with the nav stack in a single file.
I dont fully understand the new launchfile backend, but i believe some values are not shared when
launching from seperate files (this is also kinda the intention of the new system?)

myxiaojun, 2021-12-14:
I once start nav2 by "ros2 launch turtlebot3_navigation2 navigation2.launch.py use_sim_time:=True map:=/mnt/ros2/src/map.yaml".
and  I solved this problem by delete param "use_sim_time:=True".