ros-navigation/navigation2 #2545: [Bug] The goal sent to the planner is off the global costmap (even though it's not)

XDGFX, 2021-09-06:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04.2 LTS (`ros:foxy` docker container)
- ROS2 Version:
  - Foxy from `ros:foxy` container (I assume binaries?)
- Version or commit hash:
  - 0.4.7-1focal.20210701.142448
- DDS implementation:
  - I think Fast-DDS. Running host networking on all containers, and sharing PID and /dev/shm to allow communication

#### Steps to reproduce issue
Enable rolling global costmap
```yaml
global_costmap:
  global_costmap:
    ros__parameters:
      update_frequency: 1.0
      publish_frequency: 1.0
      global_frame: map
      robot_base_frame: base_link
      use_sim_time: False
      robot_radius: 0.15
      resolution: 0.05
      track_unknown_space: true
      rolling_window: true
      width: 6
      height: 6
      origin_x: -3.0
      origin_y: -3.0
```

Move robot outside original global costmap window
[image]

Attempt to navigate to a reasonable point
[image]

#### Expected behavior
The Nav2 stack receives the request and calculates a valid path

#### Actual behavior
The Nav2 stack receives the request but throws an error:
```log
[planner_server-2] [WARN] [1630919499.639906111] [planner_server]: The goal sent to the planner is off the global costmap. Planning will always fail to this goal.
[planner_server-2] [WARN] [1630919499.640326373] [planner_server]: GridBased: failed to create plan with tolerance 0.50.
[planner_server-2] [WARN] [1630919499.640414502] [planner_server]: Planning algorithm  failed to generate a valid path to (-7.29, 15.41)
[planner_server-2] [WARN] [1630919499.640629577] [planner_server_rclcpp_node]: [compute_path_to_pose] [ActionServer] Aborting handle.
```

SteveMacenski, 2021-09-07:
I believe this is where the false is being returned to case that https://github.com/ros-planning/navigation2/blob/0b92a11ded3fd27183d5bae815e0d717d1f1da98/nav2_navfn_planner/src/navfn_planner.cpp#L495 which is odd if the global costmap is actually valid. that is the same formulation of `worldToMap` this is powering the controllers that use a local rolling costmap as their defacto-standard formulation, so I don't think that's the issue.

Further, you're only seeing the error for the goal pose, not the robot pose which is fine before this request, so that might be something worth looking into to see if you ask for goals in different areas or the robot moves around that the planner can give you the other error as well (that may help figure out what the issue is or be able to better describe when this is an issue, after what parts of the costmap roll past the original window, etc). 

A PR would be appreciated as you're debugging. But I see no obvious problem at this moment to offer better advise. We need more information

XDGFX, 2021-09-07:
I realised the `origin` config was left from before I set a rolling global costmap; it seemed that the costmap would only generate around the sensor points, which for a forward-facing depth camera would often result in the robot being outside the costmap and the robot pose error you mention.

I'll try without the origin config (works fine with rolling costmap). I'll see if the error returns or if I can produce it repeatably.

XDGFX, 2021-09-13:
The origin config didn't fix it... however - 

Would I be correct in thinking that the `/compute_path_to_pose` action expects a pose in either `map` or `odom` frame (I'm finding it hard to tell looking at the code, my local costmap global frame is `odom`, and global costmap frame is `map` if that makes a difference)?

As it wants a stamped pose I assumed it would convert between frames as needed so I was giving pose requests in `base_link` frame, but after some testing it seems to not be taking that into account?

This would explain the issue I was having, as the goal would be outside the costmap if read in `map` or `odom` frame.

SteveMacenski, 2021-09-13:
The planner server should use whatever frame is set in the costmaps, but it would be good to check that in the planner server to see if you can catch somewhere where that may not be true. But if you set the start pose, it does use that as is https://github.com/ros-planning/navigation2/blob/main/nav2_planner/src/planner_server.cpp#L274 otherwise it gets the pose from the costmap which should be in whatever global frame you set in the global costmap

XDGFX, 2021-09-23:
Ok, yep that was definitely the issue. I wasn't setting start pose so that would explain the problem. Thanks!

closed 2021-09-23 by XDGFX