ros-navigation/navigation2 #2552: Refactored amcl code to include motion models as plugins

poornimajd, 2021-09-07:
----
## Feature request
Refactor the amcl code to make it more modularized to enable more flexible testing of different motion models.
#### Feature description

Currently the motion models used in AMCL are mainly restricted to the default models-  Differential and omni model.In case we want to customize the motion model to a particular robot, say try to implement a different variant of differential model or a new model (example constant velocity model) which has the same parameters as the default models,we may have to change the source code directly. In case if the motion model is modularized as a plugin,we can just change or add the motion model as a plugin,and hence without messing up with the amcl source code itself,enable different types of motion model testing.This may enable quick development and testing of motion models.
#### Implementation considerations

The feature can be implemented as a plugin.Linking a pull request here.

SteveMacenski, 2021-09-07:
AMCL is in nearly pure maintenance mode, it is my hope / intent that within the next 18 months I'll have some time to really hammer into improving the localization systems, but for now AMCL is the legacy option we have around. As such, generally I have put a stop to any major changes to AMCL that may even remotely cause issues for folks since its behavior should be the same as in ROS (1). 

But this seems reasonable in concept, though your PR needs some serious changes to be mergable. You need to largely revert your file reorganization to keep things relatively the same. You should use the **same** files that the motion models are defined in now and just update their classes to derive from the base class and register as plugins. That means a total removal of the `amcl_plugins` package you created. Your goal in that PR should be to make as **few** changes as critically required to get what you need done, done to be as disruptive as possible. 

I'd ordinarily relish the opportunity to rearchitect a package to be more general, but like I said, AMCL is a special case as a legacy package. Our primary goal is maintenance and stability on that package over all else. My hope is it won't even be in the stack in 2 years.

poornimajd, 2021-09-08:
Thank you for the detailed and speedy reply.

> AMCL is in nearly pure maintenance mode, it is my hope / intent that within the next 18 months I'll have some time to really hammer into improving the localization systems, but for now AMCL is the [...]

Yes this seems reasonable.

> But this seems reasonable in concept, though your PR needs some serious changes to be mergable. You need to largely revert your file reorganization to keep things relatively the same. You should use [...]

I do understand and agree with this,may be this way of minimal reorganizing could also serve the purpose.But based on my understanding, if I refactor it according to the suggestion here,may be the next time I need to add a new motion model I will have to change the amcl source code,and hence the purpose of refactoring may not be fully satisfied.

The only reason I made it as a separate package was to make sure that the next time to add a new model,I need to change the code only in amcl_plugins and not touch the nav2_amcl package.But yes this would again add another package overhead.

Thanks for your suggestion, and may be this issue can be addressed in the future as required,also in a improved manner.

SteveMacenski, 2021-09-08:
> may be the next time I need to add a new motion model I will have to change the amcl source code

Nope, if you create a plugin interface (e.g. header file for a general motion model), you can always include that header in your new package for a new plugin motion model type. The fact that there happen to be other implementations within the package does not limit your ability to create new ones externally. For example, see Recovery, Costmap2D.

SteveMacenski, 2021-09-14:
I'm going to close the PR to keep it on things that need review, feel free to reopen a new PR based on this discussion's results

poornimajd, 2021-09-15:
Sure,thank you for your time and feedback,I will try to stick to these suggestions in the future.

poornimajd, 2021-10-21:
A quick update,I have opened a new PR for the modularization of the motion models.I have tried to follow most of the suggestions you had given last time,to keep the number of changes to a minimum.
I have created the motion models - differential and omni models as a plugin and made the necessary changes to the other files so that the plugins can be loaded.

I am open to any new changes/suggestions I need to make to the PR.
Thank you.

closed 2021-11-11 by SteveMacenski