ros-navigation/navigation2 #2566: [Costmap2DROS] dynamic parameters

doisyg, 2021-09-16:
## Feature request
I have a personal interest in making the parameter `robot_radius` of `Costmap2DROS` dynamic (it is already in my fork). I could make a PR with only this parameter dynamic but this could also be the opportunity of making the others dynamic too.
Out of the following:
https://github.com/ros-planning/navigation2/blob/e5a4c844c748233dc4d3039b1ea5809c81dac45f/nav2_costmap_2d/src/costmap_2d_ros.cpp#L90-L116
which one should we make dynamic ?

SteveMacenski, 2021-09-16:
It is already possible, the footprint subscriber is there to update based on dynamic changes to a robot's size https://github.com/ros-planning/navigation2/blob/e5a4c844c748233dc4d3039b1ea5809c81dac45f/nav2_costmap_2d/src/costmap_2d_ros.cpp#L176-L179. You would have to specify the circle yourself though on publication (e.g. polygon that makes a circle) rather than just defining a radius, but you can use the costmap_2d function `makeFootprintFromRadius` in your application to generate it. 

In terms of which of these should be dynamic, I'd use ROS1 as a starting point https://github.com/ros-planning/navigation/blob/noetic-devel/costmap_2d/cfg/Costmap2D.cfg and use your best judgement for the rest.

SteveMacenski, 2021-10-07:
@doisyg should this be rolled into the main dynamic parameters ticket?