ros-navigation/navigation2 #2578: Can't configure odom topic for controller_server

Darkproduct, 2021-10-02:
When looking at my setup, I found one node that still subscribed to `/odom`, but I'm using `robot_localization` with the filtered odom topic `/odometry/filtered`

The node still subscribed was the `controller_server`. Is there any reason why this node should get the raw odom topic and not the filtered one?

It looks like there is no way of changing the odom topic from the parameters:
https://navigation.ros.org/configuration/packages/configuring-controller-server.html#parameters

Like with the `bt_navigator`: 
```
bt_navigator:
  ros__parameters:
    odom_topic: /odometry/filtered
```

I can rename the topic, but it is a bit odd that this is not consistent. remapping - parameters?

Darkproduct, 2021-10-02:
Ok. I found out that the `controller_server` is using the [`odom_subscriber`](https://github.com/ros-planning/navigation2/blob/foxy-devel/nav2_dwb_controller/nav_2d_utils/include/nav_2d_utils/odom_subscriber.hpp) which defines its own `odom_topic` parameter. So the this should work exactly the same as the `bt_navigator`.

Darkproduct, 2021-10-02:
And yes, this works. Tested in ROS2 Foxy. 

Then this is just an undocumented parameter. 

The only thing still missing:

> Is there any reason why this node should get the raw odom topic and not the filtered one?

SteveMacenski, 2021-10-04:
Adding a PR to our documentation website would be great, thank you!

>   Is there any reason why this node should get the raw odom topic and not the filtered one?

If estimating the current speed based on a rolling buffer of recent measurements, the raw odometry should be fine. Fusion helps longer term odometry consistency with reality. But for taking 5 measurements over the course of 0.1 seconds, it doesn't really matter. In fact, raw wheel odometry will give you a more current responsive velocity than fusion which will be tracking the target. 

But as you said, its still a parameter for you to be able to change at your will. But I would be surprised if changing it to the fused odometry improved anything.