ros-navigation/navigation2 #2581: Loading of the saved map

nedo99, 2021-10-05:
Hi,

I am trying to load the saved map. I saved the map using `ros2 run nav2_map_server map_saver_cli -f ~/map`. When I try to load it using `bringup_launch.py` passing `map` parameter, the global costmap is not updated.

## Question

**Required Info:**

- Operating system:
  - Ubuntu 20.04
- ROS Version:
  - Foxy
- SHA:
  - `foxy-devel`

#### Steps to reproduce issue
Using a real-time robot, navigate manually through the room until it is fully mapped. Then save the map using the command above.

Start the environment (robot driver, camera, SLAM, and nav stack) again, but this time using `bringup_launch.py` launch file and passing the `map` parameter with saved map.  

#### Expected behavior

Global costmap is fully loaded and the current robot location matches the one shown in the Rviz. If not, use `2D Pose Estimate` in Rviz set to the current position. 

#### Actual behavior

Only the area around the robot (within camera range) is visible in the `/map` and global cost map topic. The rest is unknown. Moreover, if I disable the SLAM  `/map` topic (remap it to something else, then I see the full global costmap, but the location of the robot is at the wrong location. If I try to set it via `2D Pose Estimate` in Rviz, it starts flickering between the location from SLAM and the one from our estimate.

When I test this using gazebo simulation it works fine with running SLAM, but in the real-world environment, I simply cannot get it to work.

What am I missing?

Regards,
Nedim

SteveMacenski, 2021-10-05:
Please ask on ROS Answers or search for other answers on ROS Answers.

closed 2021-10-05 by SteveMacenski

nedo99, 2021-10-05:
> Please ask on ROS Answers or search for other answers on ROS Answers.

Already did, but no reply. Could not find any examples or documentation that is not related to gazebo simulation.