ros-navigation/navigation2 #2596: Add default BT logic for patience near goal

SteveMacenski, 2021-10-13:
If the robot is "close" to the goal and its path is blocked:
- Wait some time to see if it becomes unblocked
- Then replan to the goal
- repeat N times

The general idea is in blocked aisleways in warehouses / retail stores or general navigation where something temporary is blocking the robot's path. If in a narrow aisle and something in temporarily in the way, don't go out of the aisle, up another, and around, simply wait a hot second to see if it clears up (someone getting something in the aisle, transient leaving the aisle, picker, forklift, etc) 

CC  @DLu

padhupradheep, 2021-11-04:
Implemented the first version of this feature, Added a `obstacle_clearance_time` parameter to the `FollowPath` BT-Node, which would eventually allow us to specify the waiting time, if we see the obstacle. Below, we can see 2 possible scenarios:

1.  Obstacle does not clear at all, with `obstacle_clearance_time` to be 3 seconds: 
[image]

2. Obstacle clears (I just delete the obstacle in gazebo - Laziness at it's peak) and you can see the robot pass through the (could have been ideally the) same path:
[image]

Thoughts?

SteveMacenski, 2021-11-09:
That looks pretty good to me! What I had in mind was to create a new Behavior Tree node plugin that would do this instead, but implementing in the controller server could work as well. I think the benefit of it in the behavior tree is more configurability (e.g. be able to use the primitive node you create in other contexts than this) and would be a bit more "structured" to the autonomy application to have an explicit pause <if condition is met>. Our example is closness to goal, but someone could manipulate a BT node to set other conditions for the robot to pause (like in a dynamic setting).

padhupradheep, 2021-11-09:
> What I had in mind was to create a new Behavior Tree node plugin that would do this instead, but implementing in the controller server could work as well. I think the benefit of it in the behavior tree is more configurability (e.g. be able to use the primitive node you create in other contexts than this) and would be a bit more "structured" to the autonomy application to have an explicit pause . Our example is closness to goal, but someone could manipulate a BT node to set other conditions for the robot to pause (like in a dynamic setting).

I understand your point, but irrespective of where the logic for pausing is, you need to anyways send a signal to the controller to send a zero velocity. That is the reason why I decided to have the logic in the pre-existing BT-Node, so that we do not have to make a redundant BT with just a pausing functionality. But, I will certainly give some more thoughts on your point, for having a separate BT. 

> But I'd like to see the implementation details of what you showed above, maybe it would work in it of itself. I'm not 100% sure how it is you implemented it. For all I know, what I suggested is basically what you did in the controller server.

Sure, here is the [commit](https://github.com/padhupradheep/navigation2/commit/e6b48063c84a8199e34a0b3507f14aafcc36693c) in my fork which has the implementation details.

 I currently do not consider the "significantly longer" path, rather I just say if the path is "just longer" then continue with the wait logic.

SteveMacenski, 2021-11-09:
> you need to anyways send a signal to the controller for sending a zero velocity

That is a good point, I did not consider that. I know it wasn't in the original prompt for this work, but I've long since thought  that we need to have `CancelXYZ` actions (e.g. plan, control, smooth, etc) so that different parts of the BT can kill the task servers if needed. This application kind of begs for that if we do it in the BT. 

The way I view this is that the "pause if close to goal and path significantly increases" is pretty "business logic"-y so it wouldn't make sense to put in the task server for every application. By putting it in the BT, people could strip it out if it didn't make sense for them, since the BT is the customizable front-end for use-case specific navigation logic. Even if software-wise its easier to put in the controller server, I would be concerned about the slippery-slope argument that feature creep keeps adding more and more semi-general semi-use-case-specific features into the server because its easier. 

What's nice about the BT is that if we make these CancelXYZ, Wait, and Path Longer/Close nodes, these are all really reusable for other applications, so none of your work would go to waste. In fact, I think there's even more tangential value in doing it this way to add more primitives to the behavior tree vocabulary for people to leverage later on. I can see all of those nodes having many more uses, and at some point, I think all of them would need to be added here anyway for other applications. Its a little more work on the front end, but then can be used in many places in the future, especially the Wait and CancelXYZ nodes.

padhupradheep, 2021-11-10:
>  But I think the project would reap many more benefits from taking the longer way around it (and you'd get to have your name enshrined as the author of multiple plugins in the documentation https://navigation.ros.org/plugins/index.html wink ).

That is surely something, which I will not deny ( A good way to lure :P ). Given the feature is not an immediate requirement, I can work on the long way around, like as you said. 

I agree on all your other points. I just wonder if we really need to kill the task servers or there maybe another way around to do the CancelXYZ. Because, it might also require another BT to restart the task servers. But, anyways, I will try to figure out the implementation details!

SteveMacenski, 2021-11-10:
> A good way to lure :P

I know, why do you think I added that field to the table :wink: 

Well if you cancel control in the patience subtree, when it leaves it, itll replan and then the plan will be sent back to the follow path BT node, so it would essentially autostart by itself once leaving this subtree dealing with pausing, no?

There are certainly ways you could use Cancel that would cause you serious issues, but as long as its used in a temporary / failure state recovery branch of the tree that later returns to the main one, it should be OK. If you include it in the main tree, then you probably won't be

padhupradheep, 2021-11-24:
FYI, I will be working on this feature this week, in case if you are wondering about it!

padhupradheep, 2021-12-02:
Small update - done with the BT-Plugin for:

- `CancelServer`
- `IsPathLonger`

Just `IsPathShorter` is left! Hope to open a PR in this weekend ! 

Note: took me a little time to get use to the other different BT Terminologies. Documentations helped big time!

padhupradheep, 2021-12-08:
Sorry for the delay. I have implemented 3 BT's that includes: 

- IsNewPathLonger (conditionBT)
- IsNewPathShorter (conditionBT)
- CancelControl

With that, if someone wants to implement the above logic. They can have the following BT:
```
<root main_tree_to_execute="MainTree">
  <BehaviorTree ID="MainTree">
    <RecoveryNode number_of_retries="6" name="NavigateRecovery">
      <PipelineSequence name="NavigateWithReplanning">
        <RateController hz="1.0">
          <RecoveryNode number_of_retries="1" name="ComputePathToPose">
            <ComputePathToPose goal="{goal}" path="{path}" planner_id="GridBased"/>
            <ReactiveFallback name="ComputePathToPoseRecoveryFallback">
              <GoalUpdated/>
              <ClearEntireCostmap name="ClearGlobalCostmap-Context" service_name="global_costmap/clear_entirely_global_costmap"/>
            </ReactiveFallback>
          </RecoveryNode>
        </RateController>
        <IsNewPathShorter path="{path}" length_factor="1.0"/>
        <RecoveryNode number_of_retries="1" name="FollowPath">
          <FollowPath path="{path}" controller_id="FollowPath"/>
          <ReactiveFallback name="FollowPathRecoveryFallback">
            <GoalUpdated/>
            <ClearEntireCostmap name="ClearLocalCostmap-Context" service_name="local_costmap/clear_entirely_local_costmap"/>
          </ReactiveFallback>
        </RecoveryNode>
      </PipelineSequence>
      <ReactiveFallback name="RecoveryPatienceFallback">
        <Sequence name="CancelingControlAndWait">
          <CancelControl name="ControlCancel" service_name="/controller_server/change_state"/>
          <Wait wait_duration="5"/>
        <Sequence name="CancelingControlAndWait">
      </ReactiveFallback>
    </RecoveryNode>
  </BehaviorTree>
</root>
```

What do you think? does it look elegant?

SteveMacenski, 2021-12-09:
A BT visualization using either groot or the behavior tree visualization script in `tools/` would be helpful. XML is tricky to parse by eye. 

What I dont see here is only executing this type of behavior when the robot is near the goal. I think with that built in, it may make sense to make some kind of combined BT node for specifically this purpose instead of many small ones (eg is path length much longer + near goal as 1 node). I’m not sure about the shorter one being needed- but maybe it is if I saw the full tree more visually. I don’t discount my lack of skills reading XML 😆

padhupradheep, 2021-12-10:
>  don’t discount my lack of skills reading XML laughing

Alright, I'm adding the image. This is way better than the XML ! 

> What I dont see here is only executing this type of behavior when the robot is near the goal.

What qualifies the robot to be near the goal? What is the distance? That is the main reason why I decided to define a `length factor`, that would allow the user to decide on how long the length of the allowed length of the new path should be. Say if the length of the new path is just 5 cm longer the the new one, which is tolerable for the users, then he can just allow the robot to go through. There is no special reason to use the `length_factor` apart from it being a simple multiplier. I could also change it to just `tolerable_new_path_length`. 

> I’m not sure about the shorter one being needed- but maybe it is if I saw the full tree more visually.
Here from the image below, it can be noted that, I use `IsNewPathShorter`, if this condition fails (which apparently means the path is longer), then we just go into the temporary fallback branch. 

The reason, I decided to implement  `IsNewPathLonger` is because of the fact that, it could be another useful primitive for the users for some other application. There is no other specific reason. 

[image]

SteveMacenski, 2021-12-13:
> What qualifies the robot to be near the goal?

Euclidean distance, integrated path length to the goal, etc. I would think integrated path distance as the best metric (which you can get as [feedback from the BT nav server](https://github.com/ros-planning/navigation2/blob/main/nav2_bt_navigator/src/navigators/navigate_to_pose.cpp#L134) no less! We could store that in the BT blackboard for use). It would be a parameterized distance for the user to set about what proximity to the goal would make sense for them to try to commit to that action before doing the 'long way around'. 

We probably want the robot to replan and go around if not near the goal. Even if a user didn't, they could simply set the distance to 1e999 so that it always happens if they like. Simply measuring the path length isn't a good metric because if you plan infrequently then it could be shorter / longer due to motion to be had during that duration, since we're replanning from the new robot's pose, not the original pose. The distance from something, like the goal, would be a more stable metric that is less dependent on replanning poses.

The distance is also then a physical value they can derive from requirements, rather than a "factor" which is more adhoc tuning. 

---

As a separate comment, I would like to see this feature integrated into the default BT so we're compounding features. We should still have a BT like what you show above available of **just** this feature in implementation for reference, I'd like the default BT also updated to have this logic. 

The logic flow, I think (which is definitely not set in stone, just my thoughts) should be something like
- Normal planning / replanning branch as you show
- BT node to check if path distance to goal is within parameterized threshold to trigger behavior & path is "significantly" longer
- If so, cancel control and wait (in retry node so it can be done N times by user parameterization). At end of N times, allow to pass path to controller and continue.
- If not, pass path to controller

So I think the cancel / wait logic should be below the BT node to check if path dist to goal / length. So you could make that a decorator instead of an action so that all of this behavior appears under 1 concise branch

padhupradheep, 2021-12-14:
> Euclidean distance, integrated path length to the goal, etc. I would think integrated path distance as the best metric (which you can get as feedback from the BT nav server no less! 

That is how I do it already, I just conveyed it to you in a very wrong manner I think. I already use the `nav2_util::geometry_utils::calculate_path_length` . So my confusion was more to the point if we want to use a length_factor or the actual_distance from the goal. This is now clear. 

So now, the point remaining is to change the `cancelControl` action node into a decorator node, and bring it under a single branch. 

Just one question, does the name of the Behaviors are fine ? or should I give it some more thoughts on it?

SteveMacenski, 2021-12-14:
I think they’re fine!

padhupradheep, 2022-01-03:
Just giving you a small nudge @SteveMacenski for this issue!

SteveMacenski, 2022-01-03:
The BT looks good, but I think you're missing the check to only run when in proximity to the goal? It could be stripped out if they would rather do that always, but it would be good to have that basic ability as the primary mode. 

For testing, you can see in `nav2_behavior_tree` package there's a `test` directory. Look at the tests for the same types of nodes as you've created and make tests that mirror their intent for your new behavior tree nodes. For linting, just run `ament_cpplint` `ament_uncrustify` and `ament_xmllint` and fix the issues they show, if any.

padhupradheep, 2022-01-04:
> The BT looks good, but I think you're missing the check to only run when in proximity to the goal?

This is indeed taken care, by the `IsPathLonger` node. The `path_length` in the BT node allows the user to set the desired goal proximity. By default the value will be set to 1e999. 

Maybe the name `path_length` should be changed!

SteveMacenski, 2022-01-04:
Yeah, I think a rename would be good!

SteveMacenski, 2022-03-21:
@padhupradheep how do you feel about adding a pause into the contextual recoveries in the BT, not just in the global recoveries subtree (e.g. where we have costmap clearing only options for global/local planner failures)? I think that could make sense for especially the local planner. if it fails, pausing for a moment might be helpful so the robot isn't so ADHD. 

What do you think? 

@jwallace42 @naiveHobo  what do you think about that?

padhupradheep, 2022-03-22:
I understand the behavior that you want to achieve, but pausing is what I started to develop initially for this particular issue on the whole. It requires parallelism I think, but I would like to take some time to think about it as a whole. Also, I know that it's every BT developers wish. 

On researching for it before, I came across this interesting [comment in the BehaviorTree.CPP repo](https://github.com/BehaviorTree/BehaviorTree.CPP/issues/239#issuecomment-719421963) for a similar issue. I think, this would be a kind of a work around solution for this particular issue.

SteveMacenski, 2022-03-30:
Merging imminent, closing.

tugbakara, 2023-07-13:
> Implemented the first version of this feature, Added a `obstacle_clearance_time` parameter to the `FollowPath` BT-Node, which would eventually allow us to specify the waiting time, if we see the obstacle. Below, we can see 2 possible scenarios:
> 
> 1. Obstacle does not clear at all, with `obstacle_clearance_time` to be 3 seconds:
> 
> [image] [ [image] ](https://user-images.githubusercontent.com/20242192/140329028-bc642b6a-10ba-49b9-93a1-a346cc592a4f.gif) [ ](https://user-images.githubusercontent.com/20242192/140329028-bc642b6a-10ba-49b9-93a1-a346cc592a4f.gif)
> 
> 2. Obstacle clears (I just delete the obstacle in gazebo - Laziness at it's peak) and you can see the robot pass through the (could have been ideally the) same path:
>    [image]
>          
>            [
>              
>                [image]
>              
>            ](https://user-images.githubusercontent.com/20242192/140328869-e09664f7-d7e3-4fc6-948b-dc7e02ccd0d6.gif)
>            
>            
>              
>                
>                  
>                
>                
>                  
>                  
>                
>              
>              [
>                
>                  
>                
>              ](https://user-images.githubusercontent.com/20242192/140328869-e09664f7-d7e3-4fc6-948b-dc7e02ccd0d6.gif)
> 
> Thoughts?
> 
> **Edit:** We can also just hardcode, the waiting value. But I think it is better to have a parameter for the user to decide on how long should the robot wait for the the obstacle clearance.

is it applicable in ros melodic side with this [repo](https://github.com/miccol/ROS-Behavior-Tree/tree/master)? I have searched about it but actually nothing I found to implement this. Do you have any idea?

SteveMacenski, 2023-07-13:
There is zero support for ROS 1 here.

SteveMacenski, 2025-08-07:
Yes, please see the configuration guide for the BT navigator for the appropriate plugin for the default BT XML path.

osmantalhauslu, 2025-08-07:
I couldn’t find the specific BT XML I wanted in the files provided in the link (https://github.com/ros-navigation/navigation2/tree/main/nav2_bt_navigator/behavior_trees), and I’m not sure if it’s necessary to locate the default XML file. If I want to add custom behavior to the default system, can I use the following configuration?

bt_navigator:
  ros__parameters:
    global_frame: map
    use_sim_time: true
    robot_base_frame: base_link
    odom_topic: odom
    bt_loop_duration: 10
    filter_duration: 0.3
    default_server_timeout: 20
    wait_for_service_timeout: 1000
    introspection_mode: "disabled"
    navigators: ["navigate_to_pose", "navigate_through_poses"]
    navigate_to_pose:
      plugin: "nav2_bt_navigator/NavigateToPoseNavigator"
      bt_xml: "../my_package/custom_bt.xml"
      enable_groot_monitoring: false
      groot_server_port: 1667
    navigate_through_poses:
      plugin: "nav2_bt_navigator/NavigateThroughPosesNavigator"
      enable_groot_monitoring: false
      groot_server_port: 1669
    error_code_name_prefixes:
      - assisted_teleop
      - backup
      - compute_path
      - dock_robot
      - drive_on_heading
      - follow_path
      - nav_thru_poses
      - nav_to_pose
      - spin
      - route
      - undock_robot
      - wait

However, when I try to start with this configuration, I don’t notice any difference in behavior from the default settings. What configuration should I use to integrate these two systems together?