ros-navigation/navigation2 #2600: Create reference implementation of multi-robot centralized planning & local refinement systems

SteveMacenski, 2021-10-13:
- Central N robot path planner
- For each N agent, refine based on current state of the world / agents around it (ORCA-D-esk) 

There are several problems to handle prior to evaluation of the issue itself
- How do we communicate each robot's pose and goal? 
- How do we communicate each robot's world view, TF, and other related data (or do we)?
- How useful would this be in practice? Would companies / groups use this itself or just as a first-order starting point before building a custom solution?
- Once a path is at an agent, does it refine that path to follow in the controller or in a local-shim-planner plugin and leverage something like OCRA-D for in intermediary refinement of the centrally developed path before going to a controller?

padhupradheep, 2022-03-30:
Found a really nice github package: https://github.com/atb033/multi_agent_path_planning 

Maybe useful, while considering any sort of implementation.

SteveMacenski, 2022-06-02:
Perhaps the local refinement of the global centralized paths could use decentralized approaches just to refine a bit in the local area

Ryujiyasu, 2022-06-02:
It is a package that moves multiple robots that I know.
I'm glad if you can use it as a reference.

https://github.com/open-rmf/rmf_demos

CiaranYoung, 2023-12-05:
great