ros-navigation/navigation2 #2622: [RPP] When NavThroughPose has an intersecting loop, it can occasionally pick the wrong path segment

SteveMacenski, 2021-10-18:
I suspect its for one of a few reasons
- If it does "skip", its because the path is fully contained within the local costmap, so the pruning doesn't get rid of the future-looping section and then the "cross" point later in the path is technically closer to the lookahead requested point than the next sequential point based on some very small amount closer (<1mm)
- The fact that it jumps to that part once the robot hits that "cross" point makes me think its actually an issue in transforming the path, where the point closest to the robot becomes the other side of the "cross" and so it uses that (<1mm).

There's a couple of potential fixes:
- When we [find the points closest to the robot to start transforming the path from](https://github.com/ros-planning/navigation2/blob/main/nav2_regulated_pure_pursuit_controller/src/regulated_pure_pursuit_controller.cpp#L590-L595), there should be some tolerance, like 0.01 or something, that if there are multiple points within that distance range of each other, use the first in the path rather than the closest one. That should help it pick the right part of the cross
- Rather than only [checking when it leaves the costmap to prune til](https://github.com/ros-planning/navigation2/blob/main/nav2_regulated_pure_pursuit_controller/src/regulated_pure_pursuit_controller.cpp#L598-L602), also add a path distance element, once the path length from the closest pose is larger than N meters, also prune past those points. That way if there's a cross in the local costmap, as long as we pick the first point, we should prune out the rest

I suppose [this part](https://github.com/ros-planning/navigation2/blob/main/nav2_regulated_pure_pursuit_controller/src/regulated_pure_pursuit_controller.cpp#L625) in erasing the passed parts should be a solution if there is no replanning (so we don't infinitely loop cauased by selecting the "first" part of the intersection in the path) if there's no replanning. So I think doing those 2 points should cover that edge case completely.

SteveMacenski, 2021-10-18:
https://user-images.githubusercontent.com/14944147/137816896-fca413cb-5eaa-482c-81e1-167f30d861c7.mp4

KavenYau, 2021-10-19:
What about using accumulation distance rather than the radius?

https://github.com/ros-planning/navigation2/blob/8c80af04b92ef537b8fc74e7f8f3b07f869a587c/nav2_regulated_pure_pursuit_controller/src/regulated_pure_pursuit_controller.cpp#L377-L380

SteveMacenski, 2021-10-19:
> accumulation distance

Can you elaborate?

SteveMacenski, 2021-12-07:
@mradams-bastian writes:  

If I understand everything correctly, I would argue that this is not only a problem in how controllers handle overlapping paths, but it is also a consequence of the default goal checker used by the controller server being the Simple Goal Checker which determines that a goal has been reached solely based on whether the current pose of the robot matches the final pose of the path. If the final pose of the path happens to overlap the trajectory of the robot, the loop in controller_server.cpp ends at that moment and publishZeroVelocity() is called.

Maybe it would make more sense to have the Stopped Goal Checker be the default? Or even create a more sophisticated goal checker?

mradams-bastian, 2021-12-07:
Would it be possible to additionally check if that pruned segment crosses or overlaps itself, and if so, only send the path up to the point of intersection?

SteveMacenski, 2021-12-07:
I don't think that would be very robust, especially in the colinear situation where there may be several overlaps close to each other or large sections of pure overlap

mradams-bastian, 2021-12-08:
What are some example situations you're thinking of where it might fail? I am imagining that while traveling overlapping sections, the first time the robot passes over the section, the controller would not even realize it is an overlapping segment because the portion of the path that wraps back around and overlaps will have been pruned. When it passes back over the overlapping section a second time, the completed portion of the path will have been [pruned away](https://github.com/ros-planning/navigation2/blob/3e31f5ad7f1e570c1d751c8944c9b61a7a1578f2/nav2_regulated_pure_pursuit_controller/src/regulated_pure_pursuit_controller.cpp#L625), so once again it will not realize that there was an overlap to start.

The biggest potential issue I can think of that if a path makes a 180 degree turnaround, the 'return' portion of the path would not become visible until the robot reaches the very end of the 'to' portion of the path, and as a consequence it would stop at the end of the 'to' portion of the path with no more path to travel before the new segment becomes available immediately after. The robot would have to come to a stop either way it it were to abruptly turn around like that, so I don't think that would be an issue. But are there any other issues situations like this would cause in the controller, etc?

SteveMacenski, 2021-12-08:
If you have multiple intersections due to roughly co-linear paths such that each segment before a "cross" is very short, resulting in extremely poor path tracking abilities since it can't see in the future far enough. 

Also imagine you have 2 colinear segments, one that leads to the other with a point that marks the change between them. As the robot approaches that point, it will start seeing sections of the other side of it causing the same issue as before if we only prune based on integrated path-distance away.  Removing segments based on only directional changes is probably going to be error prone and also has the issue I describe in the paragraph above where there's no limit for how large or small they could be. 

Overall, I think the co-linear problem is another altogether, the example you brought up is a very extreme case and I'm not sure there will be a generalized solution possible if you have path segments that are exactly or nearly overlapping for any consequential amount of time. No path planner should be giving you those results & NHP can give you it, but only if poorly selected points are used. There are other tools like the waypoint follower that could instead be used that issues only 1 point at a time. 

Some solutions here:
- Prune path `N` meters of integral path distance away, to be set based on the controller algorithms being used as a parameter
- Prune path after passed (if not already done, I think it is)
- (Potentially) Try to detect pass crosses and prune at them (see issues with that in description above) and issue them between visitation
- (Potentially) Try to detect significant portions of path overlap (see issues with that in description above) and do (?) about it

ymd-stella, 2021-12-08:
Is this implementation informative?

https://github.com/AtsushiSakai/PythonRobotics/blob/d5aca66e1b61692e239e95c928884ede6f2c94cd/PathTracking/pure_pursuit/pure_pursuit.py#L77-L108

It seems to keep track of where the robot is on the path and look for a point that is a certain distance along the path. (It's not a cumulative distance.)
However, since Nav2 recalculates the global plan from the current point, it may be difficult to keep track of where the robot is on the path.

SteveMacenski, 2021-12-09:
Yeah, we have things like that in Nav2 that are a bit more efficient. Finding some point `N` distance away or `N` integral path length forward is not challenging. Our challenge is designing a set of rules that is going to cover all of the edge cases for a robot following a path that may have crosses with itself or have segments of it that align with itself. 

Only taking the next `N` meters of the path would deal with 85% of the issue I believe and is very easy to implement (in fact, we already do that in RPP / DWB in various places).

What's our challenge is how to handle the edge cases above and additionally how to determine where the robot is along the path. If the robot is at the cross intersection in the path, how will it pick the right point 100% of the time? I can think of a few ways, but all 85% answers with a couple of edge cases missing. 

"better is good" but I'd like full solution if possible, but certainly we can fix what we can fix easily right now first.

ymd-stella, 2021-12-09:
> When we [find the points closest to the robot to start transforming the path [...]

Sorry, I realized that the above solution has already solved most of the problem. The link is not a permalink and probably points to the wrong line. That led to my misunderstanding.

mradams-bastian, 2021-12-10:
The issue with the paths being too short could be resolved in most cases if the path is not pruned when it first reaches the point of intersection, but rather, it is pruned when it comes back around and crosses that point again. Like if you were playing that snake game, it would be the point at which you lose.
[image]

Even in the case of an overlapping, or nearly overlapping path, the whole stretch of path would be free on the way there. And by the time it turns back around, the remainder of the path would appear and the portion the robot already traveled would be cleared:
[image]
The only time when the path would be super short in this case would be when the robot approaches the turnaround point, which I don't imagine would be much of an issue.

However, now that I think about it, it almost seems to me like all this wouldn't be necessary if we could get the idea to work of choosing the point that is the closest distance along the path. If there are multiple points that are about equally close to one another, and assuming we know where on the path the robot is, if we always choose the most proximate point on the path, that would take care of these specific edge cases, wouldn't it?

SteveMacenski, 2021-12-13:
> it is pruned when it comes back around and crosses that point again

Very good point, I did not consider that. That I believe as well would work. My concern with your first image is that we'd need a way to characterize the amount of distance we leave between the robot and the cross over path being exposed in frames C and D so that algorithms like DWB wouldn't try to turn around and just follow the newly exposed path of the path. But that could also be handled by the pruning N meters ahead if done in conjunction.

So perhaps a somewhat full solution for passing to the controllers would be:
- Prune passed segments permanently from stored path
- Prune path N meters of integrated length ahead, locally for 1 iteration -- fixes macro-scale path crossings
- In that already pruned path, if there is an intersection still, prune up until the intersection point wraps around

Now is the goal checker issue, it simply checks based on the end pose so if we pass the end pose as part of the loop, it'll just as done. Heuristically, we could just say look at path length and if greater than X, disregard, but that wouldn't be super principled in case something went wrong. But this only happens in the case that the end is related to this loop / colinear segment.  

We also need to update any similar logic in the controllers themselves. RPP is fine, but DWB I think mimics some of the goal checking logic internally in the critics to slow on approach / weight towards that pose. TEB I'm not sure about off hand, I'm less familiar with its internal workings.

ErikOrjehag, 2022-01-20:
We added this very small change to `transformGlobalPlan` to find `transformation_begin` with great success:
```
auto last_pose_it = std::min(global_plan_.poses.begin() + 5, global_plan_.poses.end());
  auto transformation_begin = nav2_util::geometry_utils::min_by(
    global_plan_.poses.begin(), last_pose_it,
    [&robot_pose](const geometry_msgs::msg::PoseStamped & ps) {
      return euclidean_distance(robot_pose, ps);
    });
```
This works because the global plan is constantly being pruned so transform_begin should usually be the first or second element.
[image]

SteveMacenski, 2022-01-20:
Am I to read this as you only look over the next 5 poses to which is closest? 

How did you come up with 5? This adds kind of a "magic number" into the stack that's dependent on the speed of the robot / density of the path. But I think this could be adapted to be a length (e.g. check only within the next N meters) and work fine and be a bit more general!

I think the issue still remains if you have and overlapping path (e.g. go straight, then go straight back) when you make the flip from forward to back. But that's also an edge case within an edge case, and this is definitely an improvement towards normal behavior. 

Is this something you'd be open to contributing back with a PR?

sathak93, 2022-01-23:
using integrated path length and few tweaks 

https://user-images.githubusercontent.com/57533054/150690608-1253dc75-2bac-40d5-bf4f-6fccf2823e25.mp4

this method also looks good on DWB
if you are ok with this result .i can create a pr

SteveMacenski, 2022-01-24:
Please do!

Aposhian, 2022-01-27:
This bug has affected us a lot, since our paths have lots of intersections. @sathak93 let me know if you need any help preparing the PR.

sathak93, 2022-01-27:
RPP was resolved.
dwb has a problem
some critic makes the robot turn away without executing some poses.

sathak93, 2022-01-27:
https://user-images.githubusercontent.com/57533054/151329171-f45ccf26-1716-40cb-aa0f-870913cf7256.mp4

SteveMacenski, 2022-01-27:
(copied from slack)

If this were just following waypoint goals, that looks like the goal tolerance to me (e.g. gets within roughly 0.25m of the goal, which is the default value and the value we see it end in with the last waypoint 4). But since that's not the case, I had a take a look in the DWB critic code. 

If I were to give an educated guess based on a few minutes of looking, I would wager it be the path alignment critics (but could be the path distance). They look a bit ahead of the path with a parameter `forward_point_distance` (set to 0.325 by default, but we set to 0.1 in the param file https://github.com/ros-planning/navigation2/blob/fee433ff76b1fe2997043bec16b81568fbead49e/nav2_bringup/params/nav2_params.yaml#L162) to make sure its aligning properly with the path. When you get close to those goals, it might be looking "up" the next part of the path to the next waypoint and once that hits a critical mass, it flips around and goes the other direction.

closed 2022-03-21 by SteveMacenski