ros-navigation/navigation2 #2633: Reference Velocity Smoother

vinnnyr, 2021-10-20:
https://github.com/ros-planning/navigation2/pull/2631 will remove acceleration / deceleration limits from RPP due to some issues. With the idea that these kind of things should be handled by a downstream velocity smoother anyways. 

Is there a home for a velocity smoother in Nav2, either as a tutorial, or a full implementation?

This proposed smoother could set (this list is just an initial set of ideas):
- acceleration limit 
- deceleration limit
- jerk limit 
- velocity deadband (do not command a velocity between x1 m/s and x2 m/s)

vinnnyr, 2021-10-20:
On another note, I am not sure why this issue did not pre-populate with the issue template -- perhaps bc I am using the iOS app... Apologies, I will edit to fit the template soon as possible.

wilcobonestroo, 2022-02-04:
> Aren't most of those taking odometry messages https://github.com/kobuki-base/velocity_smoother/blob/devel/src/velocity_smoother.cpp#L161 to check and limit the next velocity command by? That would make most sense to me.

I see that the Kobuko package has 3 options for input: **none**, **odometry** and **commanded velocities**. Odometry makes sure that the new `cmd_vel` value is within bounds compared to the currently estimated odometry. The commanded velocities looks at the previous `cmd_vel` that was send and assumes that the robot follows these commands. I don't understand the none option yet, but I will have a look at their code to see what it actually does. I think it makes sense to at least have the options for **odom** and for **commanded velocities** as input.

In the three implementations mentioned above I only see speed limits and acceleration limits. I think we can add jerk limits as an option.

@vinnnyr Can you describe some use cases or scenarios where you use the deadband issue? E.g. what are typical values? Is there one band or multiple? What should happen if the orginal cmd_vel is in the deadband?

I can pick up this issue and work on a nav2 velocity smoother. Should I put it in its own package or add it to an existing one? What would be a good name for the package? Simply `nav2_velocity_smoother`? There is another folder called `nav2_smoother`

vinnnyr, 2022-02-04:
Jerk Limits
--
- Yeah so I can concede that jerk limits when in **odometry** mode might not make a lot of sense due to localization noise.

Deadband
--
The velocity smoother in the version of Regulated Pure Pursuit (RPP) was in **odometry** mode. 
The deadband issue was inspired by issues we were fighting with RPP on a robot. We would give RPP a reasonable acceleration limit, but since that robot did not have a lot of control regime in the low end of velocity inputs, getting the robot to start would be tough. 

If given a reasonable acceleration limit ( that we _would have_ wanted enforced when the robot was at speed), we would enter this situation.

```
delta_t = 0.1
max_acc = 1.0

# loop 0
current_velocity = 0.0
# max_allowed_velocity = current_velocity + delta_t * max_acc
max_allowed_velocity = 0.1
cmd_velocity = 0.1

# loop 1
current_velocity = 0.01
max_allowed_velocity = 0.11
cmd_velocity = 0.11

# loop 2
current_velocity = 0.01
max_allowed_velocity = 0.11
cmd_velocity = 0.11 #and so on
```
Where loop 2 onwards, we were stuck at commanding a velocity the robot would have never been able to move.

I think in our use case, a deadband from `(0.0, 0.5]` for cmd_vel would have been useful (linear.x). 

> What should happen if the orginal cmd_vel is in the deadband?

Hm , this is a tough issue to generalize. In essence, I want the smoother to _either_ command `0.0` or `0.5`. Maybe this would just be something of this sorts: "if my current velocity is in the deadband range, do not apply the acceleration limit based on odom, but rather apply the acceleration limit based only on cmd"

wilcobonestroo, 2022-02-04:
That helps understanding the deadband issue :smile: So there should probably be several options for the behavior in the band.

Do you think we should include jerk for the commanded velocity mode? Or can we completely drop the jerk issue?

Ps. I started working on the boilerplate code and reading the parameters: https://github.com/wilcobonestroo/navigation2/tree/add-velocity-smoother

SteveMacenski, 2022-02-04:
> In the three implementations mentioned above I only see speed limits and acceleration limits. 

Some only handle X, one of the things to make sure is that it supports omni robots in the Y velocity / acceleration direction.

> Should I put it in its own package or add it to an existing one?

`nav2_velocity_smoother` sounds good to me! 

> Do you think we should include jerk for the commanded velocity mode? Or can we completely drop the jerk issue?

I think for a first-release we can not include jerk, but if @vinnnyr wanted to come in after and include it, that PR would be easily merged. After the bulk is in place, adding a couple new parameters and a new constraint would be a trivial PR to review and merge.

wilcobonestroo, 2022-02-05:
Hmmm... I had never thought about omni robots and the y-velocity. In the implementation from Care-O-bot (cob) they treat the x and y as independent things to control. I.e. they simply apply the rules for linear speed and acceleration to both components. So, in theory you can have a total combined linear speed larger the the max linear speed (e.g. when both x and y are at their max). Is this what you also had in mind?

Another question: what is the logical way to have this node running? I was thinking to listen to the original `cmd_vel` topic and in the callback immediately publish the smoothed version. There can be a timer-based check to see if there are messages coming in or not. In the other implementations, I also see an approach where the smoother has his own timer (and thus his own publish rate). Velocity message that arrive are cached and processed at the speed of the internal timer. I don't see the benefits of this. I think it only adds latency in the messages and more complexity in the code. So, I was thinking in the direction of reacting to incoming `Twist` messages on `cmd_vel`. In the meanwhile, odom messages can be cached to determine the best odom estimate once a Twist message arrives. What do think?

wilcobonestroo, 2022-02-06:
I have some initial smoothing attempts going on :smile: 

[image]

SteveMacenski, 2022-02-07:
> they treat the x and y as independent things to control

That would be fine to do, so you could make some functions that take in the min/max values (or probably a struct containing them) and then pass in each axis into them separately to smooth independently. Would be nice actually to have that abstracted architecturally. 

I'm not sure the best way off hand to deal with a maximum velocity that is higher than that of the sum of the components. I think that's also the role of the local trajectory planner. This is smoothing out the values, which I agree would be helpful to enforce the constraints. Though enforcing each direction's constraints seems good enough to me unless there's a clear way of doing that demonstrated by another method linked above.

> I was thinking to listen to the original cmd_vel topic and in the callback immediately publish the smoothed version. There can be a timer-based check to see if there are messages coming in or not.

It looks like they _all_ do that in one way or another (timer, while loop, etc). I figure there must be a reason for that? Regularity of the intervals? So that it can be run at a faster rate than local trajectory planner is executing at in order to have a smooth interpolation?

wilcobonestroo, 2022-02-08:
I have also been playing around some more. I assumed that I would always get a continuous stream of `cmd_vel` messages. However, they immediately stop once the goal has been reached. And using de deceleration constraints you need to make it stop slowly. So, I need to have the node running on its own timer anyway. Moreover, smoothly interpolating by having a higher rate is the best reason for having it that way.

wilcobonestroo, 2022-02-11:
[image]

The behavior is now like this. The ramp-up and ramp-down seem ok to me. What should be the behavior at the top? Currently, it is almost following the noisy behavior of the input, because this is within the acceleration limits. Is this behavior ok?

wilcobonestroo, 2022-02-11:
[image]

I can also do something like this with a buffer for the `cmd_vel` messages. As with all smoothing you get some delay in the signal.

SteveMacenski, 2022-02-11:
> What should be the behavior at the top? Currently, it is almost following the noisy behavior of the input, because this is within the acceleration limits. Is this behavior ok?

Yeah, but I'd test with setting the max accels to something lower to see what happens if those little variations are invalid (so it should be smoothed out). Actually, that should be more or less what you show with the buffer. 

The buffer is an interesting idea. I don't think that should be the default behavior but I'd be fine with that a parameterized option.

wilcobonestroo, 2022-02-21:
I am trying to add tests to the package, but the linker is complaining that it can't find the symbols for my class. I guess there is something wrong with the `CMakeLists.txt` in my test folder. Can someone have a look at: https://github.com/wilcobonestroo/navigation2/tree/add-velocity-smoother/nav2_velocity_smoother/test and see if I have mistakes in the CMake files?

SteveMacenski, 2022-03-01:
How's this going?

SteveMacenski, 2022-03-29:
@wilcobonestroo any update?

wilcobonestroo, 2022-03-30:
Most functionality is in place. I have to remove some buffering code. Moreover, I did not include the deadband yet and I don’t know how (or where) to write the documentation. I will try to finish it this weekend and make a PR. Code is at: https://github.com/wilcobonestroo/navigation2/tree/add-velocity-smoother

wilcobonestroo, 2022-04-19:
Could someone give me some feedback on [this code](https://github.com/wilcobonestroo/navigation2/tree/add-velocity-smoother/nav2_velocity_smoother) for this issue?

jwallace42, 2022-04-21:
@wilcobonestroo can you put in a pr? Then I can take a look :)!

SteveMacenski, 2022-04-27:
Hi,

I'm back from PTO now and will take a look tomorrow or the next day. I haven't taken a look at this yet, but I did want to point you to ruckig that's being used in https://github.com/ros-planning/navigation2/pull/2816. I think there's some natural synergies here to use that for this work potentially. I know it would be a near total rewrite regardless of what you've done so far, so I just want to bring it up, but it is not a requirement to use it. Its being used in MoveIt2 and likely to be added to RPP so if we do other kinematics "stuff" it might be good to use to be consistent with other ecosystem projects.

Steve

SteveMacenski, 2022-05-02:
@wilcobonestroo in terms of feedback
- Make sure to have this be a component node (register it) so that we can load this into the component container on launch, not only as a standalone server
- The all-caps chars are only used on get/set of variables, I think just having the string in the get/set line would be good. I don't have a strong preference about it, but that is what we do in other nodes

AlexeyMerzlyakov, 2022-05-07:
My 5 cents into review:
* I am not sure that `MAXIMUM_LINEAR_VELOCITY_X...DEADBANDS` static variables are suitable solution for declaring names of parameters. Usually these kind of variables are using to declare some global constants using in whole module or even by other modules (see `nav2_costmap_2d/include/nav2_costmap_2d/costmap_filters/filter_values.hpp` and `nav2_util/include/nav2_util/occ_grid_values.hpp` for reference). So, why don't just make something like inside class header:
  ```
  const char maximum_linear_velocity_x_param_name_ = "maximum_linear_velocity";
  ...
  const char deadbands_param_name_ = "deadbands";
  ```
* When using `declare_parameter()` API without default value (e.g. `declare_parameter(MAXIMUM_LINEAR_ACCELERATION_Y, rclcpp::ParameterType::PARAMETER_DOUBLE);`), it will throw an error in case if parameter is not set. Therefore, further usage of `get_*_parameter()` routines doesn't make sense: at the point when we will reach these routines, parameters will have a value. In this case the code be simplified to something like:
  ```
  // declare_parameter(param_name, default_value);
  // OR better as Steve mentioned:
  declare_parameter_if_not_declared(node, param_name, default_value);
  ...
  maximum_linear_velocity_x_ = get_double_parameter(param_name).as_double();
  ```
* All subscribers/publishers should be created during `on_configure()` stage and resetted at `on_cleanup()`
* It is better use `rclcpp_lifecycle::LifecyclePublisher` for publishers
* Dynamic parameters callbacks handler to set in `on_activate()` stage, and reset on `on_deactivate()`
* `RCLCPP_INFO()` in frequently calling `VelocitySmoother::get_new_speed()` calculation method will cause a garbage in output. It is better to change it to `RCLCPP_DEBUG()`
* Revise `timer_callback()` and `get_new_speed()`: it looks like `double new_speed` should be `double new_speed_x` and 
  ```
  new_setpoint = current_speed - acceleration;
  new_setpoint = std::max(std::max(new_setpoint, set_speed), -max_speed);
  ```
  could be optimized to ->
  ```
  new_setpoint = std::max(std::max(current_speed - acceleration, set_speed), -max_speed);
  ```
* I can not find any angular twist smoothing logic for `last_send_cmd_vel_`. It looks like it was built into the module design since we have `maximum_angular_velocity_` / `maximum_angular_acceleration_` / `maximum_angular_deceleration_` parameters, but never used them. Also, it looks like angular component of input velocity (be it odom or cmd_vel inputs) is never being passed to outer smoothed `last_send_cmd_vel_` velocity.

SteveMacenski, 2022-05-20:
I started to play with this using ruckig just to see how / if it would work. See branch: https://github.com/ros-planning/navigation2/tree/vel_smoother. Its still in development and incomplete. It brings up a few questions I'm still working through
- How do we set the current acceleration / jerk in the process? We don't have access to that information reliably unless we numerically differentiate the odometry which is unstable
- --> however, it does work if we do closed loop control, that we assume the last timestep we met the required velocity and so we can store the last iteration's velocity/acceleration/jerk to use as the "current" in the next timestep. That would get around the numerical differentiation of odometry for closed loop feedback, but then would we be unable to support closed loop feedback?
- How do we set the target acceleration when given an input twist, or should we?

@AndyZe over at PickNik is using Ruckig for MoveIt2, maybe he has some thoughts to share?

AndyZe, 2022-05-20:
I don't have any magic answers but it sounds like you're asking the right questions. Tough problem.

> We don't have access to that information reliably unless we numerically differentiate the odometry which is unstable for closed-loop feedback

If you end up numerically differentiating, here's a decent way to do it:

- Calculate the derivative. It could be a simple Euler calculation like accel = v(i+1) - v(i) / delta_t
- Filter from beginning to end with a low-pass filter
- Then filter from end to beginning with the same filter
- To filter forwards then backwards removes the phase delay from filtering

> it does work if we do open loop feedback (e.g. we assume the last timestep we met the required state and so we can store the last iteration's velocity/acceleration/jerk to use as the "current" in the next timestep).

I've tried this. It worked reasonably well, but path-tracking accuracy seemed a bit better on hardware when the "nominal current state" was used as input for the next iteration.

> For setting the current acceleration / jerk, I suppose we could set them all to 0 across the board and then the trajectory generated would assume the extremes for starting/stopping the trajectory. So for closed-loop set them all to 0-s since we can't estimate them reliably?

My intuition says that won't work very well. I could be wrong, though

SteveMacenski, 2022-05-20:
I don't actually understand the difference between the 2 methods you mentioned, can you elaborate on the second point? (1) would be setting the `new_{velocity, acceleration, pose}` from the `output` as the `input` of the next iteration. I'm not sure what (2) entails. There's also `pass_to_input` method that appears in the demos but never in the documentation that you could see in my branch I have open questions around precisely its nature (which might be what you're referring to).

> My intuition says that won't work very well. I think the motion will be slower than you expect. I could be wrong, though

As well. Though if we do a trivial `v min / max = v0 +/- a * t` calculation for thresholding, it's not really taking the current acceleration into account either. I need to do some thinking on that, but that feels analogous to the `ruckig` setting all current accelerations to `0`. 

I feel like this _shouldn't_ be a technology mismatch, but might end up being. Traditionally in the ROS navigation ecosystem, we've just taken some $v_i$ and used basic kinematics to find the guard rails of $v_{min}$ and $v_{max}$ based on the set acceleration min/max and threshold it. 

The 2 major uses cases of it are:
- Setting the update rate to roughly the same as the trajectory planner rate (thing making `cmd_vel`s), so 1 velocity command in = 1 velocity command out, with guard rails applied essentially.
- Setting the update rate to higher than the controller rate, so 1 velocity command in = `N` velocity commands out, somewhat applying a smoothing trajectory since each `dt` its called will update the velocity towards the commanded velocity by the acceleration profile.

Since we have a constant stream of these `cmd_vel`'s coming out of the trajectory planner at a relatively consistent rate (and with algorithms that pre-apply varying levels of feasibility constraints) the target setpoint for the velocity is constantly changing but we don't know what the future holds to be able to meaningfully set target acceleration setpoints.

SteveMacenski, 2022-05-21:
Well another thing to point out is that these all assume independent X, Y, and Theta velocity channels. Something `Yocs` does is constraining other channels when one is too high so that the velocity follows the vector previously created, just scaled down https://github.com/kobuki-base/velocity_smoother/blob/devel/src/velocity_smoother.cpp#L263-L296.

I've spent time thinking and I think that's a good idea, but I think the way this is implemented does not actually guarantee that the constraints are respected. An example being if the linear velocity blows past its limit, but then in scaling it down, it scales down the angular velocity so then its now below its minimum acceleration threshold. Further, pushing this method into 3D (X, Y, Theta) is not trivial so I've pulled out some other vector math to do it but I'm unhappy with the state of my implementation so I'm going back to the drawing board trying to come up with a better method.

AndyZe, 2022-05-21:
> so that you wouldn't be double differentiating noisy velocity data? 

Small correction here: Ruckig doesn't care what the current jerk of the robot is. It's not an input to Ruckig. So you only have to differentiate once to calculate acceleration.

> I don't actually understand the difference between the 2 methods you mentioned, can you elaborate on the second point?

The second point means:

- You have _target_state(i)_ at iteration _i_. This is the nominal, desired state at iteration _i_.
- At iteration _i+1_, you use _target_state(i)_ as the "current state" of the robot, for input into Ruckig. Basically, you assume that it reached the previous target state perfectly.

It's really similar to Point 1 except you're not using the Ruckig output, you're using the target state that you provided originally. It should be an almost perfect match, which is why I was confused to see different behavior on our hardware.

SteveMacenski, 2022-05-21:
Thanks for the clarification!

AndyZe, 2022-05-22:
If it would help, I can write some of this. Maybe the velocity differentiation part and/or the Ruckig part itself. My problem is, I have no idea where things go in the Nav2 codebase.

SteveMacenski, 2022-05-23:
I think from my current looking, ruckig is not the best choice for us unfortunately :frowning: which I'm disappointed by since it looks like it could really streamline some things. However we want to (1) use the maximum kinematic limits possible to achieve velocities ASAP and maintain them vs using the full time allotted and (2) be able to proportionately bound the velocities of the axes so that we maintain the same (or as similar as possible) commanded direction. 

From that, `ruckig` doesn't quite make sense for our application. However, I think it would be really interesting to see if we could use `ruckig` in the trajectory planners or smoothers to work with theoretical trajectories being generated versus involvement of real data. If we're using it for trajectories generated via other methods, then we should have acceleration and other information we can meaningfully use versus live data.

SteveMacenski, 2022-05-25:
#2964 is ready for testing if folks want to kick the tires!

vinnnyr, 2022-06-10:
I just noticed I missed the tags here when my input was requested for Jerk. So Sorry about that!!! I think this question is no longer relevant, but indeed I was hoping Jerk could be handled based on `cmd_vel` alone rather than trying to measure acceleration based off of sensors / localization estimates.

> (1) you can make trajectories that are themselves already jerk limited

SteveMacenski, 2022-06-10:
I think it makes sense to add jerk limiting to the trajectory planners perhaps to get around this situation, so that way the computed trajectories generating the commands are limited by it so that the velocity smoother doesn't require to do it -- and then its based on theoretical models and not actual current sensor data so you can differentiate it to your heart's desire.

SteveMacenski, 2022-06-13:
Merging imminent

closed 2022-06-13 by SteveMacenski