ros-navigation/navigation2 #2654: Nav2 build from source issues in rolling distribution

poornimajd, 2021-10-30:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04(docker)
- ROS2 Version:
  - rolling
- Version or commit hash:
  - ros2 master branch

#### Steps to reproduce issue

I am trying to build the nav2 from source.Initially I followed the build instructions for ros2 rolling distribution from this [link](https://docs.ros.org/en/rolling/Installation/Ubuntu-Development-Setup.html).I followed all the steps inside a ubuntu 20.04 docker container.I was able to successfully install the rolling version.After this I source the ros2-rolling as:
`source ./ros2_rolling/install/local_setup.bash`  [in /home directory]
And then,create nav2_ws [in /home directory] and followed the instructions from [here](https://navigation.ros.org/build_instructions/index.html#build-nav2-main) to build nav2 main.But when I execute :
`colcon build --symlink-install`
I get a error as shown below.

#### Expected behavior
Build successful
#### Actual behavior
```Finished <<< nav2_simple_commander [1.17s]
--- stderr: nav2_util                          
CMake Error at CMakeLists.txt:18 (find_package):
  By not providing "Findbondcpp.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "bondcpp", but
  CMake did not find one.

  Could not find a package configuration file provided by "bondcpp" with any
  of the following names:

    bondcppConfig.cmake
    bondcpp-config.cmake

  Add the installation prefix of "bondcpp" to CMAKE_PREFIX_PATH or set
  "bondcpp_DIR" to a directory containing one of the above files.  If
  "bondcpp" provides a separate development package or SDK, be sure it has
  been installed.

Failed   <<< nav2_util [1.83s, exited with code 1]
```
#### Additional information
I checked out some of the issues related to bondcpp in this repo,but I am still not able to solve the above issue.
Any suggestion is greatly appreciated.
Thank you

SteveMacenski, 2021-10-30:
Bond is a dependency of nav2, you should resolve it via rosdep as you may any other dependency 🙂 `rosdep install -r -y —from-path <path>`

poornimajd, 2021-10-31:
Thank you for the quick response,the error is solved.
But when I am building the nav2 main branch ,after installing all the dependencies as given in - https://github.com/ros-planning/navigation2/blob/main/tools/underlay.repos
I tried to build using `colcon build --symlink-install`  . While building gazebo_plugins as a part of the gazebo_ros_pkgs,the system hangs because all the RAM memory(15.5gb) is being used.I have attached a screenshot of the htop command below-
[The consumption goes till 14.9gb and the system hangs,I have taken the screenshot a little prior to it]
[image]

I had cloned the ros2 version of the gazebo_ros_pkgs repo .
Is there something wrong I am doing?
Any suggestion is appreciated.
Thank you.

SteveMacenski, 2021-10-31:
You’d have to inquire with the gazebo folks, but make sure you’re building with release flags. I wouldn’t recommend building the sim packages from source, there should be binaries available.

poornimajd, 2021-11-04:
Thank you for the speedy reply.I tried to install the gazebo packages using binaries itself.But whenever I launch the nav2 using - ros2 launch nav2_bringup tb3_simulation_launch.py I get the following error- 
gzserver: symbol lookup error: /opt/ros/rolling/lib/libgazebo_ros_init.so: undefined symbol: _ZN6rclcpp11InitOptionsC1E19rcutils_allocator_t

I tried to check out some of the issues in gazebo_ros_pkgs,but could not resolve this issue.
Any suggestion is appreciated.
Thank you.

SteveMacenski, 2021-11-04:
And does it work now?

poornimajd, 2021-11-05:
Yes,I was able to get it working using the gazebo binaries.
Thank you

closed 2021-11-08 by SteveMacenski

Tomorrow-SSS, 2022-07-21:
> I have the same error,Can you tell me how you solved this problem? thank you very much.

chenwu-cc, 2023-03-08:
> Bond is a dependency of nav2, you should resolve it via rosdep as you may any other dependency 🙂 `rosdep install -r -y —from-path <path>`

ERROR: Rosdep cannot find all required resources to answer your query
Missing resource —from-path
I can't use rosdep
what should I  do to solve it

chenwu-cc, 2023-03-08:
> Bond is a dependency of nav2, you should resolve it via rosdep as you may any other dependency 🙂 `rosdep install -r -y —from-path <path>`

I also meet E: Unable to locate package ros-galatic-bondcpp

mobydoby, 2024-04-20:
> Bond is a dependency of nav2, you should resolve it via rosdep as you may any other dependency 🙂 `rosdep install -r -y —from-path <path>`

Sorry, I'm running into the same issue and I still don't quite understand. Where can I find the path of Bond for <path>? Do I have to install bond and give rosdep the path?