ros-navigation/navigation2 #2658: CostmapTopicCollisionChecker footprint tilted/shifted

afrixs, 2021-11-02:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Galactic binaries
- Version or commit hash:
  - 0f6f49a3633de66db7cdf5f60ecbb951fea5883f
- DDS implementation:
  - Eclipse Cyclone DDS

#### Steps to reproduce issue
 1. Make robot move, e.g.
```
ros2 topic pub /cmd_vel geometry_msgs/msg/Twist "linear:
   x: 0.0
   y: 0.0
   z: 0.0
 angular:
   x: 0.0
   y: 0.0
   z: 1.0"
```
 2. Use a component which uses CostmapTopicCollisionChecker, e.g.
```
ros2 launch nav2_bringup navigation_launch.py
```
```
ros2 action send_goal /backup nav2_msgs/action/BackUp "target:
   x: 1.0
   y: 0.0
   z: 0.0
 speed: 0.2
 time_allowance:
   sec: 10
   nanosec: 0"
```
 3. Visualize footprints returned by CostmapTopicCollisionChecker::getFootprint (needs some code editing)

#### Expected behavior
Footprints should be aligned with the checked poses.

#### Actual behavior
Footprints have an angular/positional offset.

#### Implementation considerations
Offset is caused because timestamp is being ignored in tf lookup. The stamp used for creating the footprint message should be used for transform lookup in CostmapTopicCollisionChecker::unorientFootprint(). I.e.:
 1. Relevant pose stamp must not be discarded [when publishing footprint message](https://github.com/ros-planning/navigation2/blob/0f6f49a3633de66db7cdf5f60ecbb951fea5883f/nav2_costmap_2d/src/costmap_2d_ros.cpp#L470) (`pose.header.stamp` needs to be used instead of `now()`)
 2. The stamp must be retrievable from FootprintSubscriber. This needs discussion, since there are multiple ways to achieve it. For example it could be done as follows:
    * add `stamp = footprint_stamp;` [here](https://github.com/ros-planning/navigation2/blob/0f6f49a3633de66db7cdf5f60ecbb951fea5883f/nav2_costmap_2d/src/footprint_subscriber.cpp#L70) (maybe it was originally intened since `rclcpp::Time & stamp` parameter is non-const, but cannot say for sure since the doxygen comments for FootprintSubscriber are very unclear and the functionality they describe is absolutely different from what the methods actually do)
    * add `const nav2_util::LifecycleNode::WeakPtr & parent,` into the [constructor of `CostmapTopicCollisionChecker`](https://github.com/ros-planning/navigation2/blob/0f6f49a3633de66db7cdf5f60ecbb951fea5883f/nav2_costmap_2d/include/nav2_costmap_2d/costmap_topic_collision_checker.hpp#L52) (to retrieve node's clock)
    * use
      ```
      rclcpp::Time stamp = clock_->now();
      if (!footprint_sub_.getFootprint(footprint, stamp, [timeout])) {...
      ```
      in [`CostmapTopicCollisionChecker::getFootprint`](https://github.com/ros-planning/navigation2/blob/0f6f49a3633de66db7cdf5f60ecbb951fea5883f/nav2_costmap_2d/src/costmap_topic_collision_checker.cpp#L95)
 3. The stamp must be used in tf lookup. `const rclcpp::Time stamp = rclcpp::Time()` parameter should be added to [`nav2_util::getCurrentPose`](https://github.com/ros-planning/navigation2/blob/0f6f49a3633de66db7cdf5f60ecbb951fea5883f/nav2_util/include/nav2_util/robot_utils.hpp#L45) method and used in `CostmapTopicCollisionChecker::unorientFootprint`

SteveMacenski, 2021-11-09:
Please submit a PR, all of those sound very reasonable. I can't think I'd find any fault in that and seems like you probably already have a working implementation with that level of detail :wink:

SteveMacenski, 2021-11-16:
Any update here?

closed 2021-11-29 by SteveMacenski