ros-navigation/navigation2 #2659: Smac Hybrid Planner ignores costs around the end of the plan

zoltan-lengyel, 2021-11-02:
**Facts**
The smac hybrid planner periodically attempts to do an analytic solution to finish the plan as it expands _(related function: AStarAlgorithm<NodeT>::tryAnalyticExpansion)_.
When there is a cached analytic solution, the planner checks if there is any collision on this analytic path _(DUBIN curve)_.
If there is a collision, the planner discards the analytic solution and tries it again a few expansion later.
If there is no collision, the plan is finished using the analytic solution, regardless of the obstacle costs it crosses.
_(Technically, the planner may pick a different analytic solution further away due to a refining process, but this is not relevant for the point I am trying to make.)_

**Issues**
The analytic solution can make the plan go through really high cost areas at the last 10 meters _(lookup_table_size / 2)_, because its only criteria for success is not colliding.
This can make the plan go really close to obstacles.
In our navigation system the costs in the costmap are not only the result of obstacle inflation, they serve navigation purposes as well, like making the robots follow a preferred line. These navigation constraints are also ignored along the analytic solution.

**Ideas**
I tried simply limiting the distance the analytic solution can be triggered from, to **minimum_turning_radius * 3**.
I did not change the lookup_table_size, because I still wanted to use the analytic solutions to influence the heuristics _(NodeHybrid::getDistanceHeuristic)._
This resolved my issues fairly well. Best case performance degraded of course in simple cases, but the worst case performances for complicated planning cases were not affected significantly _(some special cases significantly improved even)_.
I my case losing some of the extremely quick results is a perfectly acceptable tradeoff for stricter cost awareness around the end of the plan, so I would be glad to have some configuration options for this.
Another idea I had was to disregard the analytic solution only if it was long and crossing high obstacle costs. The reason I didn't like this idea very much, because it doesn't help much if the goal is close to an obstacle, which happens very often in my use cases.

**Questions**
Do you have any suggestions on how to resolve this issue in a better or nicer way?
Do you think the smac hybrid planner should have some configuration options to improve this situation at the cost of performance?

afrixs, 2021-11-04:
I'm supporting this feature request since we experience better planning results too when the analytic expansion is somewhat constrained. Despite the fact that the analytic expansion is certainly needed to provide a reasonable final path section (as discretized node search often misses the goal), some constraining is needed to prevent dangerous cost-unaware maneuvers, especially in tight spaces around the goal.

There are multiple constraints that we have found improving the results while not being costly to CPU, I'll list them here to be considered:
 * `max_analytic_expansion_length` - practically equivalent to @zoltan-lengyel 's proposal, it just uses analytic path length instead of start-goal distance (cannot say which approach is better, this was just a bit less code writing for me)
 * `max_analytic_expansion_angle_range` - to prevent difficult cost-unaware turning maneuvers (happening mostly in Reeds-Shepp motion) which are usually more dangerous than straight movement and hard to push away from tight spaces by a cost-aware smoother (see note below)
 * `max_analytic_expansion_cost_subelevation` - to "wait" with analytic expansion to start from a wider/safer place (it basically prevents one concrete rare situation which is hard to explain but I will simulate an example of it when I have more time)

Note: Currently we are [merging](https://github.com/ros-planning/navigation2/pull/2569) a cost-aware smoother with @SteveMacenski into Nav2 that will push your path away from obstacles in post-process. My aim is to try to merge also the constraints mentioned above and `obstacle_heuristic_admissible` option (which brings more outlook and optimality to search at the cost of longer planning time). All of these options will help producing much safer paths, however, it will take some time until completely merged. Until then you can have a look at [our fork](https://github.com/robotechvision/navigation2/tree/smac_admissible_heuristic) (_smac_admissible_heuristic_ or _galactic_ branch) which contains the smoother and all the options (however, still lacking some cleanup and docs).

zoltan-lengyel, 2021-11-04:
Thanks @afrixs! I will test my problematic cases with the cost-aware smoother once it is merged. I expect it to improve a lot of things. Until then, I will just use my simple workaround.

SteveMacenski, 2021-11-09:
Your original ticket is entirely accurate. I did see this in testing, but this is part of the technique to make the run-time performance something nearing 2D speeds because SE2 planners will spend the vast majority of their time trying to get to the exact right bin at the right angle. This process makes it far more easy. 

> I tried simply limiting the distance the analytic solution can be triggered from, to minimum_turning_radius * 3

This isn't a bad first-order heuristic, but I'd want to understand how much that impacted runtime speed. For a given set of maps (lets say an open one, a very confined one, and a usual warehouse/retail store aisles one) how does this impact timing. I think its worth some increase in time in exchange for path quality, but there's a balance to be struck.

The original technique was paired with a cost-aware optimization-based smoother which would have taken that short-cutting analytic expansion and pushed it out more. This work was in the original implementation here but I pulled it out for a more basic smoother due to run-time and general "didn't work in places I thought it should, I didn't care for the flakiness". That work was revived somewhat recently by @robotechvision in https://github.com/ros-planning/navigation2/pull/2569 to provide that fixed up smoother (still more run-time, but sounds like you're OK with that) + my new more basic smoother into a server for use in any planner, but targeting this one. That would, I believe, largely resolve your problems. 

In `max_analytic_expansion_length` vs `simply limiting the distance`, I don't think either is inherently worse, but I would probably choose `max_analytic_expansion_length` because its more representative of the thing we're trying to measure: effective length of the analytic expansion. But I think both would work fine in practice. 

`max_analytic_expansion_angle_range` is probably not a good idea, you do want some ability for it to make tight maneuvers. You mostly just don't want those to be close to obstacles, so it seems to me a better metric would be `max_cost_in_expansion` or something similar to fail if the analytic length is too long **and** it passes through <sufficiently high cost> space, for example 200+.

zoltan-lengyel, 2021-11-09:
| Case      | Original      | Limited analytic expansion |
|-----------| ------------- | -------------------------- |
| Open area short plan | 5 ms  | 7 ms |
| Realistic warehouse long plan A | 22 ms  | 22 ms |
| Realistic warehouse long plan B | 36 ms  | 34 ms |
| Realistic warehouse long plan C | 51 ms  | 46 ms |
| Confined labyrinth very long plan | 657 ms  | 138 ms |
| Confined hammer turn in long narrow corridor | 187 ms  | 516 ms |

Here are some measurements from my benchmark setup. All of my tests are done with **cache_obstacle_heuristic = false**.The last two confined tests are kind of extreme situations, I only included them to show how significant the difference can be in some cases.

**Changes**
I limited the analytic expansion, so it is only started from less than **minimum_turning_radius * 4** manhattan distance. I used a minimum turning radius of 0.4 meter in these tests, except in the extreme confined ones where the minimum turning radius was 2 meters (these are done with a very long footprint).
I also adjusted the analytic path refinement process, which attempts to do new analytic expansions from further away after the analytic expansion was successful. This refinement can practically undo the previous distance limitation. Ideally it should only try doing refinements until the previously set distance limitation, but I didn't implement this logic properly. I just flat reduced the amount it backtracks from the default 2 meters to 0.25 meter _(assuming a resolution of 0.05 meter)_.

My implementation would need a lot of extra work and cleanup to make it ready for merging, and unfortunately I don't have time for it in the near future. So if anyone else wants to make a nice solution for this please go ahead.

afrixs, 2021-11-12:
> `max_analytic_expansion_angle_range` is probably not a good idea, you do want some ability for it to make tight maneuvers

It wouldn't disable tight maneuvers, it would just make major part of them happen during the discrete cost-aware part of the search, enabling "rotate first in a safe space, then approach goal" plans overtake "approach goal first, then rotate in tight space just before reaching the goal" plans in these situations. In other words it prevents planner from saying "ok, it's just a final turn so I'm done searching", because it's often not "_just_" a final turn, in our experience most 'robot collided'/'robot got stuck' situations happen during turns, especially when close to obstacles. Anyways the constrain would be optional and could be set to `inf` by default.

SteveMacenski, 2021-11-12:
I would support a PR to add a max distance for possible analytic expansion such that we don't short cut too early onto visibility. Would @afrixs or @zoltan-lengyel be able to submit that?  My recommended implementation:
- Store the param
- In analytic expansion after its been gotten, look at the distance `d` and compare to the param (so use actual analytic expansion distance, not L2 norm) 
- If lower, use, if higher, exit

I think that would do 80% of the need. After that I think we could add another parameter for max cost as well, but there would need to be an "out" for it. I'm thinking a minimum distance parameter, perhaps. If distance is < N, even if higher than ideal max cost (but non-lethal) accept it, because its clearly just necessary. But we can do that as another PR to discuss details or if this is even the best solution (perhaps its not).

afrixs, 2021-11-15:
I can do it as I've got it this way already, just need to isolate the update from others and write a test

SteveMacenski, 2021-11-16:
I did some testing today and definitely this number should never be less than 3x the minimum turning radius, that's the minimum threshold where it really starts to spike up CPU time and the tight maneuvers if reversing is enabled and can cause some wacky results. I've found that for a 0.4m turning radius around ~1.5-2.0 meters of length was the sweet spot where it didn't add much more compute and the quality remained high. That's 3.75x the turning radius I used, so I think 4x-5x would be a good default value.

SteveMacenski, 2021-11-20:
@zoltan-lengyel @afrixs  please test this PR, it should resolve your concern. Let me know how it goes! https://github.com/ros-planning/navigation2/pull/2697

zoltan-lengyel, 2021-11-22:
@SteveMacenski thanks!
My unit tests all succeeded with your fix. Looks a lot better than my version, which had to mess with the refinement process.

SteveMacenski, 2021-11-23:
Merged! I'm also happy to reopen or continue discussing if a min length / max cost condition would be helpful, but honestly from my testing, this did wonders

closed 2021-11-23 by SteveMacenski