ros-navigation/navigation2 #2661: Issues with building with BehaviorTreeCPP

felrock, 2021-11-04:
## Bug report

**Required Info:**

- Operating System:
- Ubuntu 18.04
  - 
- ROS2 Version:
- Foxy Binaries
  - 
- Version or commit hash:
- foxy-devel head
  - 
- DDS implementation:
- CycloneDDS
  - 

#### Steps to reproduce issue
Building nav2_behavior_tree, I have installed BehaviorTreeCPP and I get the following error, the *Config.cmake file is located
in my /usr/ folder but under a different name "BehaviorTreeV3Config.cmake"

```
--- stderr: nav2_behavior_tree
CMake Error at CMakeLists.txt:14 (find_package):
  By not providing "Findbehaviortree_cpp_v3.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "behaviortree_cpp_v3", but CMake did not find one.

  Could not find a package configuration file provided by
  "behaviortree_cpp_v3" with any of the following names:

    behaviortree_cpp_v3Config.cmake
    behaviortree_cpp_v3-config.cmake

  Add the installation prefix of "behaviortree_cpp_v3" to CMAKE_PREFIX_PATH
  or set "behaviortree_cpp_v3_DIR" to a directory containing one of the above
  files.  If "behaviortree_cpp_v3" provides a separate development package or
  SDK, be sure it has been installed.

---
Failed   <<< nav2_behavior_tree [11.6s, exited with code 1]
```

#### Expected behavior
That it would build anyway
#### Actual behavior
It doesn't build
#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

padhupradheep, 2021-11-04:
Foxy [supports only 20.04](https://docs.ros.org/en/foxy/Installation.html), you need to upgrade your ubuntu!

SteveMacenski, 2021-11-04:
Please ask on ROS Answers, but in general you need to satisfy dependencies and build the correct branches.

closed 2021-11-04 by SteveMacenski

felrock, 2021-11-08:
Made a post on ros answers about it, thanks!

BATAxjh, 2022-08-18:
hello ,How did you solve it

felrock, 2022-08-18:
The issue that I had was that i built BehaviorCPP without ament first. So you need to make sure that it's built with ament cmake or it wont work.