ros-navigation/navigation2 #2666: NavigateThroughPoses not working when robot frame != "base_link'

charlielito, 2021-11-09:
## Bug report
When using another name for the robot base frame than `base_link`, the action NavigateThroughPoses seems to be broken. 

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Galactic binaries
- Version or commit hash:
  Galactic d8f50da2ed59879112105ac1488e9139ee7f3c71
- DDS implementation:
  - Default (cycloneDDS)

#### Steps to reproduce issue
Replace in `nav2_bringup/bringup/urdf/turtlebot3_waffle.urdf` every `base_link` to let's say `chassis`.
Also replace in `nav2_bringup/bringup/params/nav2_params.yml` every `base_link` to `chassis`.
Then run the tb3 demo

```
ros2 launch nav2_bringup tb3_simulation_launch.py
```
Use Rviz to use the `Nav Through Poses Mode` and set a couple of poses to follow. Activate nav2 stack with navigate through poses action.

#### Expected behavior
The robot should follow the poses.

#### Actual behavior
BT never starts and display the following logs
```
...
[bt_navigator-11] Warning: Invalid frame ID "base_link" passed to canTransform argument source_frame - frame does not exist
[bt_navigator-11]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.2/src/buffer_core.cpp
[bt_navigator-11] Warning: Invalid frame ID "base_link" passed to canTransform argument source_frame - frame does not exist
[bt_navigator-11]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.2/src/buffer_core.cpp
[bt_navigator-11] Warning: Invalid frame ID "base_link" passed to canTransform argument source_frame - frame does not exist
[bt_navigator-11]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.2/src/buffer_core.cpp
...
```

Note that NavigateToPose and WaypointFollowing work well. I tried to look into the code to see if somewhere was `base_link`  hardcoded but couldn't find anything. Do you where might be the problem?

My mentioned and modified files

<details closed>
<summary>turtlebot3_waffle.urdf</summary>
```
<?xml version="1.0" ?>
<robot name="turtlebot3_waffle" xmlns:xacro="http://ros.org/wiki/xacro">
  <!-- <xacro:include filename="$(find turtlebot3_description)/urdf/common_properties.urdf"/>

  <xacro:property name="r200_cam_rgb_px" value="0.005"/>
  <xacro:property name="r200_cam_rgb_py" value="0.018"/>
  <xacro:property name="r200_cam_rgb_pz" value="0.013"/>
  <xacro:property name="r200_cam_depth_offset" value="0.01"/> -->

  <!-- Init colour -->
  <material name="black">
      <color rgba="0.0 0.0 0.0 1.0"/>
  </material>

  <material name="dark">
    <color rgba="0.3 0.3 0.3 1.0"/>
  </material>

  <material name="light_black">
    <color rgba="0.4 0.4 0.4 1.0"/>
  </material>

  <material name="blue">
[... 582 lines cut ...]
    global_frame: odom
    robot_base_frame: chassis
    transform_timeout: 0.1
    use_sim_time: true
    simulate_ahead_time: 2.0
    max_rotational_vel: 1.0
    min_rotational_vel: 0.4
    rotational_acc_lim: 3.2

robot_state_publisher:
  ros__parameters:
    use_sim_time: True

waypoint_follower:
  ros__parameters:
    loop_rate: 20
    stop_on_failure: false
    waypoint_task_executor_plugin: "wait_at_waypoint"   
    wait_at_waypoint:
      plugin: "nav2_waypoint_follower::WaitAtWaypoint"
      enabled: True
      waypoint_pause_duration: 200

```
</details>

SteveMacenski, 2021-11-09:
I almost guarantee its because you didn't change the base link to your actual base link in all of the places necessary. Also review the Nav2 documentation https://navigation.ros.org/configuration/index.html to see all the places where you may need to change this name.

In general though, you should use base_link to properly implement REP 105, but you should definitely be able to use another frame instead if you choose. 

That warning just indicates to me there are more places you didn't make the change.

charlielito, 2021-11-09:
> I almost guarantee its because you didn't change the base link to your actual base link in all of the places necessary. Also review the Nav2 documentation https://navigation.ros.org/configuration/index.html to see all the places where you may need to change this name.
> 
> In general though, you should use base_link to properly implement REP 105, but you should definitely be able to use another frame instead if you choose.
> 
> That warning just indicates to me there are more places you didn't make the change. Though if you actually found somewhere that we hardcoded it without a parameter override, let me know! That should be fixed!

If that were the case `NavigateToPose` shouldn't work at all, but it is working. I shared also the files I used, and there is not  `base_link` anywhere.

SteveMacenski, 2021-11-09:
You need to look at the configuration guide. Just because you don't specify base_link doesn't mean there isn't a parameter in a server or BT plugin node you're using that has it as a parameterization that defaults to base_link.

charlielito, 2021-11-09:
Yes, you are right, I missed totally that, I was thinking that the `robot_base_frame` when defined in the `bt_navigator` part of the .yaml config file file was going to propagate that to all nodes. After adding `robot_base_frame` to https://github.com/ros-planning/navigation2/blob/main/nav2_bt_navigator/behavior_trees/navigate_through_poses_w_replanning_and_recovery.xml#L13 it worked as expected.

Just one final question, is another way of defining the `robot_base_frame` for the `RemovePassedGoals` directly in the .yaml config file?

SteveMacenski, 2021-11-09:
Nope, since those are independent plugins, they all will have their own configuration setting and ports. This is so they’re independent bc user defined plugins will not have access to those same resources. 

Happy to have helped.

closed 2021-11-09 by SteveMacenski

Hercogs, 2024-10-28:
> Yes, you are right, I missed totally that, I was thinking that the `robot_base_frame` when defined in the `bt_navigator` part of the .yaml config file file was going to propagate that to all nodes. [...]

Hi!
Could you please specify what exactly did you do to solve this? I am in the same situation: NavigateToPose works, but NavigateThroughPoses does not. 

I tried your solution but, I edited my custom behaviour XML file, it got error: `"Error loading XML file: ./behavior_th.xml"`, so I am not sure if I understood your solution. 

I can see that `bt_navigator` and `behavior_server` has custom `robot_base_frame` parameter set already.

As this issue is old, I could open new issue if needed.