ros-navigation/navigation2 #2667: usage of has_parameter() before declare_parameter()

padhupradheep, 2021-11-09:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Rolling
- Version or commit hash:
  - main
- DDS implementation:
  - does not matter

There are multiple files, that still checks for the availability of the parameter using `has_parameter()` even without the declaration, something similar to #2663

@SteveMacenski : I will do a PR with all the fixes, before end of this week.

SteveMacenski, 2021-11-09:
I'd ask for them to be enumerated, but if its just a couple we can discuss over the PR. I just don't want to have you waste time if the fixes for some of them are non-trivial before discussing to make sure your understanding of their use are correct. The BT XML was an oversight, but I'd be surprised if there were many others like that.

padhupradheep, 2021-11-12:
>  There are places we use has_parameter in a context where it would have been declared in another place

I actually missed this big time! You are right!

closed 2021-11-12 by padhupradheep