ros-navigation/navigation2 #2670: RPP Rotate to heading at goal / start collision detection based on angular distance & warnings at goals near obstacles

SteveMacenski, 2021-11-11:
@kaetera @sathak93

SteveMacenski, 2021-11-20:
Any ideas on this?

sathak93, 2021-11-20:
I am busy with other stuff. Next week i can work on this.

sathak93, 2021-11-20:
Since you tried  RPP with footprint collision checking . What do you think about the warnings?

SteveMacenski, 2021-11-23:
I think there were 2 topics
- When we rotate to heading, we need the collision checker to do orientation-based discretization, not linear velocity based
- When we're approaching the goal that is very close to an obstacle, we need something to help with reducing warnings

I can work on the first one this week and have a PR ready for you both to take a look over, but its still a little bit of an open question how to handle the warnings of collision when approaching the goal / stop collision checking at the goal pose.

SteveMacenski, 2021-11-23:
@sathak93 @kaetera https://github.com/ros-planning/navigation2/pull/2701 please review!

Kaetera, 2021-11-23:
I thinks it's ok, to keep a general behaviour similar to the current one.
In the original issue you said: "I think that logic should only be used for pure rotations for the rotate to headings".
But here are my thoughts:
1. If the purpose of the collision checking in RPP is to prevent a collision in case "something goes wrong" and that the robot is no more driven by RPP but e.g. just by its inertia, then projecting the poses in the future based on the current velocity makes sense. But currently we project using the velocity command, not the current velocity, and ignore the angular velocity except for pure rotation, so this check is approximative.
2. On the other hand, if the purpose is to check that there is no obstacle on the path (e.g. if the global planner plans every 1s, then an obstacle can appear on the path between two replanning), then we should project the poses along the path.

My feeling is that the current implementation is a mix between 1. and 2.
And since we check the collision in rotation only in pure rotation, solution 1. looks incomplete since it does not covers the case of turning in curves (cf. my awesome drawing below)
If the purpose is 2. then we could implement a collision check inspired by MPC methods, that would account for that RPP will reduce the command near goal, in curves, etc. Then:
- It would naturally handle the "near goal case" (since obstacles beyond the goal are not on the path)
- Avoid the similar "curve case": e.g. if the path does a 90° turn with a small radius, the current implementation triggers collision detections, what is a little annoying in constrains environments like mine where since the robot stops frequently 

Hum while writing this comment, I realize that what I want for my particular application is the solution 2. The global planner is responsible for avoiding obstacles and RPP for following the path if it is free...

So what is the goal of RPP ? :)

[image]

SteveMacenski, 2021-11-24:
> But currently we project using the velocity command, not the current velocity

Correct, because we're checking if our newly computed velocity command is _safe_. That is the point, I believe that is straight forward?

> and ignore the angular velocity except for pure rotation ... since we check the collision in rotation only in pure rotation

We are always using angular velocity https://github.com/ros-planning/navigation2/blob/main/nav2_regulated_pure_pursuit_controller/src/regulated_pure_pursuit_controller.cpp#L436 for our collision checks. This is always taken into account. That's why when you visualize the RPP arc you will see an "arc" not a straight line https://github.com/ros-planning/navigation2/blob/main/nav2_regulated_pure_pursuit_controller/src/regulated_pure_pursuit_controller.cpp#L446. We are projecting forward the entire velocity command, linear and angular, [over timesteps to check at each cell marching forward in both normal path tracking and pure rotations](https://github.com/ros-planning/navigation2/blob/main/nav2_regulated_pure_pursuit_controller/src/regulated_pure_pursuit_controller.cpp#L416-L420) . Does this make sense now?

> if the purpose is to check that there is no obstacle on the path, then we should project the poses along the path.

That is not the purpose, because while the robot will generally speaking follow the path, this algorithm does not exactly mimick each point of the path. So we need to check that our requested velocity command is going to be in collision if we follow that velocity for some time (using the parameter [max allowed time to collision](https://github.com/ros-planning/navigation2/blob/main/nav2_regulated_pure_pursuit_controller/src/regulated_pure_pursuit_controller.cpp#L83) as  that amount of time we project forward by to make sure we're always `N` seconds from collision at all times. 

>  if the path does a 90° turn with a small radius, the current implementation triggers collision detections + drawing

Indeed, that I believe is what we're looking to solve or mitigate! Obviously your image would be of a very large `max allowed time to collision`, but I understand the image is to illustrate the point, which it does a good job of! 
 
> So what is the purpose of the collision check of RPP ? :)

To make sure the robot won't collide with something while trying to follow the path, inexactly but roughly

SteveMacenski, 2021-12-01:
@Kaetera what do you think?

SteveMacenski, 2021-12-08:
@Kaetera wanted to get your thoughts - I can implement this but wanted a sanity check from another interested party.

sathak93, 2021-12-08:
@SteveMacenski I think the above said method might not work when the robot is approaching the goal. since the distance to goal will eventually becomes less than lookahead dist.  the warnings may be reduce a bit.

SteveMacenski, 2021-12-08:
The lookahead point is always at most at the goal [see here](https://github.com/ros-planning/navigation2/blob/main/nav2_regulated_pure_pursuit_controller/src/regulated_pure_pursuit_controller.cpp#L376-L379), so I think that resolves your concern?

Kaetera, 2021-12-10:
So this change the purpose of this feature toward something closer to my point 2. :)

This sounds ok to me.

I think it would be more intuitive and fit more real-life applications. For industrial application, the "stop before collision" part would typically be handled by safety-class scanners with protective field lengths computed according to the speed and the max time to collision (note that the same "goal point" or "curve" problem may then also arise...). 
We ended to set a` lookhead_dist = maximum desirable time to collision * maximum speed` when we were experimenting with RPP at the beginning, several month ago. I think this approach with checking collision between the lookahead point and the robot is more intuitive, and would simplify a bit the tuning of RPP.

SteveMacenski, 2021-12-13:
> the "guarantee to stop before collision" part would typically be handled by safety-[systems]

Agreed, I thought I could have it here too, but turns out not the case.

Action plan:
- Check up to and including the carrot along the curve, but no further
- Update parameter name to make it clear it will check time ahead to that limitation
- Update documentation / readme to make clear its up to the carrot, so if moving fast, far away carrot positions can be important not just for stability, but for predictive collision checking

SteveMacenski, 2021-12-15:
- doc updates https://github.com/ros-planning/navigation.ros.org/pull/253
- PR with implementation https://github.com/ros-planning/navigation2/pull/2741

@Kaetera @sathak93 would you be willing to test and see that this resolves you concerns?

closed 2021-12-16 by SteveMacenski