ros-navigation/navigation2 #2673: Severe warning of nav2_amcl at runtime

gezp, 2021-11-16:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu 20.04
- ROS2 Version:
  - Rolling binaries
- Version or commit hash: 
  - `main` branch
- DDS implementation:
  - default DDS

#### Steps to reproduce issue
```
ros2 launch nav2_bringup tb3_simulation_launch.py
```

#### Expected behavior
* no severe warning
#### Actual behavior
```bash
[amcl-7] Warning: class_loader.impl: SEVERE WARNING!!! A namespace collision has occured with plugin factory for class nav2_amcl::DifferentialMotionModel. New factory will OVERWRITE existing one. This situation occurs when libraries containing plugins are directly linked against an executable (the one running right now generating this message). Please separate plugins out into their own library or just don't link against the library and use either class_loader::ClassLoader/MultiLibraryClassLoader to open.
[amcl-7]          at line 253 in /opt/ros/rolling/include/class_loader/class_loader_core.hpp
[amcl-7] Warning: class_loader.impl: SEVERE WARNING!!! A namespace collision has occured with plugin factory for class nav2_amcl::OmniMotionModel. New factory will OVERWRITE existing one. This situation occurs when libraries containing plugins are directly linked against an executable (the one running right now generating this message). Please separate plugins out into their own library or just don't link against the library and use either class_loader::ClassLoader/MultiLibraryClassLoader to open.
[amcl-7]          at line 253 in /opt/ros/rolling/include/class_loader/class_loader_core.hpp
[amcl-7] Warning: class_loader.ClassLoader: SEVERE WARNING!!! Attempting to unload library while objects created by this loader exist in the heap! You should delete your objects before attempting to unload the library or destroying the ClassLoader. The library will NOT be unloaded.
[amcl-7]          at line 127 in /tmp/binarydeb/ros-rolling-class-loader-2.1.2/src/class_loader.cpp

```
#### Additional information

reason:
* motion model is refactored as a plugin library `motions_lib`, and `amcl_core` is also a plugin library (rclcpp component), `target_link_libraries(amcl_core  motions_lib)`  causes this severe warning.
* AmclNode should load motion model at runtime, so AmclNode shouldn't link this plugin library.

currently, we can launch and run nav2 successfully, because we don't use `AmclNode` as rclcpp component (plugin). But this severe warning will cause a runtime Error when we load  `AmclNode`  rclcpp component.

SteveMacenski, 2021-11-16:
If you can submit a PR to unlink it, I'd appreciate that

SteveMacenski, 2021-11-16:
The unloading warning doesn't sound good either, I think that's probably due to this PR destroys the class loader in the context rather than storing it as part of the class like the other instances in Nav2 do. I'm a little unhappy that the user submitting this patch did not mention these errors in his PR causing this regression...

gezp, 2021-11-16:
> The unloading warning doesn't sound good either

there are some unloading warning when i close nav2 program. but i think it's caused by https://github.com/ros-planning/navigation2/pull/2562, these warnings are also generated from `controller_server`, etc.

```
ros2 launch nav2_bringup tb3_simulation_launch.py use_composition:=False
```
the output when i close nav2
```
[planner_server-10] Warning: class_loader.ClassLoader: SEVERE WARNING!!! Attempting to unload library while objects created by this loader exist in the heap! You should delete your objects before attempting to unload the library or destroying the ClassLoader. The library will NOT be unloaded.
[planner_server-10]          at line 127 in /tmp/binarydeb/ros-rolling-class-loader-2.1.2/src/class_loader.cpp
[controller_server-9] Warning: class_loader.ClassLoader: SEVERE WARNING!!! Attempting to unload library while objects created by this loader exist in the heap! You should delete your objects before attempting to unload the library or destroying the ClassLoader. The library will NOT be unloaded.
[controller_server-9]          at line 127 in /tmp/binarydeb/ros-rolling-class-loader-2.1.2/src/class_loader.cpp
```

SteveMacenski, 2021-11-16:
That shouldn't be an issue if you bring down the lifecycle state machine properly before hitting control+C, no? That's the shutdown procedure for lifecycle systems, so if you kill it in live action, you will get some errors across the board -- and largely should be expected because when you control+C, its not able to call the destructors properly in all situations (and the actual unloading is happening in lifecycle transition states which then would never be called).

If any happen when we shut it down properly, then that's a problem

poornimajd, 2021-11-16:
> The unloading warning doesn't sound good either, I think that's probably due to this PR destroys the class loader in the context rather than storing it as part of the class like the other instances in Nav2 do for the lifecycle of the program. I'm a little unhappy that the PR with this patch did not mention these errors...

I am really Sorry for missing out on the severe warning.

As rightly pointed out by @gezp motion_models is redundant.

poornimajd, 2021-11-17:
I tried to change the CMakelists.txt and removed the redundant code.I have tested the changes locally and the warnings are not there anymore,except the unloading warning for amcl and controller_server when we press ctrl+c.Should I open a separate PR for the same?

closed 2021-11-18 by SteveMacenski