ros-navigation/navigation2 #2676: Smac Hybrid Planner reinitialization issues

zoltan-lengyel, 2021-11-16:
I have a unit test setup for the Smac Hybrid Planner that works without starting ROS nodes.
I noticed that I have different test results when I run only a single test, instead of running all the test cases together.
I investigated the issue, and I found some problems in the nav2_smac_planner::HybridMotionTable initialization.
My test cases only affected each other, because the motion_table is a static variable in the NodeHybrid class.
There is no problem of course if the program (node) is restarted every time, but the HybridMotionTable initialization code looks like it wants to support reinitialization, but doesn't do it correctly.

**Issues**

node_hybrid.cpp, HybridMotionTable::initDubin and HybridMotionTable::initReedsShepp
```
  // if nothing changed, no need to re-compute primitives
  if (num_angle_quantization_in == num_angle_quantization &&
    min_turning_radius == search_info.minimum_turning_radius)
  {
    return;
  }
```
I believe this check is wrong. It skips the primitive setup when switching between DUBIN and REEDS_SHEPP motion models without changing anything else. This same check is present in both motion model init functions.

node_hybrid.cpp, HybridMotionTable::initDubin  
```
  // Create the correct OMPL state space
  if (!state_space) {
    state_space = std::make_unique<ompl::base::DubinsStateSpace>(min_turning_radius);
  }
```  
node_hybrid.cpp, HybridMotionTable::initReedsShepp  
```
  // Create the correct OMPL state space
  if (!state_space) {
    state_space = std::make_unique<ompl::base::ReedsSheppStateSpace>(
      min_turning_radius);
  }
```
These checks will skip the state_space reinitialization in case the motion_model or the min_turning_radius is changed.

**Changes**
I fixed the issue on my side by completely removing these checks, because I don't mind the slower reinitialization. Otherwise, for more proper checks the motion table should know its current motion model.

SteveMacenski, 2021-11-16:
It should matter, because that function is called every time a new goal is requested https://github.com/ros-planning/navigation2/blob/0e52d90abf542726a156eafad304854272e0dcf9/nav2_smac_planner/src/a_star.cpp#L112 so making this semi-optimized is important for planning times. 

I'd accept a PR to add in the additional checks in the `if` statements to support that reconfiguration. I think in the `init` functions you just need to add a couple more checks and in the `if (!state_space)` just needs a check also if radius or model has changed

SteveMacenski, 2021-11-17:
Merged

closed 2021-11-17 by SteveMacenski