ros-navigation/navigation2 #2686: Turning radius gets downsampled/adapated  even when "downsample_costmap" param is deactivated.

salmagro, 2021-11-18:
## Bug report

**Required Info:**
- Operating System:
  - Debian GNU/Linux 10 (buster)
- ROS2 Version:
  - `Rolling`
- Version or commit hash:
  - Branch `main`: `c65532f7e537c05c8d02c45099c42cf7ed8954bb`

#### Steps to reproduce the issue

Assign parameters to:
```
/planner_server:
    GridBased:
      downsample_costmap: false
      downsampling_factor: 4
```
The `downsampling_factor` is applied even though the `downsample_costmap` is set on  `False`. 
See: https://github.com/ros-planning/navigation2/blob/0abd2699e86fc27ed8d1c56931f3441c2208da12/nav2_smac_planner/src/smac_planner_hybrid.cpp#L134-L140

#### Actual behavior
For testing, I am changing the `downsampling_factor` while `downsample_costmap` is disabled. As can be seen in the images, the factor affects the turning radius.

[image]
[image]
[image]

## Feature request

#### Feature description

Check the parameter flag `downsample_costmap` in L134-L140. Proposed solution:

```cpp
  // convert to grid coordinates
  const double minimum_turning_radius_global_coords = _search_info.minimum_turning_radius;
  if (!_downsample_costmap)
  {
	  _downsampling_factor = 1;
  }
  _search_info.minimum_turning_radius =
		  _search_info.minimum_turning_radius / (_costmap->getResolution() * _downsampling_factor);
  _lookup_table_dim = static_cast<float>(_lookup_table_size)
		  / static_cast<float>(_costmap->getResolution() * _downsampling_factor);
`

```

SteveMacenski, 2021-11-18:
Indeed, though if you specify a parameter that is actively incorrect for your situation, that is not good either. But I can see the point of doing some casual testing you'd get funky results. I'd love a PR to fix that. 

You could set it to `1` in `on_configure()` if the downsampler parameter is `false` so that even if you did set it to something else, it would be thrown out and be OK. You'd have to add that same logic in the dyanamic parameters callback.

SteveMacenski, 2021-11-19:
Merging shortly

closed 2021-11-19 by SteveMacenski