ros-navigation/navigation2 #2689: Lifecycle manager crash on startup during heavy CPU load

afrixs, 2021-11-18:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Galactic binaries
- Version or commit hash:
  - 0f6f49a3633de66db7cdf5f60ecbb951fea5883f
- DDS implementation:
  - Eclipse Cyclone DDS

#### Steps to reproduce issue

This happened when I launched nav2 bringup while many CPU-expensive nodes were running (Gazebo, data processing, object detection, ...).

Most easily reproducible by adding 
```
std::this_thread::sleep_for(500ms);
```
at the end of `LifecycleManager::LifecycleManager(const rclcpp::NodeOptions & options)` method and launching
```
ros2 launch nav2_bringup bringup_launch.py 
```

#### Expected behavior
LifecycleManager starts.

#### Actual behavior
LifecycleManager crashes in `createLifecycleServiceClients()` method on `shared_from_this()` as the shared pointer has not been created yet when `init_timer_` fires.

#### Additional information
Can be fixed by removing `callback_group_` from `init_timer_` construction, so that the callback is fired from `rclcpp::spin(node)` after the shared ptr has been created for sure. Cannot say whether this change affects negatively different ways of bringing up the manager (other than `ros2 launch nav2_bringup bringup_launch.py `)

SteveMacenski, 2021-11-18:
That makes entire sense. Please submit a PR and I'd merge.

gezp, 2021-11-19:
>  I think that was just 1 extra callback group entry that wasn't strictly necessary for just the initialization timer to trigger after constructor completion.

yes, i agree with you, removing `callback_group_ ` from `init_timer_` construction has no negative effect. 
in addition, i look at some code about  `LifecycleManager` , `LifecycleManagerClient` , `Bond`.  in fact, i think only  `manager_srv_` need extra callback group , because it will call some services which maybe cause block (maybe wrong).

closed 2021-11-19 by SteveMacenski