ros-navigation/navigation2 #2690: Smac Hybrid planner in default configuration ignores high-cost obstacles

mangoschorle, 2021-11-18:
I am experiencing unintuitive results regarding the SMAC Hybrid planner.
It appears to me that the planner's exploration is excessively dominated by distance-, rather than obstacle-costs.

I have set up a toy environment, where the distance-wise shortest path is obstructed by a large high-cost (251) area. 

I would expect the planner to favor the (distance-wise) slightly longer path alternative path with no high-cost obstacle in between. However, the planner chooses to traverse the high-cost region.

This behavior persists irrespective of the parameterized cost_penalty. 

[image]

When shifting the starting position slightly (towards the alternative path without the high-cost are) the solution changes.

Issue:
The planner produces results, which seem to be heavily biased towards distance rather than costmap-cost. The cost_penalty coefficient has no apparent influence (neither had the analytic_expansion_ratio).

[image]

Questions
Am I missing an apparent property of the SMAC planner, is this the expected behavior, or may there actually be some issues with the two heuristics.
**Required Info:**

- Operating System:
  - Used the provided docker- development environment on a Ubuntu 20.04 host

#### Steps to reproduce issue
I have attached the maps and the parameterization 
[parameter_and_map.zip](https://github.com/ros-planning/navigation2/files/7563140/parameter_and_map.zip)

#### Expected behavior
Planner balances geometrical distance with costmap-cost and is affected by cost_penalty 

#### Actual behavior
Planner solution is heavily biased towards short solutions

SteveMacenski, 2021-11-18:
A couple of things

First see https://github.com/ros-planning/navigation2/issues/2659 which meets the thrust of your ticket. This is known and has a known solution that is being worked on (probably?). Towards the end, it will just plow through higher cost situations if there's a valid solution because that's what analytic expansion means.

Now there are 2 more "actual" clarifications I have to make from your toy situation. This is where a _real_ situation you would see success in but because you've set up an unrealistic  setting, you're getting unrepresentative results.

A.

You need to have a potential field in your map. I try to be as [explicit as I can](https://github.com/ros-planning/navigation2/tree/main/nav2_smac_planner#potential-fields) about that in the documentation. In fact, all planners in this repo you really should be doing that with, its due to poor historical examples that people just casually inflate and move on. You should always desire to have a smooth potential field so that the planners can steer away from obstacles proactively before running into them in search expansions. While that wouldn't help in some respects, it will significantly speed up the planning process as well as plan in the middle of corridors rather than skirt the edges like your example shows. Nav2 planners are not research planners built to work on binary maps to prove an algorithm, they're production implementations that take into account cost information to be drivable and safe. 

B. 

The reason your robot chooses one corridor over another has almost nothing to do with that cost zone. Hybrid-A* is a **kinematically feasible** planner, meaning it generates drivable paths in the presence of real physical constraints. In your 2 examples, the reason it goes down the higher cost hallways is because you've set your initial pose in a way that it is physically impossible to have a drivable path in the other direction. Look at the minimum turning radius in the rest of the path, its not possible for the robot to rotate that tightly to even get to the other hallway because its too narrow and would hit a wall. Therefore the planner has no choice but to go down the higher cost hallway, because the other is entirely unreachable.

SteveMacenski, 2021-11-18:
I think that fully answers all of the topics possible with this question - if there's more feel free to ask and I'll answer,but closing since this is not an actual bug anywhere and there's no actionable changes for the repo (I believe).

mangoschorle, 2021-11-19:
Thank you very much for your thorough explanation. 
The example i provided stems from one of our industrial plants. To get behind the mechanics of the planner I did indeed highly simplify the scenario. The center area (roughly 50mx40m)  is where production takes place and vehicles are not granted access. This area is surrounded by a traffic lane, where vehicles may freely navigate.

Concerning your statement in B, I must unfortunately disagree. The starting points in both plans **have the same orientation**.  They only differ in translation (1 meter along the lower path). To take the low-cost path the vehicle must do nothing but drive forwards. To take the upper, high-cost path the vehicle must do a turning maneuver (which it does). I provided close-up images where this behavior becomes more apparent.

Concerning your statement A, I have seen the previous issue #2659. The high cost-area is some 50m away from the goal and starts at around 50% of the path. I didn't consider this to be _around the end of the plan_.

Anyway, I integrated your advice in my setup. I added very smooth inflations around the obstacles, and made a very smooth transition into the high cost area. I also shifted it further towards the start to not run into the analytic expansion issue. The issues persist. Instead of just driving forward along the low cost route the planner produces a result, where the vehicle actively changes orientation to then plough thorough a smoothly increasing high-cost area.

Shown below is the new setup. Again, the **start in the right is just shifted by one meter in translation** (to demonstrate the crucial bias towards distance rather than anything the costmap says). In my opinion this cost-neglecting behavior will make future endeavors e.g. working with costmap filters to create preferred lanes ect... extremely cumbersome.

[image]

The following is a closeup of the starting-area. 

[image]

As a side note, but this is a different issue.
I observed that the smoother unfortunately seems to destroy the kinematic feasibility in the beginning. The picture below shows this. On the left you have the path before smoothing. It appears to be fine concerning the kinematic constraints. On the right is the smoothed result
[image]
.

mangoschorle, 2021-11-19:
This is the result of Theta*. To me ,it seems to produce more explainable results.

[image]

SteveMacenski, 2021-11-19:
Ah, I see, the close up images help alot, thanks. That is a serious smoother hiccup, I haven't seen it quite that bad before, that will require some attention. It was a known issue I describe in detail here https://github.com/ros-planning/navigation2/pull/2660#pullrequestreview-800764769 but was told that people really didn't functionally see it so I passed it off. 

Now that I understand better, what I think is happening is that our [search heuristic](https://github.com/ros-planning/navigation2/blob/main/nav2_smac_planner/src/node_hybrid.cpp#L462-L488) is looking for a 2D non-feasible path to the goal that is cost aware to steer the search. Once we're in the expansion phase, we use that and costed-traversal functions to avoid higher cost regions, but if the shortest path through the system itself goes through a higher cost zone than a competing branch solution, then it just does it anyway (but will avoid costs along that branch solution fine, just chooses the wrong branch). In this ticket https://github.com/ros-planning/navigation2/issues/2498 I'm discussing with some other folks about changing this heuristic computation to include more search time in the heuristic to keep going until the graph has been expanded with a solution and exhausting search in parallel branches until the cost is higher than the alternative solution. That would let the search continue to expand down the other hallway to find a solution and potentially use that solution instead to guide the heuristic search.  

tl;dr I see 2 things
- Smoother needs to deal with boundary conditions better
- Heuristic expansion needs to keep it up

I think the heuristic item is already pretty much done and we're waiting on @robotechvision to submit a PR (we recently did a big Smac Planner framework re-architecture for State Lattice that was merged a few days ago, so getting back on these smaller issues) so I don't think there's really any action you need to take there. We should have a solution there shortly. 

I would however really love your help on improving the boundary conditions for the smoother to make this more reasonable.

afrixs, 2021-11-20:
I believe the heuristic will resolve the problem, @mangoschorle could you check out [our fork](https://github.com/robotechvision/navigation2/tree/smac_planner_safe) (branch smac_planner_safe) and test the planner on your map with parameter `obstacle_heuristic_admissible: true` (or send us your map and nav2 params)?

mangoschorle, 2021-11-20:
@afrixs The changes appear to resolve the issue. I am getting very intuitive results. Also the cost_penalty parameter also yields the response one would assume. Fantastic!

[image]

Unsurprisingly, the issue with the smoother is still there. It quite consistently, in the beginning of the path invalidates the min turning radius constraint.

[image]

SteveMacenski, 2021-11-21:
@afrixs thanks! Do you have a potential idea of when you may be able to submit a cleaned up version as a PR?

afrixs, 2021-11-22:
2 months I guess

SteveMacenski, 2021-11-22:
With the most recent updates, isn't that mostly nulled out? [That's just the setup now after the last PR was merged](https://github.com/ros-planning/navigation2/blob/main/nav2_smac_planner/src/node_hybrid.cpp#L465-L466) for all situations.

I think the only thing that needs to be changed now is the queuing changes you made to the heuristic to continue expansion until the cost is the same as the solution cost. The computation of the Heuristic and Travel costs should be admissible now

afrixs, 2021-11-23:
Oh I see the updated computation, we didn't use it in our admissibility fix, but it's good to speed up the truly admissible search (as described below) or improve quality of the path for the current non-truly-admissible search and I think it will be OK from admissibility point of view.
(The term can be optimized into `((i <= 3) ? 1.0 : sqrt_2)*(1.0 + (cost_penalty * cost / 252.0))`)

However, what I was talking [here](https://github.com/ros-planning/navigation2/issues/2498#issuecomment-966080340) (the first two paragraphs) about, is need to update also [this line](https://github.com/ros-planning/navigation2/blob/ed30d704e66770d7683267a3f6f85f7d6fa31ebb/nav2_smac_planner/src/node_hybrid.cpp#L317) in the same manner, i.e. to
`NodeHybrid::travel_distance_cost * (1 + motion_table.cost_penalty * normalized_cost)`

Just after this change the search is truly admissible, behaving, like @mangoschorle called it, intuitively, i.e. finding optimal path with respect of the actual cost penalties given in parameters. However, it will be slower to a certain degree (although I'm curious about how updating the term in heuristic like you did will speed it up), so I would recommend an on/off parameter for it ('on' for safety critical searches, 'off' for others).

So yes, the update can be divided into 2 PRs,
 1. Using the Differential A* 2D search for obstacle heuristic to solve the two different-cost passages problem (and other problems from the same class)
 2. Updating the [traversal cost term](https://github.com/ros-planning/navigation2/blob/ed30d704e66770d7683267a3f6f85f7d6fa31ebb/nav2_smac_planner/src/node_hybrid.cpp#L317), making the search truly admissible. It's just 1 line but needs maybe some speed tests and to add a re-tuned recommended configuration as it changes a lot.

I will create a partially cleaned-up PR for (1.) so that @SteveMacenski can perform speed tests as well and then we can discuss about whether to use an on/off parameter for this part too or if the old code can be fully replaced.

SteveMacenski, 2021-11-24:
> However, what I was talking here (the first two paragraphs) about, is need to update also this line in the same manner, i.e. to
NodeHybrid::travel_distance_cost * (1 + motion_table.cost_penalty * normalized_cost)

I think that was an oversight on my part. I updated it to include the cost again with my old formulation while I was testing around the change to the H computation. I added the cost there and then moved onto the state lattice admissibility, (which is that)[https://github.com/ros-planning/navigation2/blob/main/nav2_smac_planner/src/node_lattice.cpp#L287]. It looks like I just didn't go back and update how that was being computed as well. 

> I would recommend an on/off parameter for it ('on' for safety-critical searches, 'off' for others).

If it makes a big difference, I like that idea. If it doesn't, I'm OK just updating it to use the fully admissible formulation. 

I can look at characterizing the topics in (2) tomorrow.

SteveMacenski, 2021-11-24:
I characterized it on some first order tests I've been using - I see that this change is about ~2x slower but I'd need some more testing to really narrow into the specifics. I see that it expands about an order of magnitude more search iterations to get a path that is slightly different, but often not consequentially so. 

However, when I kill the penalty functions for non-straight / change, I see the numbers go back down to the prior numbers.

**That makes me think that we can make the fully admissible work**. I'll just need to retune the defaults... again. This is not something I'm going to get to today,  but certainly next week.

SteveMacenski, 2021-11-29:
So in this I've found that there's a balance that needs to be struck between path quality and run-time using this formulation. After the smoother, things run pretty good actually. However, since long term we are going to move the smoothers into another server, I'm concerned about the raw search-quality path with this formulation where the run-time is in the similar ballpark. 

Using the same defaults (change 0.2, non-straight 1.25) the planner is about ~1.5-2.5x slower than it was before. This is where the path quality is about on par. To get performance to about only ~1.2-1.5x slower (what I think is a reasonable trade off using change ~0.0-0.10, non-straight ~1.15-1.2), the path quality is pretty not amazing directly out of search - but perhaps livable. After the basic smoother, its unnoticeable that there's been a change. 

So that speaks to me that perhaps having some rough smoothing in the planner's output might be good even if being fed into another smoother later. I'm not in love with that idea -- or I suppose we could just make the smoother server be on by default so that this isn't an issue for most users with the default behavior tree. 

I'll submit a PR shortly for you to take a look at and let me know what you think / test

SteveMacenski, 2021-11-30:
https://github.com/ros-planning/navigation2/pull/2713

SteveMacenski, 2021-12-06:
https://github.com/ros-planning/navigation2/pull/2727 also allows you to turn off the smoother via `smooth_path: false` so that you can disable it due to current issues. I'm working on a fix, but it won't be a complete fix, it will be a rough fix to improve behavior but still will be capable of generating infeasible smoothing effects. The real smoother described in #2498 from previous work of mine that @afrixs picked up and fixed up will be kinematically feasible.

SteveMacenski, 2021-12-07:
I have a prototype going right now, needs some serious clean up, but it proves the point that this works! I should be able to push a solution before the weekend for you to test @mangoschorle to confirm it resolves your concerns.

SteveMacenski, 2021-12-09:
https://github.com/ros-planning/navigation2/pull/2731 Please check it out @mangoschorle - makes the smoother feasible again!

mangoschorle, 2021-12-09:
I check it out and get back to you. Should be no later than beginning of next week.

afrixs, 2021-12-09:
@SteveMacenski I haven't done the downsampler work we were talking about, but something with downsampler was needed to be done for the heuristic to be admissible: use _min_ function instead of _max_ for downsampling (since heuristic cost must be <= real cost) so I had to add a param to choose the function in downsampler.

Currently, as far as I remember, the only extra things in there are the analytic expansion constraints mentioned [here](https://github.com/ros-planning/navigation2/issues/2659#issuecomment-960997194) + some debug logs/visualizations etc.

Currently I'm working on ceres smoother cleanup/tests + I have some deadlines until 2021/12/17 but I think I will be able to create a PR until Christmas.

SteveMacenski, 2021-12-09:
@mangoschorle that looks very good to me, but I do notice that the angle seems a little off in the middle example for the starting heading. I'm not sure why that would be given the solution, it should be rather exact. I'd be interested if you added a print [here](https://github.com/ros-planning/navigation2/pull/2731/files#diff-64cb9b4335e41c3dd5b807ad6aa9805a3fd4456673681efe2e6a5d0c0b1d49e7R448) (or in the `End` counter part depending if that's the start/end pose) to see if its being overrided or not in that situation. I wonder if the boundary conditions didn't update due to a failure somewhere. I tested on 0.2-0.4 m turning radius. I'll test on some larger ones like you have today. Or this may not actually be an issue and its my eyes playing a trick on myself. 

Something that brought to my mind though is that instead of using the starting / ending poses from the `path` generated by seearch, we can instead have the smoother be passed the starting / ending poses from **the planning request** so that you actually would have more exact boundary condition matching than would be possible from pure search alone, which is limited by the number of angular quantizations you selected in the parameters. I'm not sure you'd notice much from Hybrid-A* (with defaults at 5 deg increments) but for state lattice, that would make a real difference since that only has 16 bins (22.5 deg, on average). 

---

@afrixs thanks for the update. I look forward to it! I'll probably leave that to you then and move onto some other Smac / Nav2 related problems in the meantime, but happy to help if its useful.

mangoschorle, 2021-12-09:
@SteveMacenski your "here" isn't very specific about the line number you want your print at.
I used 64 angle quantizations. I will add the start and goal from the request to the path and let you know what happens.

mangoschorle, 2021-12-09:
Smoother result where I added the start (from request) to the plan.
After smoothing, the first point in the result plan has a yaw value of 44.4774 degrees as compared to 0 degrees for the desired start.
[image]

In this example i set the smoother iterations to zero, and again, added the first point in the result plan. Interestingly: even though the smoother iterations is set to zero, the first point in the plan has a changed yaw (3 degrees) after smoothing, i.e. the smother alters the first point of the plan even with num_smoother_iterations = 0
[image]

mangoschorle, 2021-12-09:
@SteveMacenski 
There is no printing, since
  if (best_expansion_idx > boundary_expansions.size()) 
evaluates to true and enforceStartBoundaryConditions returns

mangoschorle, 2021-12-09:
The points collide hence the function returns no valid index.

Here, in

// Check for collision
    unsigned int mx, my;
    costmap->worldToMap(x, y, mx, my);
    if (costmap->getCost(i) >= INSCRIBED) {
      expansion.in_collision = true;
      return;
    }
    
    You are checking the for loop index.
    
    Are you sure this shouldnt be more like the linear index as inferred from     costmap->getIndex(unsigned int mx, unsigned int my)

SteveMacenski, 2021-12-09:
Ahhh! Thanks. I didn't catch that - thanks for looking it over :smile:

SteveMacenski, 2021-12-09:
FYI - I see that the smoothed path starts at a different position than the original searched path due to the discrete error into translating the global map poses into the discrete grid search space. So this actually makes the paths better, but it does look a little odd if you don't know why that would be happening.  

[image]

SteveMacenski, 2021-12-10:
I was planning on merging this today, but I had some issues come up that I'm not entirely sure how they developed when I picked this up this afternoon. This is delayed until further notice -- I'm having an issue where we get lots of strong deviations from the path now and I'm not sure why I didn't see this at all yesterday but now seems like it happens ~50% of the time. Such large deviations are not acceptable as they can drive the robot closer to obstacles rather than just refining the approach.

SteveMacenski, 2021-12-13:
Smoother work to be merged shortly. Renaming the ticket to reflect the current scope to be fixed: Obstacle heuristic

afrixs, 2021-12-22:
Update: can't make the PR until Christmas, you can expect it around 2022-1-15

SteveMacenski, 2022-01-19:
Update: @afrixs has done some awesome work and his PR is close to being merged. We're going to strip out my initial implementation and replace with his and then largely it should be good to go and will resolve this issue.

mangoschorle, 2022-01-24:
@SteveMacenski I checked out the new work of  @afrixs, played around with the retrospective_penalty and checked the resuts. Great work!

SteveMacenski, 2022-01-29:
Merged and tuned to my satisfaction, should be good to go!

closed 2022-01-29 by SteveMacenski