ros-navigation/navigation2 #2710: [Smac planner] Minor anomalies in path

afrixs, 2021-11-26:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Galactic binaries
- Version or commit hash:
  - 0f6f49a3633de66db7cdf5f60ecbb951fea5883f
- DDS implementation:
  - Eclipse Cyclone DDS

#### Steps to reproduce issue
 1. Plan a path via Hybrid or Lattice planner
 2. Examine it in detail

#### Expected behavior
Path is kinematically feasible and pose intervals are approximately same.

#### Actual behavior
Path has minor anomalies, which are usually OK (and therefore unnoticed until now), but can potentially cause problems in some cases are cheap to fix.

#### Implementation considerations
While doing detailed admissibility analysis I noticed two minor problems:
 1. Analytic expansion path skips poses as noted [here](https://github.com/ros-planning/navigation2/blob/b8d3611d52df78be4ddb099f62c93cc4392cbed4/nav2_smac_planner/src/analytic_expansion.cpp#L223). It's not a big deal when 1 pose is skipped, but this happens commonly for 3-4 consequent poses. Can be fixed by creating a new node instead of skipping, adding it to a list of `_detached_nodes` and cleaning the list up after backtracing the path (e.g. in call of `_expander->cleanup()`)
 2. Original pose restoring which is done [here](https://github.com/ros-planning/navigation2/blob/b8d3611d52df78be4ddb099f62c93cc4392cbed4/nav2_smac_planner/src/node_hybrid.cpp#L643), [here](https://github.com/ros-planning/navigation2/blob/b8d3611d52df78be4ddb099f62c93cc4392cbed4/nav2_smac_planner/src/node_lattice.cpp#L526) and [here](https://github.com/ros-planning/navigation2/blob/b8d3611d52df78be4ddb099f62c93cc4392cbed4/nav2_smac_planner/src/analytic_expansion.cpp#L201) should be done also [here](https://github.com/ros-planning/navigation2/blob/b8d3611d52df78be4ddb099f62c93cc4392cbed4/nav2_smac_planner/src/a_star.cpp#L318), in an `else {}` clause.
 
For this to be done `getNeighbors()` must retrieve `NodeCandidateVector` instead of `NodeVector` (where `NodeCandidate` would be a pair of `<NodePtr, Coordinates (i.e. initial_pose_coords)>`.

Alternatively (and maybe more elegantly) `getNeighbors()` could be changed to `expandToNeighbors()` and called with an additional parameter
```
std::function<bool(const nav2_smac_planner::NodeHybrid * & node, nav2_smac_planner::NodeHybrid * & neighbor)> & AddNeighborToSearchIfInteresting
```
where the [for loop body](https://github.com/ros-planning/navigation2/blob/b8d3611d52df78be4ddb099f62c93cc4392cbed4/nav2_smac_planner/src/a_star.cpp#L309) would be performed, allowing node to restore the coords if neighbor not added to graph.

SteveMacenski, 2021-11-29:
I have as well noticed that and found the same conclusion that it didn't impact feasibility so I passed over it. But if it actually causes an issue in application, we can resolve it (once again, thank you for your extreme attention to detail, I appreciate the push back to improve things. I think this is the last un-written quirk of Smac that you haven't already brought up). 

<details>
  <summary>I believe unimportant thoughts due to the comment below, but want to leave in case that doesn't work out</summary>
  
For 1, I think you could simply reset `_detached_nodes` at the end of `tryAnalyticExpansion`, I think that would have the same effect without adding a new API,  we do not need to store that between runs of attempting expansion. 

For 2, `getNeighbors` is what generates the initial poses, so adding an `else` at the `if (g_cost < neighbor->getAccumulatedCost())` would be resetting the "initial pose" back to `0,0,0` as if never visited. Further, there's some queuing logic in `NodeBasic` to store the current state of the pose information for a specific node that is being expanded, such that we can support many `NodeT`'s and deal with the case that a single node has been queued  multiple times. But it should only be visited / expanded into 1 time. Setting the poses back to initial I think would mess that process up and I'm not sure if it would be necessary. 

The node is already added to the graph [here](https://github.com/ros-planning/navigation2/blob/b8d3611d52df78be4ddb099f62c93cc4392cbed4/nav2_smac_planner/src/a_star.cpp#L245-L254) regardless if we add the node to the priority queue for potentially being visited. When you use the [node getter](https://github.com/ros-planning/navigation2/blob/b8d3611d52df78be4ddb099f62c93cc4392cbed4/nav2_smac_planner/src/a_star.cpp#L122), it will return the instance of the node at those indices if it exists, or create a new one if it does not already exist. 
</details>

---

This check https://github.com/ros-planning/navigation2/blob/b8d3611d52df78be4ddb099f62c93cc4392cbed4/nav2_smac_planner/src/analytic_expansion.cpp#L222-L231 exists because of "loops" that can theoretically be found in the path using analytic expansions. The aim was that if there was a loop, that the fact that we visited the same node 2x wouldn't cause is harm via an infinite pointer loop. Thinking about that context, I think there's a simpler solution. 

Maybe, a better check here is to check instead not if it was `visited` but if the particular node is either
- A: In the current analytic expansion path already (which... shouldn't be possible I do not believe)
- B: In the current global path from search. That is the situation we're trying to avoid. And we're perfectly find overriding visited cells values at this point because `setAnalyticPath` is only called when we've found a valid analytically expanded path to the goal and we're immediately going to return it and exit the planning problem. 

We could use the `node` to virtually backtrace and store a vector of indices of points in its path. If one of those indices matches the `expanded_nodes` indices, we drop it. Otherwise, we override it because we don't care if we invalidate that node since the planning problem if about to be cleared out.

afrixs, 2021-11-30:
> For 1, I think you could simply reset `_detached_nodes` at the end of `tryAnalyticExpansion`
`backtracePath` would crash if the nodes have been deleted in `tryAnalyticExpansion`.

On the backtracing alternative solution you propose, yes, that is possible but I think it's too complicated for such a minor fix... It would add computation time and would be harder to implement, but it is possible to do it this way (I would suggest to use at least `set` instead of `vector` of indices for speedup) if you consider adding one `_expander->cleanup()` call too ugly.

> adding an `else` at the `if (g_cost < neighbor->getAccumulatedCost())` would be resetting the "initial pose" back to 0,0,0 as if never visited
No, this wouldn't happen since `if (g_cost < neighbor->getAccumulatedCost())` is `true` if node has never been visited

> Further, there's some queuing logic in `NodeBasic` to store the current state of the pose information for a specific node
Oh I see the fix has already been done using `NodeBasic`, sorry I didn't notice it. However, as I see the speed and API is important for you, the `NodeBasic` queueing could be avoided and a lighter, more elegant and faster version with `expandToNeighbors()` could be used, but this can be done in another issue/PR as it is just a question of computation time/API.

SteveMacenski, 2021-11-30:
> backtracePath would crash if the nodes have been deleted in tryAnalyticExpansion.

:+1: 

> if you consider adding _expander->cleanup() call too ugly.

I don't mind it that much, just trying to think about how to do it with the fewest changes as possible. But if its necessary, its necessary. 

> yes, that is possible but I think it's too complicated for such a minor fix... According to speed tests I did expandToNeighbors approach brings minor speedup (in addition to being a bit easier to read)

Personally, I find that to be the easiest solution to think about, but that's only important if I'm the one doing it. If you have something working already, please submit a PR and happy to go the other route if its already done - especially if its faster. 

I just don't like the idea of putting this block https://github.com/ros-planning/navigation2/blob/b8d3611d52df78be4ddb099f62c93cc4392cbed4/nav2_smac_planner/src/a_star.cpp#L303-L320 into the `NodeT`s since it is part of the core A* algorithm. Putting that stuff in there would break my intended separation of concerns to have a complete A* algorithm that can work across any Node search type. If we could put that back in place, that would be most ideal. I'm not sure moving that block into the NodeT's would speed up search, so I don't think that's where the 10% speed up came from?

SteveMacenski, 2021-11-30:
https://github.com/ros-planning/navigation2/pull/2717 backwards fix will be merged once CI is complete

afrixs, 2021-12-01:
Most important for us is the analytic expansion nodes skipping avoidance, I'll create a PR for it tomorrow.

Other things are just ideas for speed-up, but I'll address them here since I've done some work with them:

> I just don't like the idea of putting this block into the `NodeT`

Yeah, I think most of it would be needed to be put into the `NodeT`. Although I originally thought the block could remain in a_star.cpp in similar way like the `neighborGetter`, but I found that to be really effective, only the `addNode` line could remain in `a_star.cpp` from the whole block. However, I agree that it looks better when it's in `a_star.cpp` and we don't care so much about speed, so I'll leave the decision upon you. Just sharing my observations from further speed tests:

 * I performed a test for a pure node graph search using `cache_obstacle_heuristic: True` (i.e. not biased by costmap downsampling and obstacle heuristic search), which showed that the change* sped up the expansions by 32% instead of 10% (70ms -> 48ms for a search with ~56000 expansions) so maybe it starts to be interesting
 * I performed a valgrind analysis, showing an interesting fact that the costmap downsampling in `resetObstacleHeuristic` took 40ms, i.e. 40% of whole search (map size was 80mx80m, resolution 0.05, but the duration will rise quadratically for larger maps). I see a space for optimization here (e.g. lazy downsampling)

change*: Yeah, moving the block from one place to another doesn't speed up anything but placing the `g_cost` check into the node can simplify current process of
```
 1. for all neighbors: store initial pose and replace by new one
 2.                    if new pose is valid then replace motion state with new one
 3.                    else restore initial pose
 4. return all valid neighbors
 5. for all valid neighbors: compute `g_cost` and check if interesting
 6.                          if interesting, then store current pose and motion state to NodeBasic and add to queue
 7. when retrieving from queue, restore current motion state (because it could have been replaced meanwhile)
```
to
```
 1. for all neighbors: place the new pose to a helper static node
 2.                    if valid then: place new motion state to the helper node
 3.                                   compute `g_cost` using the helper node and check if interesting
 4.                                   if interesting, then copy the pose and motion state from helper node to neighbor and add to queue
```
It does have some elegancy flaws (helper node, more boilerplate in `NodeT`s) but is faster as restorals and the `NodeBasic` maintenance is not needed.

SteveMacenski, 2021-12-01:
We already have a few Smac tickets open where we're talking about a breadth of topics. I'm starting to have a hard time keeping track what we're talking about where :disappointed_relieved:. This covers the analytic expansion dropping poses on visited nodes, lets handle that first and then circle back to performance improvements. 

Maybe open a separate ticket for `Smac Potential Performance Improvements` for that discussion. Though admittedly, I'm biased toward fixing the things we currently have open (cost aware smoother plugin, obstacle H improvements, this) first since those fix issues. But I'd be happy to also discuss performance improvements to squeeze as much performance as can get out of it, while keeping with the core design. I think there's a middle ground we can hit where it keeps the "core A*" in place but streamlines the getting neighbors with these checks using iterators and adding an extra input field -- though admittedly I'm not entirely sure what the 'helper node' is in your breakdown. 

The downsampler was added because running the obstacle heuristic for the full resolution of costmap was really, really bad for performance. So I wanted to make it a little more sparse to make it faster but still maintaining relative quality of the costs (since a 2x downsample isn't very significantly different). But there's definitely more improvements that can be made there and I'm open to that.  

For now, I'd like to not put part of the A* core algorithm in the `NodeT`'s for the purposes of fixing the analytic expansion issue, but we could talk it over for performance improvements.

afrixs, 2021-12-02:
> I'm starting to have a hard time keeping track what we're talking about where

Sorry for digression, just wanted to finish the analysis so that I can clean it up for good and checkout to currently more important fixes.

> The downsampler...

I find the 2x downsampling quite good, just thinking of a lazy downsampler (cells are downsampled in the same way but only on demand, not all at once), it would significantly improve the performance for short paths in large maps, while not adding big overhead (just one more `if`)

SteveMacenski, 2021-12-02:
That's a good idea!

SteveMacenski, 2021-12-02:
After we close out this ticket with a fix for the anomalies due to analytic expansion crossing over with visited cells, I'll open a new ticket around Smac optimizations and post what we discussed above for further action.

afrixs, 2021-12-02:
Yes, that would do the job, ideally the approaches could be compared for speed since the `getDownsampledCost(u, v)` method can bring more overhead when the same cell is requested multiple times but on the other hand spares time elsewhere - no downsampled map allocation etc. Either way, I think both approaches could live inside the current downsampler (and enabled/disabled by a flag, e.g. `eager`/`lazy`/`cached_lazy`) and no separate implementations are needed.

SteveMacenski, 2021-12-02:
> when the same cell is requested multiple times

When `getDownsampledCost()` is called we cache it for re-requests to get a stored value? Similarly, if we do that in a data structure, the look up in something like an `unordered_map` may be heavier than just recomputing it. 

Potentially, we could keep a cached `VisitedDownsampledCells` vector that we could check if a non-`-1` value exists, if so, use it, if not do  `getDownsampledCost()` and then store the results in `VisitedDownsampledCells` for re-requests. 

But agreed, that's tricky business.

afrixs, 2021-12-02:
Yes, vector or a simple C array would be better than map. After all Costmap2d uses C array of `uchar`s for storing costs so the indexes would be at least compatible.

SteveMacenski, 2021-12-03:
https://github.com/ros-planning/navigation2/issues/2725 I opened a new ticket specifically on performance optimizations topics for that continued discussion

closed 2021-12-03 by SteveMacenski